Ysurac/openmptcprouter #665: Error messages on LUCI.

Marctraider, 2019-09-18:
## Expected Behavior

No error message in luci.

## Actual Behavior

Both System -> System and Network -> DHCP and DNS throw same error.

## Steps to Reproduce the Problem

  1. Clean factory image flash
2. observe error navigating to these luci elements.

## Specifications

    OpenMPTCProuter version: 0.52.2
    OpenMPTCProuter VPS version: 0.1001
    OpenMPTCProuter platform: WRT3200ACM

[image]

Marctraider, 2019-09-18:
Today, I saw it only once, and now i can both access System -> System and Network -> DHCP and DNS without any warning like this.

I have not rebooted since.
Looks like its random bug?

Ysurac, 2019-09-18:
I think it's more a browser cache issue.

Marctraider, 2019-09-18:
> 
> 
> I think it's more a browser cache issue.

You might actually be right as it has disappeared completely now. Not the first problem with Openwrt and firefox in general :)

Sorry to bother you with this.

closed 2019-09-18 by Marctraider