Ysurac/openmptcprouter #668: Glorytun TCP can't start up

JorgeMoort89, 2019-09-18:
If I install VPS side on fresh installed machine I have no problem at all, on a cloned machine I have this problem:
Connecting OMR with public IP and KEY give me this error:

OMR side
[image]

VPS side:
[image]

How can I check the log to fix this problem?

## Specifications

  - OpenMPTCProuter version: 0.49
  - OpenMPTCProuter VPS version: 0.9962
  - OpenMPTCProuter platform: x86_64

Marctraider, 2019-09-18:
You run on old version?

JorgeMoort89, 2019-09-18:
Yes, 0.49 on OMR and 0.9962 on VPS

Ysurac, 2019-09-18:
I don't make support on old release.
Glorytun TCP seems to no be able to start, maybe a problem with your clone but it's not related to OpenMPTCProuter

closed 2019-09-18 by Ysurac