Ysurac/openmptcprouter #669: No AES HW encryption support by default on 0.52.3 installation

xabix, 2019-09-19:
## Expected Behavior

Hello,

I read on other issues: https://github.com/Ysurac/openmptcprouter/issues/653#issuecomment-530524444 that for people who have AES CPU that aes-256-gcm should be activated by default at the installation. Therefore I have decided to reinstall to start from a clean sheet of paper :)

But at the end Glorytun had still chacha ticket and shadowsocks was still refereing to chacha20 and not aes-256-gcm

` PACKAGE:     openmptcprouter
 VERSION:     v0.52.3
-----------------------------------------------------
root@OpenMPTCProuter:~# cat /proc/cpuinfo | grep aes
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ss syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon rep_good nopl xtopology cpuid tsc_known_freq pni pclmulqdq ssse3 fma cx16 pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand hypervisor lahf_lm abm 3dnowprefetch cpuid_fault invpcid_single pti ssbd ibrs ibpb fsgsbase tsc_adjust bmi1 hle avx2 smep bmi2 erms invpcid rtm mpx rdseed adx smap clflushopt xsaveopt xsavec xgetbv1 xsaves arat md_clear`

## Actual Behavior
VPN->GloryVPN says
[image]
and 
[image]

To avoid to reinstall again should:
1/ put aes-256-gcm manually, and
2/ untick chacha20 in GloryVPN page too?

## Steps to Reproduce the Problem

  1.Fresh install

## Specifications

  - OpenMPTCProuter version: 0.52.3
  - OpenMPTCProuter VPS version: 1001
  - OpenMPTCProuter platform: x86_64

xabix, 2019-09-19:
I have done **manually** 1:
[image]

and 2:
[image]

but I get "Can't get public IP address from ShadowSocks" on the VPS on and off. With no many details on the logs

OMR says:
`Thu Sep 19 12:35:05 2019 user.notice omr-bypass: Restart dnsmasq...
Thu Sep 19 12:35:05 2019 daemon.info dnsmasq[7830]: exiting on receipt of SIGTERM
Thu Sep 19 12:35:05 2019 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Thu Sep 19 12:35:05 2019 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Thu Sep 19 12:35:05 2019 user.notice dnsmasq: Allowing RFC1918 responses for domain plex.direct
Thu Sep 19 12:35:05 2019 user.notice omr-bypass: OMR-ByPass is running
Thu Sep 19 12:35:05 2019 daemon.info dnsmasq[17466]: started, version 2.80 cachesize 150
Thu Sep 19 12:35:05 2019 daemon.info dnsmasq[17466]: DNS service limited to local subnets
Thu Sep 19 12:35:05 2019 daemon.info dnsmasq[17466]: compile time options: IPv6 GNU-getopt no-DBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
Thu Sep 19 12:35:05 2019 daemon.info dnsmasq[17466]: using local addresses only for domain test
Thu Sep 19 12:35:05 2019 daemon.info dnsmasq[17466]: using local addresses only for domain onion
Thu Sep 19 12:35:05 2019 daemon.info dnsmasq[17466]: using local addresses only for domain localhost
Thu Sep 19 12:35:05 2019 daemon.info dnsmasq[17466]: using local addresses only for domain local
Thu Sep 19 12:35:05 2019 daemon.info dnsmasq[17466]: using local addresses only for domain invalid
Thu Sep 19 12:35:05 2019 daemon.info dnsmasq[17466]: using local addresses only for domain bind
Thu Sep 19 12:35:05 2019 daemon.info dnsmasq[17466]: using nameserver 127.0.0.1#5353
Thu Sep 19 12:35:05 2019 daemon.info dnsmasq[17466]: using local addresses only for domain lan
Thu Sep 19 12:35:05 2019 daemon.info dnsmasq[17466]: read /etc/hosts - 4 addresses
Thu Sep 19 12:35:05 2019 daemon.info dnsmasq[17466]: read /tmp/hosts/dhcp.cfg01411c - 0 addresses
Thu Sep 19 12:35:12 2019 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Thu Sep 19 12:35:20 2019 daemon.info vnstatd[8411]: Info: Traffic rate for "tun0" higher than set maximum 10 Mbit (20s->27262976, r4289707906 t4290871192, 64bit:0), syncing.
Thu Sep 19 12:35:36 2019 daemon.info omr-tracker-ss: Shadowsocks is up (can contact 212.27.48.xx)
Thu Sep 19 12:37:57 2019 daemon.info omr-tracker-ss: Shadowsocks is down (can't contact 103.224.182.xx, 198.27.92.xx, 88.191.250.xx)
Thu Sep 19 12:37:57 2019 user.notice Shadowsocks: Rules DOWN`

VPS says:
`Sep 19 14:39:12 XabiX systemd[1]: Started Session 18 of user root.
Sep 19 14:39:12 XabiX ss-server[697]:  2019-09-19 14:39:12 ERROR: failed to handshake with 92.184.110.xx: invalid address type
Sep 19 14:39:18 XabiX ss-server[697]:  2019-09-19 14:39:18 ERROR: failed to handshake with 92.184.110.xx: invalid address type`

Should I reboot or change anything somewhere else for the handshake to work correctly?

Merci
XabiX

Ysurac, 2019-09-19:
Same as #653 
Can you do a ```cat /proc/cpuinfo``` on the VPS ?
The cipher is not changed on the VPS side, before changing it all was green on the status page ?

Ysurac, 2019-09-20:
When wizard is used for first time it check if aes flag is on the router and on the VPS, if both have aes flag then it's enabled. I think there was problems to communicate with the admin script.

xabix, 2019-09-21:
can't this be automatically enabled? not only at the installation but anytime the HW detected on both sides support HW encryption?

At least if a note can be displayed to notify what best HW encryption options are avalaible.

I ll close the issue. Shame we couldn't identify the issue to fix it (anyway I discovered a new option merci:))

Ysurac, 2019-09-21:
No and this will not be the case. If a user choose another encryption I will not change it automatically, so it's only on install.
I can make an option on wizard to choose available encryption, maybe this would be easier.

closed 2019-09-24 by xabix