Ysurac/openmptcprouter #670: Multipath Seems to be Blocked

ccmks, 2019-09-20:
## Expected Behavior

 Utilized all WAN interface to get the benefit of aggregation

## Actual Behavior
Only WAN1 and WAN3 are according to graph in MPTCP bandwidth aggregating but the WAN2 not and showing Multipath seems to be blocked

[image]

## Steps to Reproduce the Problem

  1. Fresh install on both client and VPS
  2. Configure All WAN to use DHCP
  3. Run the following test:

OMR-test-speed
% Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 10.0G    0 84.7M    0     0  2079k      0  1:24:01  0:00:41  1:23:20 1825k^C

OMR-test-speed WAN1
 % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 10.0G    0 31.2M    0     0  2388k      0  1:13:10  0:00:13  1:12:57 2359k^C

OMR-test-speed WAN2
 % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 10.0G    0  311k    0     0   9840      0  12d 15h  0:00:32  12d 15h  4827^C

OMR-test-speed WAN3
 % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 10.0G    0  120k    0     0   4509      0  27d 13h  0:00:27  27d 13h  7670^C

I do notice that both WAN2 and WAN3 run slow on OMR speed test. However, when I connect to them individually and do some browsing and run download test using IDM, they are all showing decent speed. with WAN2 speed is 7 Mbps and WAN3 speed is 15 Mbps

For aggregation test, I noticed that the aggregated speed is slower than just single WAN connection which is very ironic. It defeat the purpose of having aggregating WAN speed if the single WAN connection is faster. 

I also simply download various download file from:
http://speedtest-sgp1.digitalocean.com/1gb.test
http://www.ovh.net/files/1Gio.dat

Using IDM to download those files above and getting result:

- 2000 - 2300 KBps for aggregating download but only show up WAN1 and WAN3 are doing the aggregation whereas the WAN2 just sitting idle

## Specifications

  - OpenMPTCProuter version: 0.52.3
  - OpenMPTCProuter VPS version: 0.1001 4.19.67 using both OVH and Vultr for VPS test
  - OpenMPTCProuter platform: Both Banana Pi R2 and x86

Ysurac, 2019-09-20:
For speed test you should wait more than 30s (1-2 minutes).
But I think there is a problem for wan2 and wan3 (or a problem to contact OVH server from these providers). Latency on wan3 seems to be very high.
For wan2, if multipath is really blocked you have to use glorytun UDP or MLVPN and disable shadowsocks on the wizard if you want to aggregate, but you need to fix wan2 & wan3 speed problem before that.

ccmks, 2019-09-20:
Aside using OMR test, what other tools I can use to test the WAN2 and WAN3 speed? How can I also test the speed performance on the VPS. Keep in mind that I am using 2 different VPS (OVH SG and Vultr SG) with identical setup and configuration. Both of VPS location are located in Singapore whereas my location is in Indonesia.

The reason why I choose Singapore server because I couldn't find any decent VPS in Indonesia that has decent overseas connection.

Ysurac, 2019-09-20:
No other tools available. This only download a file using wan to test speed.
wan1 and wan2 latency to the VPS are good. Only a 835ms latency on wan3 that is really too big.

To test speed on the VPS you can download the same file: ```curl -4 http://proof.ovh.net/files/10Gio.dat >/dev/null || echo```

ccmks, 2019-09-21:
Already run the test on VPS and the result looks good
% Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  6 10.0G    6  658M    0     0  10.2M      0  0:16:36  0:01:04  0:15:32 11.0M^C

Also just checked on the router page and the WAN3 is showing low latency. 

[image]

But still WAN2 not utilizing its bandwidth

omr-test-speed
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  2 10.0G    2  276M    0     0  2999k      0  0:58:15  0:01:34  0:56:41 3126k

[image]

Please advise.

Thank you

Ysurac, 2019-09-21:
As I said, if multipath is really blocked you have to use glorytun UDP or MLVPN and disable shadowsocks on the wizard if you want to aggregate.
You can check where it's blocked using tracebox : ```omr-tracebox wan2``` (when you have -TCPOptionMPTCPCapable there is a problem)

ccmks, 2019-09-21:
> As I said, if multipath is really blocked you have to use glorytun UDP or MLVPN and disable shadowsocks on the wizard if you want to aggregate.

I disabled the shadowsocks and use Glorytun UDP. The speed is very slow. Only getting around 3 Mbps when do aggregating, although the aggregating works on all WAN interface. However, the speed is way too slow compare to WAN1 speed which is capable 20 Mbps.

I tried to use MLVPN, but never get it to work. Keep showing the VPN is not running or down on status page. As the result, I got no internet on using MLVPN.

> You can check where it's blocked using tracebox : `omr-tracebox wan2` (when you have -TCPOptionMPTCPCapable there is a problem)

Here is the result as I don't know how to read it
1: 192.168.40.1 
2: * 
3: 10.198.248.113 TCP::DataOffset TCP::CheckSum IP::DiffServicesCP IP::TotalLength IP::TTL IP::CheckSum +TCPOptionMaxSegSize 
4: 10.199.100.82 TCP::DataOffset TCP::CheckSum IP::DiffServicesCP IP::TotalLength IP::TTL IP::CheckSum +TCPOptionMaxSegSize +TCPOptionPad -TCPOptionMPTCPCapable 
5: 112.215.37.54  [PARTIAL] +PartialTCP IP::DiffServicesCP IP::TotalLength IP::TTL IP::CheckSum 
6: 210.176.138.230  [PARTIAL] +PartialTCP IP::DiffServicesCP IP::TotalLength IP::TTL IP::CheckSum 
7: 210.176.138.229 TCP::DataOffset TCP::CheckSum IP::DiffServicesCP IP::TotalLength IP::TTL IP::CheckSum +TCPOptionMaxSegSize +TCPOptionPad -TCPOptionMPTCPCapable 
8: 63.218.165.85 TCP::DataOffset TCP::CheckSum IP::DiffServicesCP IP::TotalLength IP::TTL IP::CheckSum +TCPOptionMaxSegSize +TCPOptionPad -TCPOptionMPTCPCapable 
9: 63.217.24.86  [PARTIAL] +PartialTCP IP::DiffServicesCP IP::TotalLength IP::TTL IP::CheckSum 
10: 10.79.1.161  [PARTIAL] +PartialTCP IP::DiffServicesCP IP::TotalLength IP::TTL IP::CheckSum 
11: 10.79.0.230  [PARTIAL] +PartialTCP IP::DiffServicesCP IP::TotalLength IP::TTL IP::CheckSum 
12: * 
13: 45.77.41.111 TCP::SrcPort TCP::DstPort TCP::SeqNumber TCP::AckNumber TCP::DataOffset TCP::Flags TCP::WindowsSize TCP::CheckSum IP::TotalLength IP::Identification IP::TTL IP::CheckSum IP::SourceIP IP::DestinationIP +TCPOptionMaxSegSize -TCPOptionMPTCPCapable 

Result:
< Ethernet (14 bytes) :: DestinationMAC = 00:15:5d:22:c7:04 , SourceMAC = d4:a1:48:3c:2b:f6 , Type = 0x800 , >
< IP (20 bytes) :: Version = 4 , HeaderLength = 5 , DiffServicesCP = 0 , ExpCongestionNot = 0 , TotalLength = 44 , Identification = 0x0 , Flags = 2 , FragmentOffset = 0 , TTL = 54 , Protocol = 0x6 , CheckSum = 0x504 , SourceIP = 45.77.41.111 , DestinationIP = 192.168.40.100 , >
< TCP (20 bytes) :: SrcPort = 65101 , DstPort = 58473 , SeqNumber = 586494693 , AckNumber = 133758709 , DataOffset = 6 , Reserved = 0 , Flags = ( SYN ACK ) , WindowsSize = 29200 , CheckSum = 0xa70d , UrgPointer = 0 , >
< TCPOptionMaxSegSize (4 bytes) :: Kind = 2 , Length = 4 , MaxSegSize = 1380 , >

Ysurac, 2019-09-21:
The problem:
```4: 10.199.100.82 TCP::DataOffset TCP::CheckSum IP::DiffServicesCP IP::TotalLength IP::TTL IP::CheckSum +TCPOptionMaxSegSize +TCPOptionPad -TCPOptionMPTCPCapable```
So your ISP seems to really block MPTCP.
For Glorytun UDP, did you set speed of each wan in the wizard ?

ccmks, 2019-09-21:
> For Glorytun UDP, did you set speed of each wan in the wizard ?

No I didn't. Should I set it? If so, I can set it on WAN1 since it is a fiber connection and fixed limit speed. However, how should I set for WAN2 and WAN3? Those 2 WAN connections are LTE which doesn't have any fixed speed limit.

Their average speed is so variable sometimes faster than 20 Mbps sometimes, even slower than 10 Mbps, depending on congestion in my area. This applies to any LTE based ISP.

Ysurac, 2019-09-21:
Yes you have to for Glorytun UDP. For LTE you can try to fix the max speed you can get with them.

ccmks, 2019-09-21:
I have set the speed per your instruction. Here is what I set

WAN1 Fiber - 20Mbps down and 4Mbps up
WAN2 LTE - 100Mbps down and 100Mbps up (LTE theoretical Speed)
WAN3 LTE - 7Mbps down and 2Mbps up (This is the max speed I can get from them)

However, the aggregation speed doesn't improve. The best speed I can get around 6 Mbps

Ysurac, 2019-09-21:
Can you give me the result of ```glorytun-udp path dev tun0``` on the router via SSH ?

ccmks, 2019-09-21:
Here is the result

path UP
  status:   OK
  bind:     192.168.80.10 port 5000
  public:   125.162.209.121 port 5000
  peer:     45.77.41.111 port 65001
  mtu:      1464 bytes
  rtt:      33.092 ms
  rttvar:   2.471 ms
  rate tx:  4096000 bytes/sec
  rate rx:  20480000 bytes/sec
  total tx: 658369 packets
  total rx: 1131522 packets
path UP
  status:   OK
  bind:     192.168.40.100 port 5000
  public:   112.215.173.148 port 5000
  peer:     45.77.41.111 port 65001
  mtu:      1436 bytes
  rtt:      105.296 ms
  rttvar:   18.426 ms
  rate tx:  102400000 bytes/sec
  rate rx:  102400000 bytes/sec
  total tx: 3237026 packets
  total rx: 1350976 packets
path UP
  status:   OK
  bind:     192.168.8.100 port 5000
  public:   120.188.84.42 port 54312
  peer:     45.77.41.111 port 65001
  mtu:      1472 bytes
  rtt:      86.479 ms
  rttvar:   15.351 ms
  rate tx:  4096000 bytes/sec
  rate rx:  8192000 bytes/sec
  total tx: 433982 packets
  total rx: 470407 packets

ccmks, 2019-09-21:
You issued me the new release https://www.openmptcprouter.com/release/4.19/ from the other thread. Do you have the Hyper-V image (VHD) for me to test on my router?

Ysurac, 2019-09-21:
It's same image as .img.gz, only converted using qemu-img.

ccmks, 2019-09-21:
So for the ISP that doesn't support MPTCP, is there anyway for me to use shadowsock? I believe, shadowsocks is the fastest method to do the aggregation.

Ysurac, 2019-09-21:
I can do nothing about that.

ccmks, 2019-09-21:
I can draw conclusion here then, if the ISP doesn't support MPTCP, then pretty much you won't be able to use any TCP based VPN to do aggregation, instead use UDP based. But the drawback using UDP based VPN is the speed is not fast, even not faster then single WAN connection, per my scenario.

Ysurac, 2019-09-21:
True. MPTCP support multipath only for TCP. So if MPTCP doesn't work, TCP based VPN can't do aggregation.
UDP VPN available in OpenMPTCProuter use own aggregation system, but I never get really good result with them.
If you know other choices, I'm open to test anything...

ccmks, 2019-09-21:
How about FEC Udp, like Updspeeder or TinyFECVPN?

ccmks, 2019-09-21:
Also, I am thinking about using Softether as another VPN layer to bypass ISP that doesn't support MPTCP routing. Basically, you create tunnel between client to server for each WAN using softether, then use MPTCP shadowsocks to do the bonding. I have been using softether VPN for years and it runs great and solid!

What is your advise?

Llorx, 2019-10-14:
> Also, I am thinking about using Softether as another VPN layer to bypass ISP that doesn't support MPTCP routing. Basically, you create tunnel between client to server for each WAN using softether, [...]

Seems like a great idea. I'm also going to test this, as I have 1 ISP that also strips MPTCP header.

Llorx, 2019-10-14:
Ok, its been a few hours since I tried this VPN thingy and now I'm pretty lost.

I've managed to create a `tun0` with OpenVPN, binding to the interface that has MPTCP headers stripped. Its IP is `10.0.1.2` and the server one is `10.0.1.1`.

Now I can send traffic directly to my server IP `10.0.1.1`, and it replies back without problem.

The thing is that now I can't tell him to connect to anything outside the server. I mean, if I do:
```
curl --interface tun0 ifconfig.co
```
It won't work. It says that the connection is refused. But if I do
```
curl --interface tun0 10.0.1.1
```
It works (I have a web server right there).

I'm kinda lost at this point...

Ysurac, 2019-10-15:
You may have to configure shorewall on the VPS side.

Llorx, 2019-10-15:
I've disabled both firewalls and still nothing... :-/

ccmks, 2019-10-16:
@Llorx Great! I knew it is going to work!

However, your next step is to configure the tunnel interface to be member of MPTCP so aggregation will work over tunnel instead of direct WAN interface.

Also will need to see its performance (internet speed and CPU usage) since you are aggregating the connection over tunnel.

Ysurac, 2019-10-16:
I'm testing with wireguard for now. I will see what is the result...

Llorx, 2019-10-16:
I had to create a tap interface and disable all firewalls (just in case, don't want to have problems with that). I had to set the server to `server-bridge nogw` in bridged mode and create `tun0` interface in the client like so:
```
config interface 'vpn0'
        option ifname 'tap0'
        option proto 'dhcp'
        option ipv6 '0'
        option metric '5'
        option defaultroute '0'
        option peerdns '0'
        option multipath 'on'
```
The dhcp proto will get an IP from the server DHCP server and the router configure all routes when gets up.

Remember that all interfaces must allow promiscuous mode and mac spoofing (just in case you are under a VM. I had a lot of headaches because of that).

Also, in the client I had to configure vpn with `local [interface without MPTCP enabled IP]` and `lport 0`.

With this I got everything working. I only have one problem now: https://serverfault.com/questions/988279/openvpn-tap-interface-not-connecting-to-server-public-ip-through-tunnel

max4186, 2019-10-19:
> I'm testing with wireguard for now. I will see what is the result...

Hopefully you can get WireGuard working, as far as I know it actually does support MPTCP well so will be another option for a low overhead high speed VPN.

Llorx, 2019-10-19:
The VPN don't has to go over MPTCP. Must be a tunnel just before MPTCP thingy happens.

Ysurac, 2019-10-21:
I'm not able to bind softether or Wireguard to a specific interface (even if this may be possible), so I tested with OpenVPN and finally I get something that look like a working result with shadowsocks over OpenVPN over UDP (I will make more tests).
This feature need many changes, so I will add them asap to develop branches.

ccmks, 2019-10-22:
@Ysurac Great work! I appreciate your effort on getting this even better.

Just a suggestion, I would use UDPSpeeder (UDP FEC) for better UDP performance. According to the guy who wrote this protocol, the UDPSpeeder can be combined with OpenVPN TCP VPN and get very good result, better speed, a lot of stable performance.

Not sure if you can integrated tho

Llorx, 2019-10-28:
> @Ysurac Great work! I appreciate your effort on getting this even better.
>
> Just a suggestion, I would use UDPSpeeder (UDP FEC) for better UDP performance. According to the guy who wrote this [...]

There's a problem. You can't bind to an interface. All UDP traffic will go through first `eth`. I tried to fiddle with iptables but was impossible. I created an issue just in case: https://github.com/wangyu-/UDPspeeder/issues/224

ccmks, 2019-11-06:
@Llorx 

I am not sure what was the IPtable that you have tried so far. I am not a linux guy, so my knowledge of using iptable is very limited.

However, I am curious that if you have implemented the iptables with the following logical routing:

- Deny all traffic except going to VPS server on eth1
- Allow all traffic that going out to eth2 or the tunnel interface.

Basically, you are filtering the iptable on IP based which is layer 3 based, rather than the TCP/UDP based filtering

Llorx, 2019-11-06:
I'm rather newbie with iptables too so I'm sure I've doing something wrong. By the way I created a pull request to able to bind to an IP or to an interface: https://github.com/wangyu-/UDPspeeder/pull/227

closed 2019-12-20 by Ysurac