Ysurac/openmptcprouter #672: SEGFAULT in MLVPN on x86 client

heathfx, 2019-09-22:
## Expected Behavior

 mlvpn authenticated with VPS and ability to transmit packets via the VPS.

 IP lookup from a PC on the LAN reports "my ip" as the ip of the VPS.

## Actual Behavior
 observe a flood of messages in the serial console of the router/client:
```
mlvpn[11779]: segfault at b7f8dda2 ip b7f21a2c sp bfa5ae30 error 4 in libc.so[b7efe000+4b000]
[158948.598607] Code: 83 ec 1c 8b 5d 08 89 45 e4 8b 43 1c 85 c0 74 05 46 89 c3 eb f4 0f bf 43 40 83 ec 0c 6b c0 44 29 c3 8d 7b 3c 57 e8 d6 9c 01 00 
<66> 8b 43 42 83 c4 10 29 f0 66 85 c0 66 89 43 42 75 13 89 5d 08 8b
```
 no relevant log messages observed on the VPS while monitoring with journalctl -xef.

 status page on router reports "direct output" and is confirmed with an IP lookup from a PC on the LAN.

## Steps to Reproduce the Problem

  1. Install VPS on ubuntu 18.04 x64 and set boot kernel to vmlinuz-4.19.67-mptcp
  2. Verified VPS running kernel with uname -r (4.19.67-mptcp)
  3. open all ports as described in the VPS install instructions for ubunutu
  4. Install x86 client in KVM virtual machine
  5. configure 1 LAN interface and 2 WAN interfaces
  6. use wizard to configure VPS server ip and server key
  7. verify successful connection (indicated on status page and performing an internet speedtest with OpneMPTCProuter set as default gateway) using default shadowsocks and Glorytun config
  8. disable shadowsocks and set VPN to MLVPN
  9. reboot router VM and observe segfault errors flooding the serial console

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.52.3 r0+11006-713561a10b / LuCI Master (git-19.260.67281-bb408f2)
  - OpenMPTCProuter VPS version: 0.1001 4.19.67-mptcp
  - OpenMPTCProuter platform: x86

Ysurac, 2019-09-22:
Why do you really want to use MLVPN ?

heathfx, 2019-09-22:
less overhead than MPTCP

Ysurac, 2019-09-22:
MLVPN is not really and doesn't give good results for me. But it is available for config that work with it.
Subflow on fullmesh should never be increased.

heathfx, 2019-09-23:
I'll take your word for it, that MLVPN doesn't give good results, I'm sure you've done enough testing to know, but I still imagine that you would want to fix the segfault issue regardless.

I set "Fullmesh subflows for each pair of IP addresses" to 1 and when I do a speed test only one of my LTE links sees any sort of utilization. when i set it to 2, I get about 50% utilization on each link, when I set it to 4 I get about 90% utilization, I get a little more by going to 6 subflows, but then I get a kernel panic. How can I get good aggregation of my links with the subflows set to 1?

heathfx, 2019-09-23:
A new discovery: when setting MLVPN and then going back to glorytun, something gets permanently messed up with the routing table. forwarded ports yielded only 2mbps despite my upload being 20mbps, after installing a fresh router I got full utilization of both links with fullmesh subflows at 1, once i changed to mlvpn then back to glorytun, it performed like crap. so, something goes permanently wrong when switching between VPN modes.

Ysurac, 2020-03-10:
MLVPN support back in OpenMPTCProuter v0.54. Should work.

closed 2020-03-10 by Ysurac