Ysurac/openmptcprouter #673: VPN on VPS ?

iamfreaky, 2019-09-23:
## Expected Behavior

VPN-Client on VPS. All trafic from your network should go trough the VPN.

## Actual Behavior

VPN-Client can be etablished but no traffic get routed rough it. 

## Steps to Reproduce the Problem

  1. Install OooenVPN on VPS.
  2. Use a VPN Provider
  3. Try to etablis a VPN Connection

## Specifications

  - OpenMPTCProuter version:  v0.51.5
  - OpenMPTCProuter VPS version: 1000
  - OpenMPTCProuter platform: x86_64

Ysurac, 2019-09-23:
OpenVPN is installed by default on the VPS.
If you want an external VPN, you need to you configure it properly on the VPS, but it's out of the scope of OpenMPTCProuter.
You can also configure it on the router side.

closed 2019-09-23 by Ysurac

iamfreaky, 2019-09-23:
@Ysurac I already tried to use it on the Router side but performance is really bad. In fact worse than one link on its own. So this option pretty much doesnt work. 

SO how should i configure it on a VPS ? OpenVPN as a Client configures it itself on its own. It gets ignored by openmptcp

Ysurac, 2019-09-23:
You can use a VPN in TCP mode to get better performances.
On the VPS, there is nothing special, you need to set routes and configure the firewall.

iamfreaky, 2019-09-23:
@Ysurac Sure VPN in TCP is better, but also sadly not an Option, I already discussed this with my VPN Provider. 

Routes are set by OpenVPN automatically, so you mean Glorytun/Shadowsocks Routes to OpenVPN?

Ysurac, 2019-09-23:
no, I mean routes via OpenVPN. You should set/check them and configure the firewall.

iamfreaky, 2019-09-23:
okay then i will try this.

Th0rGer, 2020-01-06:
did it work?