Ysurac/openmptcprouter #675: Internet Speed is very less

ranusha14, 2019-09-25:
Before mptcp configuration wan1 speed is 80Mbps in web page.
After mptcp configuration wan1 speed is very less that is around 30 mbps in web page.
I have compiled openmptcprouter  for mediatek ralink mips MT7688 board.

## Specifications

  - OpenMPTCProuter version: (0.52.2)
  - OpenMPTCProuter VPS version: (0.1000)
  - OpenMPTCProuter platform: (MT7688)

Ysurac, 2019-09-25:
So it's not a supported platform. With a 580Mhz CPU and 256Mo RAM it's really too slow I think...
What do you mean with before and after mptcp configuration ?

ranusha14, 2019-09-25:
I mean to say that without VPS congfiguration i am getting 80Mbps.
After VPS configuration i am getting 20-30Mbps.

Ysurac, 2019-09-25:
OK. Shadowsocks and glorytun are encrypted by default, with a slow processor this can slow down the connection. By default chacha20 is used (or AES if there is hardware support), you can try another encryption.

closed 2019-12-20 by Ysurac