Ysurac/openmptcprouter #676: Problème reception mail (depuis mon serveur mail)

Slivikkus, 2019-09-25:
## Actual Behavior
Bonjour, j'ai un problème quand je suis censé recevoir un mail Google me dit que mon serveur refuse la connexion alors que ça ne peut pas être ça j'ai ouvert les port pour les mail mais il faut faire autre chose pour ça ?

## Steps to Reproduce the Problem

  1. Envoyer un mail depuis Gmail
  2. Attendre des heures avant le mail de Google
  3. Recevoir un mail de Google disant que le serveur refuse la connexion

## Specifications

  - OpenMPTCProuter version: (0.52.3)
  - OpenMPTCProuter VPS version: (0.999)
  - OpenMPTCProuter platform: (x86_64)

Ysurac, 2019-09-25:
Est-ce que la redirection des ports depuis le serveur est activé ? Dans Système->OpenMPTCProuter puis onglet paramètres avancés.

Slivikkus, 2019-09-25:
**Bien sûr puisque j'héberge des sites internet sans problème**

La réponse était :
The recipient server did not accept our requests to connect. Learn more at https://support.google.com/mail/answer/7720 [mail.xxxxx.com.: generic::failed_precondition: connect error (111): Connection refused]

Ysurac, 2019-09-25:
Pas d'IPv6 ?
Et que donne un dmesg sur le VPS ?

Slivikkus, 2019-09-25:
Non pas d'IPv6

Slivikkus, 2019-09-25:
et le dmesg me donne un tas d'info

Ysurac, 2019-09-25:
C'est le but ;) Mais il faut repérer si il y a quelque chose lié au port 25

Slivikkus, 2019-09-25:
SMTP : 587 (pour moi)

Mais c'est le port 143 qui pose problème

Slivikkus, 2019-09-25:
Et puis je ne sais pas comment je vais voir récupérer l'intégralité je contrôle directement le VPS à distance ducoup je ne peut pas tout voir

Ysurac, 2019-09-25:
Tu as essayé un telnet sur les ports voir si ça répond ?

Slivikkus, 2019-09-25:
Je fait le telnet ou ?

Ysurac, 2019-09-25:
sur l'IP du VPS

Slivikkus, 2019-09-25:
Je m'en doute mais sur quel appareil ?

Ysurac, 2019-09-25:
peu importe.

Slivikkus, 2019-09-25:
Je ne peut pas faire de telnet y'a pas la commande de partout

Slivikkus, 2019-09-28:
Comment je fait pour savoir ducoup ? @Ysurac

Slivikkus, 2019-10-01:
```
root@adky-vps-linux-1:~# telnet 51.75.202.56
Trying 51.75.202.56...
telnet: Unable to connect to remote host: Connection refused
root@adky-vps-linux-1:~#
```

Voilà ce que j'obtient

closed 2019-10-01 by Slivikkus