Ysurac/openmptcprouter #679: RPC error on admin/network/hosts

PxxBxx, 2019-09-29:
## Expected Behavior

Clean install, access to all network panels on Luci

## Actual Behavior

After a clean-install (dd if=/dev/zero on all storage beforehand), I can access the password settings, the SSH settings, the DHCP settings, but when I try to access `hosts`, I get an error :

RPC call failed with error -32002: Access denied
  at ClassConstructor.handleCallReply (http://192.168.100.1/luci-static/resources/rpc.js:58:16)

http://192.168.100.1/ubus/?1569742672305
0: {jsonrpc: "2.0", id: 1, error: {code: -32002, message: "Access denied"}}
  error: {code: -32002, message: "Access denied"}
  id: 1
  jsonrpc: "2.0"
1: {jsonrpc: "2.0", id: 2, result: [0, {changes: {}}]}

## Specifications

  - OpenMPTCProuter version: 0.52.3 ext4 image (non-UEFI)
  - OpenMPTCProuter VPS version: 0.1001
  - OpenMPTCProuter platform: x86_64

Ysurac, 2019-09-29:
Clean your browser cache.

PxxBxx, 2019-09-29:
lol, if it's that, I'll hurt myself (and it needs to be in a FAQ somewhere).
IT'S THAT !!! <====== I'm deeply sorry :'(

closed 2019-09-29 by PxxBxx