Ysurac/openmptcprouter #681: Problème ESCXI et changement de DNS

ProxFR, 2019-09-30:
Bonsoir,
Je souhaitais déjà vous remercier pour ce superbe projet qui m'aide beaucoup.
Cela fait trois jours que je prends en main OpenMPTCProuter, et j'ai eu pas mal de problèmes...
J'avais dans l'optique de l'utiliser pour additionner ma connexion ADSL (5 Mbit/s en down et 0.5 Mbit/s en up) et un routeur 4G (±100 Mbit/s en down et 20 Mbit/s en up).
J'avais la possibilité de l’installer sur mon serveur ESXI ou sur un RPi 3b.
Le RPi ayant une interface 100 Mbit/s, je choisis ESXI.

Mais premier problème, je n'ai jamais réussi à accéder aux réseaux WAN1 et WAN2 que j'avais configurés correctement. Même avec un PC dans le réseau impossible de ping les interfaces... J'ai suivi votre guide (https://github.com/Ysurac/openmptcprouter/issues/87) et l'unique différence est avec mon adaptateur réseau, vous conseillez le e1000, mais j'ai un VCenter Distributed Port Group, pour faire du load balancing. Peut-être que le problème venait de là ?

Du coup, après 2 jours d'essais infructueux, je me suis rabattu sur mon RPi.
Et là tout a fonctionné du premier coup ! Sauf, que j'ai un DNS sur mon Windows Server pour pouvoir me connecter à mon AD, et que dès que je passe un PC en adressage manuel, et que je le configure avec mon DNS je n'ai plus aucune connexion réseau... 
Mon but final serrait d'utiliser le serveur DHCP de mon NAS Synology, et le DNS du Windows Server. 

Merci par avance pour votre aide !

## Specifications

  - OpenMPTCProuter version: (v0.52.3 r0+11006-713561a10b)
  - OpenMPTCProuter VPS version: (0.1001 4.19.67-mptcp)
  - OpenMPTCProuter platform: (RPI3)

Ysurac, 2019-09-30:
Pour ESXI il faut pas oublier le mode promiscous. Sinon je ne peux pas aider, j'utilise virtualbox.
Il est fortement recommandé d'utiliser le DNS d'OpenMPTCProuter, sinon omr-bypass notamment ne fonctionnera pas correctement. 
Il faut que le DNS de windows server puisse aller sur internet et se connecter à des DNS qui fonctionne (je ne sais pas si il se connecte aux root DNS ou si il faut personnaliser les DNS qui ne doivent pas être ceux d'un FAI).

Scaff31, 2019-10-07:
Bonjour,
Donc je veux bien t'aider avec esxi perso, ca tourne sur un esxi avec 2vSwitch et des Vlans sans aucun souci.

Pour ta finalité il te suffit de configurer ton windows server avec comme DNS principal 127.0.0.1 et en secondaire l'ip de L'omr afin qu'il puisse forward ver l'omr-bypass 

Sur ton DHCP sygnology il te suffit de config les options de plage, en gateway tu met l'ip OMR et en DNS tu met l'ip de ton DNS

ProxFR, 2019-10-07:
Bonjour @Scaff31,
Merci de me proposer ton aide ;-) 
J'ai pu réessayer de déployer OMR sur ESXI ce weekend, j'ai bien vérifié que le mode promiscous était activé sur mon vSwitch mais toujours impossible d’accéder au reste du réseau et donc de configurer les WAN...
J'ai un T2600G-28TS derrière mon ESXI, est-ce lui qui pourrait poser un problème avec les VLANs ?
Du coup je suis toujours sur mon RPi 3 mais je trouve que ça manque de puissance...

Merci a @Ysurac pour ton aide (j'avais pas pris le temps de répondre)

Scaff31, 2019-10-07:
Si tu as deja des VLAN de configuré pense à modifier le type d'interface dans OMR

Onglet - Réseaux --> Interfaces --> Ton_WAN --> Parametres physiques 

Type d'interface = Défault

ProxFR, 2019-10-07:
Je viens de remonter un VM avec OMR.
Mon switch et configuré par défaut, il y a juste un groupe LAG de configurer sur les 4 interfaces du serveur ESXI.
J'ai suivi ta procédure :
Onglet - Réseaux --> Interfaces --> Ton_WAN --> Parametres physiques
Mais ça n'a pas arrangé mon problème... 

Voila ce que je vois dans la console :
[image]

Mais du coup état ne me rend rien et reste vide :
[image]

[image]

J'ai pas vraiment d'idée sur comment debugger ça...
Merci 😉

Ysurac, 2019-10-07:
Je ne sais pas si c'est lié, mais le masque de sous réseau de wan1 est trop large.

ProxFR, 2019-10-07:
C'est le masque qui est utilisé par mon architecture aujourd'hui : mon réseau est en 192.168.0.0/16
Mon but final est d'avoir le WAN1 (4G) en 172.16.0.0/24 le WAN2 (ADSL) en 172.16.1.0/24 et eth0 en 192.168.0.0/16 pour que mon architecture présente fasse le changement "invisiblement".
Mais je ne peux pas modifier l'adresse de WAN1 pour le moment comme la configuration ne fonctionne pas et que les machines sont dépendantes du serveur AD.

En tout cas ce plan d'adressage fonctionne sur le RPI.

Après un redémarrage, j'ai accès à l'état mais je ne vois plus mes interfaces :
[image]

[image]

J'ai uniquement changé ceci :
Onglet - Réseaux --> Interfaces --> Ton_WAN --> Parametres physiques
Type d'interface = Défault

Ysurac, 2019-10-07:
Logique, tu utilisais macvlan je pense et tu l'as désactivé.
Vu que tu es sous ESXI tu devrais créé 2 interfaces réseau supplémentaire dans ESXI au lieu d'utiliser MacVLAN.

ProxFR, 2019-10-07:
Je suis nouveau sur l'interface d'OpenWRT 😉 
J'ai du mal a voir ce que tu veux dire par créer 2 interfaces de plus sur ESXI... 2 interfaces qui seraient liées à mon interface virtuelle principale ?

ProxFR, 2019-10-07:
Je laisse la configuration réseau de la VM : 
[image]

@Scaff31 J'ai vu que tu recommandais d'utiliser le e1000 en NIC sur ESXI, mais je n'ai accès qu'au VMXNET 3. 

Mais d'un coté je me dis que ça ne vient pas de ça puisque je peux accéder a l'interface web. 
Si je comprends bien le fonctionnement de OMR, les interfaces WAN sont dans des VLANs séparée, et l'interface eth0 est en "direct" ? Donc mise a part les VLANs je ne vois pas trop de quoi cela peut venir...

Scaff31, 2019-10-07:
je me demande si ce n'est pas ton LAG qui pose problème.

Peux tu crée dans ton ESXI un groupe de port appeler WAN sans tag VLAN ni LAG que tu assigne a ta  VM et tu y colle ta 4g ou ton adsl peux importe et tu regarde dans l'interface OMR si le wan remonte

ProxFR, 2019-10-08:
Je viens de vérifier les paramétrés de mon vSwitch, et j'ai changé "Autoriser les transmissions forgées" et "Autoriser les modifications MAC" sur Oui. 
Et là toutes mes interfaces WAN se sont configurées sans problème ! Merci beaucoup ! 😉 

J'ai donc par la suite suivi la procédure de @Scaff31 pour utiliser le DNS de mon AD :
> Pour ta finalité il te suffit de configurer ton windows server avec comme DNS principal 127.0.0.1 et en secondaire l'ip de l'omr afin qu'il puisse forward vers l'omr-bypass
> 
> Sur ton DHCP sygnology il te suffit de config les options de plage, en gateway tu met l'ip OMR et en DNS tu met l'ip de ton windows server

Mais j'ai de nouveaux problèmes :
 - J'ai donc changé mes adresses réseaux, 172.16.0.0/24 (4G), 172.16.1.0/24 (ADSL), et 192.168.0.0/16 pour le LAN.
 - J'ai configuré le serveur AD comme ça :
[image]

 - J'ai ensuite modifié les paramètres du DHCP du NAS :
[image]

Voici la configuration du OMR, dans la section DHCP & DNS :
[image]

J'ai voulu désactiver le DHCP de OMR, mais il semble qu'il y ait toujours des clients qui ouvrent des bails (je pense ne pas avoir compris comment désactiver le DHCP de OMR) :
[image]

Quand un client est connecté sur le LAN, il semble récupérer une adresse sur le DHCP du Synology.

Le problème est surtout pour les clients en wifi, se connectant directement sur le réseau wifi de la box 4G, et qui mettent soit plusieurs minutes à récupérer une adresse IP, soit n'en récupère pas du tout...

Quand je récupère une IP sur mon PC via le DHCP du Synology, il m'est impossible d’accéder aux partages de mon NAS qui sont configurés avec le nom de domaine du serveur AD (nas.proxfr.local)
Alors que si je renseigne l'adresse du Synology ou du AD dans un navigateur, j’accède directement à leur interface web.

je laisse quand même la configuration IP que j'ai du DHCP : 
[image]
 
Merci pour votre aide 👍

Scaff31, 2019-10-08:
Donc si je résume tes problèmes a l'heure actuelle.
- Pas de DHCP sur le wifi du routeur 4g
- Pas de partage des ressources du domaine.

Pour le DHCP sur le wifi, vérifie que ton routeur est configuré en relais DHCP et non pas en serveur.
Pour les ressources du domaine (partage), a tu initialement ouvert une connexion avec tes credentials de domaine pour pouvoir les utiliser?

De plus sur ta capture de OMR et de ton AD quelque chose me chagrinne, ton ad forward les requêtes DNS a OMR qui lui même les forward à ton AD.

ProxFR, 2019-10-08:
Le routeur 4G est un MR1100, sur la page de configuration je peux uniquement activer/désactiver le serveur DHCP, il est donc désactivé.
Maintenant que OMR fonctionne sous une VM, je me demande si la solution ne serait pas de d’utiliser le raspberry pi avec un openWRT pour en faire un relais DHCP, et donc régler ce problème. 

Concernant les requêtes DNS, la solution a suivre serait de supprimer les forwards de OMR, et de laisser celles de l'AD ?

Ce qui est de plus étonnant, c'est que ce problème de DNS est "différent" suivant les machines :
Sur ma machine principal après avoir récupéré une adresse sur le DHCP, j'arrive a accéder a internet, et au NAS. Mais sur une autre machine qui récupère une adresse sur le même serveur, je n'ai aucun accès a internet ou au NAS...

Scaff31, 2019-10-10:
Pour le DNS le principe et celui-ci

Ton lan interroge ton AD pour faire une résolution si celui-ci à la réponse il répond sinon il demande à son forwarder ( OMR ) si OMR à la réponse il renvoi à ton AD puis à ton LAN sinon il demande à son forwarder (DNS Externe: cloudflare, google, FAI, etc) qui eux même demande aux root servers puis aux registrar

Pour le DHCP sur les machine qui n'ont pas internet n'oublie pas de faire un ipconfig /flushdns puis ipconfig /release et ipconfig /renew histoire de repartir sur de bonne base

ProxFR, 2019-10-11:
Alors voici un plan du réseau :
[image]

Cette semaine j'ai décidé de tout arrêter le temps de trouver une solution, j'ai donc tout re-basculer sur la box 4G comme précédemment. 
J'ai essayé tout ce que je savais faire entre lundi et mardi, mais impossible de faire fonctionner ce DNS...

J'ai donc configurer mon AD comme ci :
[image]

Donc, d’après ce que j'avais compris, les requêtes DNS étaient toutes envoyées au DNS de l'AD, si elles n’étaient pas utilisées par celui-ci, elles étaient donc forward vers le 192.168.0.1 (OMR), qui doit les renvoyer vers le DNS de google (par exemple).

Donc une fois l'adresse récupérée par les clients via le DHCP, j'ai donc tous les redirections vers les équipements locaux qui fonctionnent.
SAUF que je parviens à accéder a l'interface de ESXI et de OMR 1 fois sur 20 si ce n'est pas moins...
Impossible de ping aucune des 2 interfaces, impossible de savoir pourquoi...

Il y a surement un conflit quelque part, mais je n'ai jamais réussi a voir d'où il vient... 

Mais par moment je parviens a accéder a l'interface web de OMR, mais toujours impossible d’accéder a internet... 

Comme je change l'adresse IP de OMR, pour passer de 192.168.100.1/24 à 192.168.1.0/16, je pense que son DNS ne suit pas, comme si il continuait d'écouter sur le 192.168.100.1.

Quand j'ai voulu changer l'adresse IP de OMR, j'ai voulu passer par son interface web, je modifie directement LAN1 dans interface, quand je valide j'ai le décompte de 30s qui est interminable. Je peux uniquement rafraîchir la page, et je me retrouve connecté sur l'adresse par défaut.
Je passe donc par le CLI, et modifie le fichier /etc/config/network et je reboot par le suite.
Dans ce cas la l'adresse est bien prise en compte.

Mais donc une fois intégré au réseau déjà existant, j'ai l’impression que c'est le début de tous les problèmes.

En tout cas, merci beaucoup @Scaff31 de m'aider ! 👍

Ysurac, 2019-10-11:
Pour le changement d'IP, il y a effectivement un décompte de 30s, qui dure... 30s. Comme le site ne répond bien entendu plus à l'adresse initiale il revient à la configuration précédente et affiche une boite de dialogue proposant d'appliquer quand même les changements sans vérifications.

Scaff31, 2019-10-11:
Alors je n'ai pas d'outil de diagrammes sous la main je vais tenté de te faire un schéma d'archi en ASCII qui me semble plus simple. Je te détail la configuration des équipements en dessous.

WAN1 (adsl)  --------                                                      --------------------- NAS
                                 |                                                     |
                             Sw_wan ----------- ESXI ---------- Sw_LAN 
                                 |                                                     |
WAN2 (4G) ----------                                                      |---------------------- PC
                                                                                       |---------------------- Imprimante
                                                                                       |---------------------- Borne Wifi

Wan1 : 
     - ip = 192.168.10.254/24
Wan2 : 
     - ip = 192.168.20.254/24

ESXI :
     - 2 vSwitch
     - 2 groupe de port

vSwitch0: 
    - nom: inter_wan
    - option de sécurité: accepte tout
    - connecté à VMNIC0

vSwitch1:
    - nom: vSwitch_Lan
    - option de sécurité: par default
    - connecté à VMNIC1

Groupes de ports1:
    - Assigné au vSwitch0
    - nom: WAN
    - Connecté à OMR
    - Connecté à Pfsense

Groupes de ports2:
    - Assigné au vSwitch_lan
    - nom: LAN
    - Connecté à Pfsense
    - Connecté à ton AD
    - Connecté à ton vcenter

ProxFR, 2019-10-13:
@Ysurac C'est ce que j'avais compris mais même durant la période des 30s, si je fais un ifconfig je ne voit pas la nouvelle adresse IP appliquée pendant ce décompte.

@Scaff31 Merci pour ce schéma qui m'intéresse beaucoup. Donc pour toi la meilleur solution serait de dédier un switch pour les WAN, qui serait directement connecté à ESXI ? Pour pouvoir séparer les réseaux entre le LAN et le WAN pour par qu'il n'y est de "conflit" ? 

Il faut que je me penche la dessus 😉 Mais il faut surtout que je trouve un créneau pour pouvoir bloquer internet... parce qu'entre les cours et le reste... c'est pas facile !

Scaff31, 2019-10-14:
Yes jsui partisan de reprendre tout au plus simple et de complexifier au fur et à mesure

closed 2019-12-20 by Ysurac