Ysurac/openmptcprouter #684: The opkg update command failed with code 5.

xabix, 2019-10-03:
## Expected Behavior

When I ask to update the packages (opkg), I get the follow error: The opkg update command failed with code 5.

 `Collected errors:
 * opkg_download: Failed to download http://downloads.openwrt.org/snapshots/packages/x86_64/luci/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.
 * opkg_download: Failed to download http://downloads.openwrt.org/snapshots/packages/x86_64/packages/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.
 * opkg_download: Failed to download http://downloads.openwrt.org/snapshots/packages/x86_64/base/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.
 * opkg_download: Failed to download http://downloads.openwrt.org/snapshots/packages/x86_64/routing/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.
 * opkg_download: Failed to download http://downloads.openwrt.org/snapshots/packages/x86_64/telephony/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.`

My question is to understand if it s an issue on my side or if the repositories are bad.

I tried checking from OMR if I could manually get this file and I get:
`root@OpenMPTCProuter:~# wget http://downloads.openwrt.org/snapshots/packages/x86_64/luci/Packages.gz
Downloading 'http://downloads.openwrt.org/snapshots/packages/x86_64/luci/Packages.gz'
Connecting to 2a01:4f8:150:6449::2:80
Connection error: Connection failed
root@OpenMPTCProuter:~# nslookup downloads.openwrt.org
Server:         127.0.0.1
Address:        127.0.0.1#53

Name:      downloads.openwrt.org
downloads.openwrt.org   canonical name = mirror-02.infra.openwrt.org
Name:      mirror-02.infra.openwrt.org
Address 1: 176.9.48.73
downloads.openwrt.org   canonical name = mirror-02.infra.openwrt.org
Address 2: 2a01:4f8:150:6449::2`

## Actual Behavior

Is it normal or a bad config on my side or an issue with the configuration of the default repositories?

## Specifications

  - OpenMPTCProuter version: 0.52.3
  - OpenMPTCProuter VPS version: 1001
  - OpenMPTCProuter platform: x86_64

Ysurac, 2019-10-03:
Do you have IPv6 enabled or not ?

xabix, 2019-10-03:
In which part? My network is not IPv6, my interfaces have 'IPv6 assignment length' desactivated

Ysurac, 2019-10-03:
on openmptcprouter

xabix, 2019-10-03:
Sorry not sure which section of OMR allow to say if IPv6 is enable? within MTCP? 

[image]

I think it s not enabled by this one:
[image]

Ysurac, 2019-10-03:
Yes this is disabled. It shouldn't try to download via IPv6...
It's an old imported backup ?

xabix, 2019-10-03:
no it s fresh install because I wanted to start from a clean to have AES enabled by default (which didn't work) cf: https://github.com/Ysurac/openmptcprouter/issues/669

xabix, 2019-10-03:
What do you think is the issue? 

```
root@OpenMPTCProuter:~# ifconfig 
6in4-omr6in4 Link encap:IPv6-in-IPv4  
          inet6 addr: fe80::a00:2/128 Scope:Link
          inet6 addr: fe80::aff:fb02/64 Scope:Link
          UP POINTOPOINT RUNNING NOARP  MTU:1280  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth0      Link encap:Ethernet  HWaddr 3E:87:6E:02:F4:D2  
          inet addr:192.168.1.254  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fd77:d00e:45dd::1/60 Scope:Global
          inet6 addr: fe80::3c87:6eff:fe02:f4d2/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:2975491 errors:0 dropped:2252 overruns:0 frame:0
          TX packets:1545023 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:2160963653 (2.0 GiB)  TX bytes:2015400360 (1.8 GiB)

ifb4wan1  Link encap:Ethernet  HWaddr 6A:58:AF:8F:78:78  
          inet6 addr: fe80::6858:afff:fe8f:7878/64 Scope:Link
          UP BROADCAST RUNNING NOARP  MTU:1500  Metric:1
          RX packets:485873 errors:0 dropped:0 overruns:0 frame:0
          TX packets:485873 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:32 
          RX bytes:340632075 (324.8 MiB)  TX bytes:340632075 (324.8 MiB)

ifb4wan2  Link encap:Ethernet  HWaddr 9A:D1:9A:30:A7:BA  
          inet6 addr: fe80::98d1:9aff:fe30:a7ba/64 Scope:Link
          UP BROADCAST RUNNING NOARP  MTU:1500  Metric:1
          RX packets:270243 errors:0 dropped:0 overruns:0 frame:0
          TX packets:270243 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:32 
          RX bytes:191900627 (183.0 MiB)  TX bytes:191900627 (183.0 MiB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:67179 errors:0 dropped:0 overruns:0 frame:0
          TX packets:67179 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:5261760 (5.0 MiB)  TX bytes:5261760 (5.0 MiB)

tun0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          inet addr:10.255.255.2  P-t-P:10.255.255.1  Mask:255.255.255.255
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1500  Metric:1
          RX packets:126209 errors:0 dropped:0 overruns:0 frame:0
          TX packets:111328 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:92159580 (87.8 MiB)  TX bytes:27007204 (25.7 MiB)

wan1      Link encap:Ethernet  HWaddr 72:95:ED:EF:2A:6D  
          inet addr:192.168.2.2  Bcast:192.168.2.255  Mask:255.255.255.0
          inet6 addr: fe80::7095:edff:feef:2a6d/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:490656 errors:0 dropped:1156 overruns:0 frame:0
          TX packets:431828 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:100 
          RX bytes:349834571 (333.6 MiB)  TX bytes:59576633 (56.8 MiB)

wan2      Link encap:Ethernet  HWaddr 16:2C:B4:F7:4E:B8  
          inet addr:192.168.3.2  Bcast:192.168.3.255  Mask:255.255.255.0
          inet6 addr: fe80::142c:b4ff:fef7:4eb8/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:272570 errors:0 dropped:1127 overruns:0 frame:0
          TX packets:190217 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:100 
          RX bytes:194602036 (185.5 MiB)  TX bytes:19912167 (18.9 MiB)

wan3      Link encap:Ethernet  HWaddr 56:F0:8A:41:15:3F  
          inet addr:192.168.9.2  Bcast:192.168.9.255  Mask:255.255.255.0
          inet6 addr: fe80::54f0:8aff:fe41:153f/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1297234 errors:0 dropped:1098 overruns:0 frame:0
          TX packets:416510 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:100 
          RX bytes:1475200247 (1.3 GiB)  TX bytes:96384334 (91.9 MiB)
```

Ysurac, 2019-10-03:
I would need the config, from system->OpenMPTCProuter and "show all settings" tab, without keys or vps ip.

xabix, 2019-10-04:
Would be nice to have a button to export the settings but hiding the keys and vps ip :) ?

Here is my config:

> dhcp.@dnsmasq[0]=dnsmasq
> dhcp.@dnsmasq[0].domainneeded='1'
> dhcp.@dnsmasq[0].boguspriv='1'
> dhcp.@dnsmasq[0].filterwin2k='0'
> dhcp.@dnsmasq[0].localise_queries='1'
> dhcp.@dnsmasq[0].rebind_protection='1'
> dhcp.@dnsmasq[0].rebind_localhost='1'
> dhcp.@dnsmasq[0].local='/lan/'
> dhcp.@dnsmasq[0].domain='lan'
> dhcp.@dnsmasq[0].expandhosts='1'
> dhcp.@dnsmasq[0].authoritative='1'
> dhcp.@dnsmasq[0].readethers='1'
> dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
> dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.auto'
> dhcp.@dnsmasq[0].nonwildcard='1'
> dhcp.@dnsmasq[0].localservice='1'
> dhcp.@dnsmasq[0].server='127.0.0.1#5353'
> dhcp.@dnsmasq[0].noresolv='1'
> dhcp.@dnsmasq[0].nonegcache='1'
> dhcp.@dnsmasq[0].rebind_domain='plex.direct'
> dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1'
> dhcp.lan=dhcp
> dhcp.lan.interface='lan'
> dhcp.lan.ra_slaac='1'
> dhcp.lan.ra_flags='managed-config' 'other-config'
> dhcp.lan.ignore='1'
> dhcp.wan=dhcp
> dhcp.wan.interface='wan'
> dhcp.wan.ignore='1'
> dhcp.odhcpd=odhcpd
> dhcp.odhcpd.maindhcp='0'
> dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
> dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
> dhcp.odhcpd.loglevel='4'
> dropbear.@dropbear[0]=dropbear
> dropbear.@dropbear[0].PasswordAuth='on'
> dropbear.@dropbear[0].RootPasswordAuth='on'
> dropbear.@dropbear[0].Port='22'
> dscp.@classify[0]=classify
> dscp.@classify[0].direction='both'
> dscp.@classify[0].proto='icmp'
> dscp.@classify[0].class='cs5'
> dscp.@classify[0].comment='ICMP'
> dscp.@classify[1]=classify
> dscp.@classify[1].direction='both'
> dscp.@classify[1].proto='udp'
> dscp.@classify[1].class='cs5'
> dscp.@classify[1].src_port='53,123,5353'
> dscp.@classify[1].comment='DNS udp and NTP'
> dscp.@classify[2]=classify
> dscp.@classify[2].direction='both'
> dscp.@classify[2].proto='tcp'
> dscp.@classify[2].class='cs5'
> dscp.@classify[2].src_port='53,5353'
> dscp.@classify[2].comment='DNS tcp'
> dscp.@classify[3]=classify
> dscp.@classify[3].direction='both'
> dscp.@classify[3].proto='tcp'
> dscp.@classify[3].class='cs6'
> dscp.@classify[3].dest_port='65001,65301,65011'
> dscp.@classify[3].comment='OMR vpn'
> dscp.@classify[4]=classify
> dscp.@classify[4].direction='both'
> dscp.@classify[4].proto='udp'
> dscp.@classify[4].class='cs6'
> dscp.@classify[4].dest_port='65001'
[... 1139 lines cut ...]
> upnpd.config.enabled='0'
> upnpd.config.enable_natpmp='1'
> upnpd.config.enable_upnp='1'
> upnpd.config.secure_mode='1'
> upnpd.config.log_output='0'
> upnpd.config.download='1024'
> upnpd.config.upload='512'
> upnpd.config.internal_iface='lan'
> upnpd.config.port='5000'
> upnpd.config.upnp_lease_file='/var/run/miniupnpd.leases'
> upnpd.config.igdv1='0'
> upnpd.@perm_rule[0]=perm_rule
> upnpd.@perm_rule[0].action='allow'
> upnpd.@perm_rule[0].ext_ports='1024-65535'
> upnpd.@perm_rule[0].int_addr='0.0.0.0/0'
> upnpd.@perm_rule[0].int_ports='1024-65535'
> upnpd.@perm_rule[0].comment='Allow high ports'
> upnpd.@perm_rule[1]=perm_rule
> upnpd.@perm_rule[1].action='deny'
> upnpd.@perm_rule[1].ext_ports='0-65535'
> upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
> upnpd.@perm_rule[1].int_ports='0-65535'
> upnpd.@perm_rule[1].comment='Default deny'
> vnstat.@vnstat[0]=vnstat
> vnstat.@vnstat[0].interface='wan1' 'wan2' 'wan3'

Ysurac, 2019-10-04:
I see no problems here.
Can you give me the result of ```ip -6 a``` and ```ip -6 route``` via SSH on the router ?

Marctraider, 2019-10-04:
I have seen this happen at times as well, no idea what causing it.

xabix, 2019-10-06:
> ip -6 route

Sure here it is:

> ip -6 a
> 1: lo: <LOOPBACK,UP,LOWER_UP,80000> mtu 65536 state UNKNOWN qlen 1000
>     inet6 ::1/128 scope host 
>        valid_lft forever preferred_lft forever
> 2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP,80000> mtu 1500 state UP qlen 1000
>     inet6 fd77:d00e:45dd::1/60 scope global noprefixroute 
>        valid_lft forever preferred_lft forever
>     inet6 fe80::3c87:6eff:fe02:f4d2/64 scope link 
>        valid_lft forever preferred_lft forever
> 7: wan1@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 state UP qlen 100
>     inet6 fe80::7095:edff:feef:2a6d/64 scope link 
>        valid_lft forever preferred_lft forever
> 8: wan2@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 state UP qlen 100
>     inet6 fe80::142c:b4ff:fef7:4eb8/64 scope link 
>        valid_lft forever preferred_lft forever
> 9: wan3@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 state UP qlen 100
>     inet6 fe80::54f0:8aff:fe41:153f/64 scope link 
>        valid_lft forever preferred_lft forever
> 10: 6in4-omr6in4@NONE: <POINTOPOINT,NOARP,UP,LOWER_UP,80000> mtu 1280 state UNKNOWN qlen 1000
>     inet6 fe80::a00:2/128 scope link 
>        valid_lft forever preferred_lft forever
>     inet6 fe80::aff:fb02/64 scope link 
>        valid_lft forever preferred_lft forever
> 23: ifb4wan1: <BROADCAST,NOARP,UP,LOWER_UP,80000> mtu 1500 state UNKNOWN qlen 32
>     inet6 fe80::dc52:96ff:fe5b:89da/64 scope link 
>        valid_lft forever preferred_lft forever
> 27: ifb4wan2: <BROADCAST,NOARP,UP,LOWER_UP,80000> mtu 1500 state UNKNOWN qlen 32
>     inet6 fe80::546c:53ff:fe14:a252/64 scope link 
>        valid_lft forever preferred_lft forever
 
>  ip -6 route
> default from fe80::a00:2 dev 6in4-omr6in4 proto static metric 7 pref medium
> fd77:d00e:45dd::/64 dev eth0 proto static metric 2 pref medium
> unreachable fd77:d00e:45dd::/48 dev lo proto static metric 2147483647 pref medium
> fe80::a00:2 dev 6in4-omr6in4 proto static metric 7 pref medium
> fe80::/64 dev eth0 proto kernel metric 256 pref medium
> fe80::/64 dev wan1 proto kernel metric 256 pref medium
> fe80::/64 dev wan2 proto kernel metric 256 pref medium
> fe80::/64 dev wan3 proto kernel metric 256 pref medium
> fe80::/64 dev 6in4-omr6in4 proto kernel metric 256 pref medium
> fe80::/64 dev ifb4wan1 proto kernel metric 256 pref medium
> fe80::/64 dev ifb4wan2 proto kernel metric 256 pref medium
> default via fe80::a00:1 dev 6in4-omr6in4 proto static metric 7 pref medium

```

Ysurac, 2019-10-07:
Can you try to disable "omr6in4" interface, in "network"->"interfaces" menu,  and try again to update package list ?

xabix, 2019-10-10:
If I disable this interface, yes it does work. Do I need it or do you know what is the issue?

BTW are we suppose to be able to upgrade the packages directly through opkg or you don't recommend? Just wondering if it s a good thing to do or a waste of time and or a risky option. In this case maybe worth removing all of this menu to avoid people to upgrade packages by themselves..

Merci

[image]

Ysurac, 2019-10-10:
You can install new applications, but you shouldn't update this way.

Ysurac, 2019-10-10:
Because this give IPv6, and this seems to use it. So it's easy: in next release omr6in4 will be down when not needed

xabix, 2019-10-11:
Ok so I don't need it and I can disable it. If I reboot it will stay disabled right?

Ysurac, 2019-10-11:
yes

closed 2019-10-11 by Ysurac

Candyffm, 2022-02-08:
This still happens on v0.58.5 r0+16336-b36068d35d
My Solution: Services >> Terminal >> login >> opkg install wget-ssl  --no-check-certificate

cagnulein, 2022-02-20:
Confirm that it's still happening

Brain2000, 2022-12-27:
> I have seen this happen at times as well, no idea what causing it.

This is happening to me right now. It's quite amazing how OpenWRT is so advanced, yet so primitive at times.

macminoz, 2023-12-28:
it's a DNS problem. I manually added the downloads.openwrt.org to the /etc/hosts and with this manually method works