Ysurac/openmptcprouter #685: can't get public IP address from shadowsocks

Yanaem, 2019-10-05:
## Expected Behavior

No having this error.

## Actual Behavior

1) Is that a real problem? how to fix it ?

2) How can i use my Dyndns ? (is there somewhere to configure login/password) ?

## Specifications

  - OpenMPTCProuter version: (v0.52)
  - OpenMPTCProuter VPS version: (0.1001)
  - OpenMPTCProuter platform: RPI364)

Ysurac, 2019-10-06:
1. Yes it's a problem. You should reboot the VPS, you are on a 4.14 kernel release...
2. You can install " luci-app-ddns" package, but with the VPS you should have a fixed IP. So why ?

Yanaem, 2019-10-06:
1. Ok i reboot it , but the problem is still there now with two message. Maybe a problem with my keys?

[image]
[image]

Ysurac, 2019-10-06:
Yes, it can't contact admin script. "omr-admin" should be running on VPS side and key need to be correct.

Yanaem, 2019-10-06:
Ok i checked the key , and all is good. 

I also checked the  /var/log/daemon.log. Maybe the error is there , take a look :
[image]

Ysurac, 2019-10-06:
Yes it can't authenticate. You use another install of shadowsocks ? You should reinstall VPS.

closed 2019-12-20 by Ysurac