Ysurac/openmptcprouter #689: Aggregation speed less

abhishek1390, 2019-10-09:
## Expected Behavior
Aggregation speed should be sum of two ISP.

## Actual Behavior
We have connected two Huawei dongles. One is giving the speed around 5 to 6 Mbps, other is giving around 3 to 4 Mbps. But aggregated speed is not as expected, And in status page it is showing all green.And we are using Amazon Web server (AWS) as VPS. For getting more speed we have configured shadow socks encryption to 'none'.

[image]

For reference attached the config file.
[config.txt](https://github.com/Ysurac/openmptcprouter/files/3707761/config.txt)

## Specifications

  - OpenMPTCProuter version: 0.52.3
  - OpenMPTCProuter VPS version: 0.999
  - OpenMPTCProuter platform: x86_64

Ysurac, 2019-10-09:
What is the aggregated speed ?
What is the result of ```omr-test-speed wan1```, ```omr-test-speed wan2``` (replace wan1 and wan2 by real device name, check that using ```ip a```) and the aggregated speed ```omr-test-speed``` ?

abhishek1390, 2019-10-10:
Here it is...

[image]

[image]

Ysurac, 2019-10-10:
Not really good...
You can try to disable fast-open.
You can also try another TCP congestion control in the Network->MPTCP interface.

abhishek1390, 2019-10-10:
"fast-open" is enabled on all shadowsocks instance,
Should i disable all?

Ysurac, 2019-10-10:
Yes, or use checkbox in System->OpenMPTCProuter and Advanced settings tab.

abhishek1390, 2019-10-10:
ok will try and get back to you. Thanks for quick response.

abhishek1390, 2019-10-11:
I am getting this error continuously while testing the speed. What does this mean? 
Sometimes this error comes on both interface and VPN tunnel will also be down. 

[image]

And i have changed the VPS to OVH.

Ysurac, 2019-10-11:
It means that server doesn't answer to ping in less than 1 second.
I think there a big latency between the server and you.

abhishek1390, 2019-10-11:
Is this reason VPN tunnel becoming down?
Can i increase that 1 second?

Ysurac, 2019-10-11:
The 1 seconds is only for status page, it's not a real problem.
If you really want to change it, via SSH on the router:
```
uci set openmptcprouter.settings.status_vps_timeout='10'
uci set openmptcprouter.settings.status_getip_timeout='10'
uci commit openmptcprouter
```

But you should check system log, to see what is the real problem.

abhishek1390, 2019-10-11:
As i understand omrvpn is getting down..What can be the reason?
Attached the logfile
[log.txt](https://github.com/Ysurac/openmptcprouter/files/3716048/log.txt)

Can you see what could be problem? And after disabling `fast-open` in shadowsocks instances i am not seeing any significant difference in speed.

closed 2019-12-20 by Ysurac