Ysurac/openmptcprouter #690: OMR-Bypass for airlines compagny

Scaff31, 2019-10-10:
## Expected Behavior

Acces website for airlines compagny (easyjet, airfrance, etc...)

## Actual Behavior

Access denied

## Steps to Reproduce the Problem

  1. try to navigate on www.airfrance.fr
  2. on omr-bypass add airfrance.fr / airfrance.com / klm.com 
  3. try to reserve on website

## Specifications

  - OpenMPTCProuter version: 0.52.3
  - OpenMPTCProuter VPS version: 0.1001
  - OpenMPTCProuter platform: RPI3 + ESXI

Ysurac, 2019-10-10:
You have problem on access or when you want to pay or at another step ?
You need to find what domain block your VPS IP using the developer mode of your browser.

Scaff31, 2019-10-10:
Problem for accessing, without OMR-Bypass reservation form not load and with omr-bypass loading can be use but when you want click on search you get and denied access

Ysurac, 2019-10-10:
What is the VPS provider ?
Do you have an example of public URL that fail ?

Scaff31, 2019-10-10:
ovh

wwws.airfrance.fr

Scaff31, 2019-10-10:
Ok i think i've found solution with multiple domains ans subdomains trying 
[image]

Ysurac, 2019-10-10:
subdomains are bypassed when domain is set. But you may have also static-kl.com and some others.
Fun: IPs from firstheberg and OVH are blocked but not scaleway and Vultr.

Scaff31, 2019-10-10:
Great info for IP pool

closed 2019-12-20 by Ysurac