Ysurac/openmptcprouter #691: VPN tunnel DOWN

kyouran38, 2019-10-10:
Hi,

I ve got a problem on my state tab my router is indicate a VPN tunnel DOWN status ...

I have also these logs : 

Thu Oct 10 20:58:42 2019 kern.info kernel: [55108.170959] IPv6: ADDRCONF(NETDEV_UP): tun0: link is not ready
Thu Oct 10 20:58:42 2019 kern.info kernel: [55108.334712] IPv6: ADDRCONF(NETDEV_UP): tun0: link is not ready
Thu Oct 10 20:58:43 2019 daemon.err glorytun[27703]: 217.xx.x.xxx.65001: connected
Thu Oct 10 20:58:43 2019 daemon.info glorytun[27703]: STARTED tun0
Thu Oct 10 20:58:43 2019 daemon.err glorytun[27703]: tun write: I/O error
Thu Oct 10 20:58:43 2019 daemon.info glorytun[27703]: STOPPED tun0
Thu Oct 10 20:59:13 2019 daemon.info glorytun[28741]: INITIALIZED tun0
Thu Oct 10 20:59:13 2019 kern.info kernel: [55139.423674] IPv6: ADDRCONF(NETDEV_UP): tun0: link is not ready
Thu Oct 10 20:59:13 2019 kern.info kernel: [55139.625062] IPv6: ADDRCONF(NETDEV_UP): tun0: link is not ready
Thu Oct 10 20:59:14 2019 daemon.err glorytun[28741]: 217.xx.x.xxx.65001: connected
Thu Oct 10 20:59:14 2019 daemon.info glorytun[28741]: STARTED tun0
Thu Oct 10 20:59:14 2019 daemon.err glorytun[28741]: tun write: I/O error
Thu Oct 10 20:59:14 2019 daemon.info glorytun[28741]: STOPPED tun0
Thu Oct 10 20:59:44 2019 daemon.info glorytun[29789]: INITIALIZED tun0
Thu Oct 10 20:59:44 2019 kern.info kernel: [55170.644953] IPv6: ADDRCONF(NETDEV_UP): tun0: link is not ready
Thu Oct 10 20:59:44 2019 kern.info kernel: [55170.833980] IPv6: ADDRCONF(NETDEV_UP): tun0: link is not ready
Thu Oct 10 20:59:45 2019 daemon.err glorytun[29789]: 217.xx.x.xxx.65001: connected
Thu Oct 10 20:59:45 2019 daemon.info glorytun[29789]: STARTED tun0
Thu Oct 10 20:59:45 2019 daemon.err glorytun[29789]: tun write: I/O error
Thu Oct 10 20:59:45 2019 daemon.info glorytun[29789]: STOPPED tun0
Thu Oct 10 21:00:15 2019 daemon.info glorytun[30855]: INITIALIZED tun0
Thu Oct 10 21:00:15 2019 kern.info kernel: [55201.853877] IPv6: ADDRCONF(NETDEV_UP): tun0: link is not ready
Thu Oct 10 21:00:16 2019 kern.info kernel: [55202.034086] IPv6: ADDRCONF(NETDEV_UP): tun0: link is not ready
Thu Oct 10 21:00:17 2019 daemon.err glorytun[30855]: 217.xx.x.xxx.65001: connected
Thu Oct 10 21:00:17 2019 daemon.info glorytun[30855]: STARTED tun0
Thu Oct 10 21:00:17 2019 daemon.err glorytun[30855]: tun write: I/O error
Thu Oct 10 21:00:17 2019 daemon.info glorytun[30855]: STOPPED tun0
Thu Oct 10 21:00:47 2019 daemon.info glorytun[31926]: INITIALIZED tun0
Thu Oct 10 21:00:47 2019 kern.info kernel: [55233.041828] IPv6: ADDRCONF(NETDEV_UP): tun0: link is not ready
Thu Oct 10 21:00:47 2019 kern.info kernel: [55233.215048] IPv6: ADDRCONF(NETDEV_UP): tun0: link is not ready
Thu Oct 10 21:00:48 2019 daemon.err glorytun[31926]: 217.xx.x.xxx.65001: connected
Thu Oct 10 21:00:48 2019 daemon.info glorytun[31926]: STARTED tun0
Thu Oct 10 21:00:48 2019 daemon.err glorytun[31926]: tun write: I/O error
Thu Oct 10 21:00:48 2019 daemon.info glorytun[31926]: STOPPED tun0
Thu Oct 10 21:01:18 2019 daemon.info glorytun[339]: INITIALIZED tun0
Thu Oct 10 21:01:18 2019 kern.info kernel: [55264.203167] IPv6: ADDRCONF(NETDEV_UP): tun0: link is not ready
Thu Oct 10 21:01:18 2019 kern.info kernel: [55264.375258] IPv6: ADDRCONF(NETDEV_UP): tun0: link is not ready
Thu Oct 10 21:01:19 2019 daemon.err glorytun[339]: 217.xx.x.xxx.65001: connected
Thu Oct 10 21:01:19 2019 daemon.info glorytun[339]: STARTED tun0
Thu Oct 10 21:17:11 2019 daemon.info glorytun[339]: STOPPED tun0
Thu Oct 10 21:17:12 2019 daemon.err glorytun[339]: 217.xx.x.xxx.65001: connected
Thu Oct 10 21:17:12 2019 daemon.info glorytun[339]: STARTED tun0
Thu Oct 10 21:17:12 2019 daemon.err glorytun[339]: tun write: I/O error
Thu Oct 10 21:17:12 2019 daemon.info glorytun[339]: STOPPED tun0
Thu Oct 10 21:17:42 2019 daemon.info glorytun[27723]: INITIALIZED tun0
Thu Oct 10 21:17:42 2019 kern.info kernel: [56248.301081] IPv6: ADDRCONF(NETDEV_UP): tun0: link is not ready
Thu Oct 10 21:17:42 2019 kern.info kernel: [56248.468679] IPv6: ADDRCONF(NETDEV_UP): tun0: link is not ready
Thu Oct 10 21:17:43 2019 daemon.err glorytun[27723]: 217.xx.x.xxx.65001: connected
Thu Oct 10 21:17:43 2019 daemon.info glorytun[27723]: STARTED tun0
Thu Oct 10 21:20:36 2019 user.notice post-tracking-post-tracking: Get status and settings for wan2...

Is there anyone have an idea ?

## Specifications

  - OpenMPTCProuter version: 0.52.3
  - OpenMPTCProuter VPS version: 0.1001 4.19.67
  - OpenMPTCProuter platform: Raspberry Pi 3 Model B Plus Rev 1.3

Thanks :)

Ysurac, 2019-10-11:
The "tun write: I/O error" is strange.
You don't have an other VPN enabled ?

kyouran38, 2019-10-11:
Hi nop, i don't have any other vpn running.

 

REgards

 

Sent: Friday, October 11, 2019 at 7:14 AM

Ysurac, 2019-10-11:
You tried to reboot the router and checked the status page ?

kyouran38, 2019-10-11:
i did change nothing :(

[Pierrick Lebrun - Chat @ Spike](https://www.spikenow.com/?ref=spike-organic-signature&_ts=76y84)	[76y84]

kyouran38, 2019-10-12:
I did nothing change :(

Ysurac, 2019-10-12:
All is green in status page ?
What is your config ?

kyouran38, 2019-10-13:
All is green, i reinstall the router and change vps, pur glorytun on udp mode, but still the same muisc :(

 

Sun Oct 13 11:20:20 2019 daemon.info glorytun[3484]: STOPPED tun0
Sun Oct 13 11:20:21 2019 daemon.err glorytun[3484]: 51.15.XX.XX.65001: connected
Sun Oct 13 11:20:21 2019 daemon.info glorytun[3484]: STARTED tun0
Sun Oct 13 11:20:41 2019 daemon.err glorytun[3484]: read: Connection reset by peer
Sun Oct 13 11:20:41 2019 daemon.info glorytun[3484]: STOPPED tun0
Sun Oct 13 11:20:42 2019 daemon.err glorytun[3484]: 51.15.XXX.XXX.65001: connected
Sun Oct 13 11:20:42 2019 daemon.info glorytun[3484]: STARTED tun0
Sun Oct 13 11:21:02 2019 daemon.info glorytun[3484]: STOPPED tun0
Sun Oct 13 11:21:03 2019 daemon.err glorytun[3484]: 51.15.XXX.XXX.65001: connected
Sun Oct 13 11:21:03 2019 daemon.info glorytun[3484]: STARTED tun0
 

 

Sent: Saturday, October 12, 2019 at 5:14 PM

Ysurac, 2019-10-13:
So why do you use glorytun UDP ?

kyouran38, 2019-10-13:
I read somewhere it can solve problem, but it change nothing
 

Sent: Sunday, October 13, 2019 at 1:33 PM

kyouran38, 2019-10-13:
I read somewhere it help for some people :(

I put it back to tcp, it's weird, i never had this kind of problem before, it worked well

kyouran38, 2019-10-18:
Je reviens à la charge, si quelqu'un à une idée car je n'ai pas d'idée, est ce qu'on peut bien me confirmer la version que je dois télécharger pour mon RBPI :+1: 

OpenMPTCProuter platform: Raspberry Pi 3 Model B Plus Rev 1.3

Merci d'avance

Anthony010101, 2019-10-19:
Bonjour,

Nous avons le même problème.. Installation à jour et le VPN avec GloryTun.

kyouran38, 2019-10-20:
Still have this up down problem

Ysurac, 2019-12-20:
You can test with v0.53, if this still doesn't work you can reopen this issue.

closed 2019-12-20 by Ysurac

MistiqueX, 2023-12-24:
Hi there, dear Ysurac, have the same problem + another.
[image]

dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.d/resolv.conf.auto'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].ednspacket_max='1232'
dhcp.@dnsmasq[0].server='127.0.0.1#5353' '/lan/' '/use-application-dns.net/'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/zoom.us/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1' '/tv.milkywan.fr/omr_dscp-cs5,omr_dscp6-cs5'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.dhcpv4='server'
dhcp.lan.ra_slaac='1'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.lan.force='1'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dhcp.modem=dhcp
dhcp.modem.interface='modem'
dhcp.modem.ignore='1'
dhcp.modem.start='100'
dhcp.modem.limit='150'
dhcp.modem.leasetime='12h'
dhcp.modem.ra_flags='none'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].RootPasswordAuth='on'
dropbear.@dropbear[0].Port='22'
dscp.@classify[0]=classify
dscp.@classify[0].direction='both'
dscp.@classify[0].proto='icmp'
dscp.@classify[0].class='cs7'
dscp.@classify[0].comment='ICMP'
dscp.@classify[1]=classify
dscp.@classify[1].direction='both'
dscp.@classify[1].proto='udp'
dscp.@classify[1].class='cs4'
dscp.@classify[1].src_port='53,123,5353'
dscp.@classify[1].comment='DNS udp and NTP'
dscp.@classify[2]=classify
dscp.@classify[2].direction='both'
dscp.@classify[2].proto='tcp'
dscp.@classify[2].class='cs4'
dscp.@classify[2].src_port='53,5353'
dscp.@classify[2].comment='DNS tcp'
[... 1672 lines cut ...]
v2ray.omrrouting.inbound_tag='omrbridge'
v2ray.omrrouting.outbound_tag='omrout_tunnel'
v2ray.omrrouting.domain='full:omr.lan'
v2ray.omrroutingo=routing_rule
v2ray.omrroutingo.type='field'
v2ray.omrroutingo.inbound_tag='omrbridge'
v2ray.omrroutingo.outbound_tag='out'
v2ray.main_routing=routing
v2ray.main_routing.enabled='1'
v2ray.main_routing.rules='omrrouting' 'omrroutingo'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='wan1' 'wan2'
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.path='platform/soc/fe300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio0.channel='36'
wireless.radio0.band='5g'
wireless.radio0.htmode='VHT80'
wireless.radio0.disabled='1'
wireless.default_radio0=wifi-iface
wireless.default_radio0.device='radio0'
wireless.default_radio0.network='lan'
wireless.default_radio0.mode='ap'
wireless.default_radio0.ssid='OpenWrt'
wireless.default_radio0.encryption='none'

MistiqueX, 2023-12-24:
[image]
when restart.

Ysurac, 2023-12-24:
You should try snapshot release: https://github.com/Ysurac/openmptcprouter/wiki/Snapshots
Please, if you still have problem, open a new issue.