Ysurac/openmptcprouter #692: IP FailOver direct output

Anthony010101, 2019-10-13:
Hello,

Is it possible to put the IP FailOver in IP output?
I added it with the VPS, but that's not enough.

Thank you

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.52.3 r0+11006-713561a10b
  - OpenMPTCProuter VPS version: 0.52.3
  - OpenMPTCProuter platform: x86_64

Ysurac, 2019-10-13:
I think you speak about OVH IP Failover, so you should check OVH docs to set it as default output using routes.

Anthony010101, 2019-10-13:
Yes, at the exit I have the ip of the VPS. While I wish to have that of FailOver.

flapvhd, 2019-10-13:
Hi All,
For that purpose, I use Xen on the physical server, configured in router mode. OpenMPTCP VPS is running on a domU VM and is using its own IP Failover address...

Anthony010101, 2019-10-13:
In fact I meet a problem due to the additional OVH ip.
Ping and port redirects work on the additional IP. But not the IPSec VPNs, which they only work with the IP of the VPS.

closed 2019-12-20 by Ysurac