Ysurac/openmptcprouter #693: Reuse-after-free in kernel messages

ragefan, 2019-10-13:
## Expected Behavior
No error in Kernel Message

## Actual Behavior

Sporadically, on raspberry pi 3 b+ : use-after-free error

[63787.880963] ------------[ cut here ]------------
[63787.886942] refcount_t: underflow; use-after-free.
[63787.893030] WARNING: CPU: 1 PID: 0 at lib/refcount.c:187 refcount_sub_and_test_checked+0x58/0xb0
[63787.904299] Modules linked in: rtl8192cu rtl8192c_common rtl_usb rt2800usb rt2800lib qcserial option ipw brcmfmac usb_wwan usb_serial_simple ti_usb_3410_5052 sierra rtlwifi rtl8xxxu rtl8187 rt2x00usb rt2x00lib pppoe ppp_async pl2303 oti6858 nf_nat_pptp nf_conntrack_pptp mt7601u mos7720 mmc_spi mct_u232 mac80211 keyspan iptable_nat ipt_REJECT ipt_MASQUERADE garmin_gps ftdi_sio ebtable_nat ebtable_filter ebtable_broute cypress_m8 cp210x ch341 cfg80211 cdc_mbim belkin_sa bas_gigaset ark3116 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_owner xt_ndpi xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iface xt_hl xt_helper xt_hashlimit xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_condition xt_comment xt_addrtype xt_TRACE xt_TPROXY
[63787.989242]  xt_TCPMSS xt_REDIRECT xt_LOG xt_IPMARK xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY xt_ACCOUNT visor usbserial usbhid usb_gigaset ts_fsm ts_bm sr9700 sierra_net ser_gigaset rtl8150 rndis_host r8712u(C) r8152 qmi_wwan pppox ppp_generic pegasus of_mmc_spi nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_proto_gre nf_nat_irc nf_nat_ipv4 nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_proto_gre nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount mcs7830 macvlan kaweth kalmia iptable_raw iptable_mangle iptable_filter
[63788.077299]  ipt_ECN ipheth ip6table_raw ip_tables huawei_cdc_ncm hso hid_generic gigaset ezusb ethoc ebtables ebt_vlan ebt_stp ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_among ebt_802_3 dm9601 crc7 crc_itu_t crc_ccitt compat_xtables compat cdc_wdm cdc_subset cdc_ncm cdc_ether cdc_eem cdc_acm brcmutil ax88179_178a asn1_decoder asix arptable_filter arpt_mangle arp_tables sch_cake sch_teql sch_sfq sch_red sch_prio sch_pie sch_multiq sch_gred sch_fq sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp act_simple act_police act_pedit act_ipt act_gact act_csum sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred snd_bcm2835(C) hid evdev spi_ks8995 xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport
[63788.167056]  ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat ip6t_NPT ip6t_MASQUERADE nf_nat_ipv6 nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 isdn_bsdcomp dss1_divert isdn slhc ifb sit sctp libcrc32c tunnel4 ip_tunnel veth tun snd_rawmidi snd_seq_device snd_pcm_oss snd_pcm_dmaengine snd_pcm snd_timer snd_mixer_oss snd_hwdep snd_compress snd soundcore mISDN_dsp l1oip mISDN_core nls_utf8 eeprom_93cx6 sha256_generic seqiv jitterentropy_rng drbg md5 hmac ghash_generic gf128mul gcm ctr vfat
[63788.256205]  fat nls_iso8859_1 nls_cp437 xhci_plat_hcd xhci_hcd ehci_platform ehci_hcd gpio_button_hotplug ptp broadcom bcm_phy_lib pps_core
[63788.272576] CPU: 1 PID: 0 Comm: swapper/1 Tainted: G         C        4.19.72 #0
[63788.283619] Hardware name: Raspberry Pi 3 Model B Plus Rev 1.3 (DT)
[63788.291717] pstate: 80000005 (Nzcv daif -PAN -UAO)
[63788.298293] pc : refcount_sub_and_test_checked+0x58/0xb0
[63788.305367] lr : refcount_sub_and_test_checked+0x58/0xb0
[63788.312365] sp : ffffff800800bb40
[63788.317299] x29: ffffff800800bb40 x28: ffffffc03a39fa38 
[63788.324207] x27: 0000000000000000 x26: 0000000000010600 
[63788.331065] x25: 0000000000000001 x24: ffffffc038f8c059 
[63788.337865] x23: ffffff80088f39c0 x22: ffffffc034ed4ec0 
[63788.344607] x21: ffffffc035b0e7b8 x20: ffffff80088f8000 
[63788.351287] x19: 0000000000000000 x18: 0000000000000000 
[63788.357928] x17: 0000000000000000 x16: 0000000000000000 
[63788.364487] x15: 0000000000000001 x14: 0720072007200720 
[63788.370983] x13: 0720072007200720 x12: ffffff8009309dd8 
[63788.377431] x11: 00000000ffffffff x10: 0000000000000001 
[63788.383826] x9 : 0000000000000008 x8 : 0000000000000000 
[63788.390161] x7 : 0000000000000007 x6 : 0000000000000261 
[63788.396421] x5 : 0000000000000001 x4 : 0000000000000001 
[63788.402605] x3 : 0000000000000007 x2 : 0000000000000007 
[63788.408725] x1 : 0000000000000006 x0 : 0000000000000026 
[63788.414789] Call trace:
[63788.417933]  refcount_sub_and_test_checked+0x58/0xb0
[63788.423613]  refcount_dec_and_test_checked+0x14/0x20
[63788.429253]  mptcp_mpcb_put+0x18/0x58
[63788.433607]  mptcp_sock_destruct+0xb8/0x120
[63788.438476]  __sk_destruct+0x2c/0x150
[63788.442819]  sk_destruct+0x38/0x40
[63788.446889]  __sk_free+0x30/0x40
[63788.450783]  sk_free+0x28/0x38
[63788.454499]  tcp_v4_rcv+0x894/0xe20
[63788.458653]  ip_local_deliver_finish+0x130/0x1c8
[63788.463953]  ip_local_deliver+0xb0/0xc8
[63788.468477]  ip_rcv_finish+0x40/0x60
[63788.472745]  ip_rcv+0x74/0x98
[63788.476387]  __netif_receive_skb_one_core+0x3c/0x48
[63788.481970]  __netif_receive_skb+0x6c/0x78
[63788.486766]  process_backlog+0xa8/0x170
[63788.491320]  net_rx_action+0x110/0x298
[63788.495812]  __do_softirq+0xe0/0x1dc
[63788.500142]  irq_exit+0x60/0xb8
[63788.504056]  handle_IPI+0xac/0x140
[63788.508208]  bcm2836_arm_irqchip_handle_irq+0x8c/0xb8
[63788.514038]  el1_irq+0xb0/0x140
[63788.517992]  arch_cpu_idle+0x10/0x18
[63788.522347]  do_idle+0x10c/0x228
[63788.526346]  cpu_startup_entry+0x20/0x28
[63788.531041]  secondary_start_kernel+0x1ac/0x1c0
[63788.536360] ---[ end trace 70aba623a1a31d77 ]---

impact no know

## Steps to Reproduce the Problem

Sporadic behaviour, default scheduler

## Specifications

  - OpenMPTCProuter version: 0.52.3
  - OpenMPTCProuter VPS version: 0.999
  - OpenMPTCProuter platform: RPI3B+

ragefan, 2019-10-20:
This happens again :

Impact is that a list temporary the internet connection :

371866.073779] ------------[ cut here ]------------
[371866.079645] refcount_t: underflow; use-after-free.
[371866.085670] WARNING: CPU: 2 PID: 0 at lib/refcount.c:187 refcount_sub_and_test_checked+0x58/0xb0
[371866.096834] Modules linked in: rtl8192cu rtl8192c_common rtl_usb rt2800usb rt2800lib qcserial option ipw brcmfmac usb_wwan usb_serial_simple ti_usb_3410_5052 sierra rtlwifi rtl8xxxu rtl8187 rt2x00usb rt2x00lib pppoe ppp_async pl2303 oti6858 nf_nat_pptp nf_conntrack_pptp mt7601u mos7720 mmc_spi mct_u232 mac80211 keyspan iptable_nat ipt_REJECT ipt_MASQUERADE garmin_gps ftdi_sio ebtable_nat ebtable_filter ebtable_broute cypress_m8 cp210x ch341 cfg80211 cdc_mbim belkin_sa bas_gigaset ark3116 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_owner xt_ndpi xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iface xt_hl xt_helper xt_hashlimit xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_condition xt_comment xt_addrtype xt_TRACE xt_TPROXY
[371866.180893]  xt_TCPMSS xt_REDIRECT xt_LOG xt_IPMARK xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY xt_ACCOUNT visor usbserial usbhid usb_gigaset ts_fsm ts_bm sr9700 sierra_net ser_gigaset rtl8150 rndis_host r8712u(C) r8152 qmi_wwan pppox ppp_generic pegasus of_mmc_spi nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_proto_gre nf_nat_irc nf_nat_ipv4 nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_proto_gre nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount mcs7830 macvlan kaweth kalmia iptable_raw iptable_mangle iptable_filter
[371866.269436]  ipt_ECN ipheth ip6table_raw ip_tables huawei_cdc_ncm hso hid_generic gigaset ezusb ethoc ebtables ebt_vlan ebt_stp ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_among ebt_802_3 dm9601 crc7 crc_itu_t crc_ccitt compat_xtables compat cdc_wdm cdc_subset cdc_ncm cdc_ether cdc_eem cdc_acm brcmutil ax88179_178a asn1_decoder asix arptable_filter arpt_mangle arp_tables sch_cake sch_teql sch_sfq sch_red sch_prio sch_pie sch_multiq sch_gred sch_fq sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp act_simple act_police act_pedit act_ipt act_gact act_csum sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred snd_bcm2835(C) hid evdev spi_ks8995 xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport
[371866.356689]  ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat ip6t_NPT ip6t_MASQUERADE nf_nat_ipv6 nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 isdn_bsdcomp dss1_divert isdn slhc ifb sit sctp libcrc32c tunnel4 ip_tunnel veth tun snd_rawmidi snd_seq_device snd_pcm_oss snd_pcm_dmaengine snd_pcm snd_timer snd_mixer_oss snd_hwdep snd_compress snd soundcore mISDN_dsp l1oip mISDN_core nls_utf8 eeprom_93cx6 sha256_generic seqiv jitterentropy_rng drbg md5 hmac ghash_generic gf128mul gcm ctr vfat
[371866.445903]  fat nls_iso8859_1 nls_cp437 xhci_plat_hcd xhci_hcd ehci_platform ehci_hcd gpio_button_hotplug ptp broadcom bcm_phy_lib pps_core
[371866.462356] CPU: 2 PID: 0 Comm: swapper/2 Tainted: G         C        4.19.72 #0
[371866.473482] Hardware name: Raspberry Pi 3 Model B Plus Rev 1.3 (DT)
[371866.481660] pstate: 40000005 (nZcv daif -PAN -UAO)
[371866.488327] pc : refcount_sub_and_test_checked+0x58/0xb0
[371866.495481] lr : refcount_sub_and_test_checked+0x58/0xb0
[371866.502563] sp : ffffff8008013b40
[371866.507587] x29: ffffff8008013b40 x28: ffffffc03a3b4a38 
[371866.514581] x27: 0000000000000000 x26: 0000000000010600 
[371866.521528] x25: 0000000000000001 x24: ffffffc034f840a5 
[371866.528411] x23: ffffff80088f39c0 x22: ffffffc034d89180 
[371866.535241] x21: ffffffc034d88778 x20: ffffff80088f8000 
[371866.542011] x19: 0000000000000000 x18: 0000000000000000 
[371866.548737] x17: 0000000000000000 x16: 0000000000000000 
[371866.555384] x15: 0000000000000001 x14: 0720072007200720 
[371866.561963] x13: 0720072007630761 x12: ffffff8009309dd8 
[371866.568498] x11: 00000000ffffffff x10: 0000000000000001 
[371866.574985] x9 : 0000000000000008 x8 : 0000000000000000 
[371866.581406] x7 : 0000000000000007 x6 : 0000000000000263 
[371866.587755] x5 : 0000000000000000 x4 : 0000000000000000 
[371866.594031] x3 : 0000000000000000 x2 : ffffffc03a3b0e98 
[371866.600256] x1 : 0000004031b9c000 x0 : 0000000000000026 
[371866.606451] Call trace:
[371866.609720]  refcount_sub_and_test_checked+0x58/0xb0
[371866.615525]  refcount_dec_and_test_checked+0x14/0x20
[371866.621293]  mptcp_mpcb_put+0x18/0x58
[371866.625777]  mptcp_sock_destruct+0xb8/0x120
[371866.630772]  __sk_destruct+0x2c/0x150
[371866.635141]  sk_destruct+0x38/0x40
[371866.639238]  __sk_free+0x30/0x40
[371866.643157]  sk_free+0x28/0x38
[371866.646899]  tcp_v4_rcv+0x894/0xe20
[371866.651082]  ip_local_deliver_finish+0x130/0x1c8
[... cut ...]
[371866.679348]  process_backlog+0xa8/0x170
[371866.683923]  net_rx_action+0x110/0x298
[371866.688428]  __do_softirq+0xe0/0x1dc
[371866.692760]  irq_exit+0x60/0xb8
[371866.696653]  handle_IPI+0xac/0x140
[371866.700792]  bcm2836_arm_irqchip_handle_irq+0x8c/0xb8
[371866.706597]  el1_irq+0xb0/0x140
[371866.710500]  arch_cpu_idle+0x10/0x18
[371866.714828]  do_idle+0x10c/0x228
[371866.718806]  cpu_startup_entry+0x20/0x28
[371866.723484]  secondary_start_kernel+0x1ac/0x1c0
[371866.728779] ---[ end trace 2c52dffdab2deb30 ]---

Ysurac, 2019-12-20:
You can test with v0.53, if this still doesn't work you can reopen this issue.

closed 2019-12-20 by Ysurac

rustequal, 2020-03-12:
I have the same issue on v0.53.3 and RPi 4 (VPS version 0.1012):

```
[33169.758711] ------------[ cut here ]------------
[33169.763371] refcount_t: underflow; use-after-free.
[33169.768222] WARNING: CPU: 3 PID: 0 at lib/refcount.c:187 refcount_sub_and_test_checked+0x50/0xa0
[33169.777012] Modules linked in: ath9k ath9k_common rtl8192cu rtl8192c_common rtl_usb rt2800usb rt2800lib qcserial option ipw brcmfmac ath9k_hw ath usb_wwan usb_serial_simple ti_usb_3410_5052 sierra rtlwifi rtl8xxxu rtl8187 rt2x00usb rt2x00lib pppoe ppp_async pl2303 oti6858 nf_nat_pptp nf_conntrack_pptp mt7601u mos7720 mmc_spi mct_u232 mac80211 keyspan iptable_nat ipt_REJECT ipt_MASQUERADE garmin_gps ftdi_sio ebtable_nat ebtable_filter ebtable_broute cypress_m8 cp210x ch341 cfg80211 cdc_mbim belkin_sa bas_gigaset ark3116 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_owner xt_ndpi xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iface xt_hl xt_helper xt_hashlimit xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_condition xt_comment
[33169.848378]  xt_addrtype xt_TRACE xt_TPROXY xt_TCPMSS xt_REDIRECT xt_LOG xt_IPMARK xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY xt_ACCOUNT wireguard visor via_velocity via_rhine usbserial usbhid usb_gigaset tulip ts_fsm ts_bm sr9700 solos_pci sky2 skge sis900 sis190 sierra_net ser_gigaset rtl8150 rndis_host r8712u(C) r8169 r8152 r6040 qmi_wwan pppox ppp_generic pegasus pcnet32 of_mmc_spi nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_proto_gre nf_nat_irc nf_nat_ipv4 nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_proto_gre nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast
[33169.920403]  ts_kmp nf_conntrack_amanda nf_conncount ne2k_pci mcs7830 macvlan kaweth kalmia iptable_raw iptable_mangle iptable_filter ipt_ECN ipheth ip6table_raw ip_tables huawei_cdc_ncm hso hid_generic gigaset forcedeth ezusb ethoc et131x ebtables ebt_vlan ebt_stp ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_among ebt_802_3 e1000e e100 dm9601 crc7 crc_itu_t crc_ccitt compat_xtables compat cdc_wdm cdc_subset cdc_ncm cdc_ether cdc_eem cdc_acm brcmutil bnx2 ax88179_178a atl2 atl1e atl1c atl1 asn1_decoder asix arptable_filter arpt_mangle arp_tables 8390 8250_pci 8139too 8139cp sch_cake sch_teql sch_sfq sch_red sch_prio sch_pie sch_multiq sch_gred sch_fq sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp act_simple act_police act_pedit act_ipt act_gact act_csum sch_tbf sch_ingress sch_htb
[33169.991616]  sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred snd_bcm2835(C) hid evdev spi_ks8995 xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat ip6t_NPT ip6t_MASQUERADE nf_nat_ipv6 nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 isdn_bsdcomp dss1_divert isdn slhc 3c59x vmxnet3 e1000 ifb ip6_udp_tunnel udp_tunnel sit sctp libcrc32c tunnel4 ip_tunnel hfcpci hfcmulti
[33170.062324]  veth tun snd_rawmidi snd_seq_device snd_pcm_oss snd_pcm_dmaengine snd_pcm snd_timer snd_mixer_oss snd_hwdep snd_compress snd soundcore mISDN_dsp l1oip mISDN_core br2684 atm nls_utf8 natsemi eeprom_93cx6 sha256_generic seqiv jitterentropy_rng drbg md5 hmac ghash_generic gcm ctr vfat fat nls_iso8859_1 nls_cp437 ahci libahci libata ehci_platform ehci_hcd gpio_button_hotplug tg3 b44 ssb ptp realtek pps_core
[33170.099346] CPU: 3 PID: 0 Comm: swapper/3 Tainted: G         C        4.19.84 #0
[33170.106744] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
[33170.112580] pstate: 60000005 (nZCv daif -PAN -UAO)
[33170.117378] pc : refcount_sub_and_test_checked+0x50/0xa0
[33170.122694] lr : refcount_sub_and_test_checked+0x50/0xa0
[33170.128005] sp : ffffff800801bb10
[33170.131318] x29: ffffff800801bb10 x28: ffffffc07ffb9a38 
[33170.136638] x27: 0000000000000000 x26: 0000000000000001 
[33170.141956] x25: 0000000000000001 x24: ffffffc073321ed1 
[33170.147273] x23: ffffff80089db680 x22: ffffffc074996040 
[33170.152590] x21: ffffffc078b64600 x20: ffffff80089e0000 
[33170.157907] x19: 0000000000000000 x18: 0000000000000000 
[33170.163223] x17: 0000000000000000 x16: 0000000000000000 
[33170.168540] x15: 0000000000000001 x14: 0720072007200720 
[33170.173857] x13: 0720072007200720 x12: 0720072007200720 
[33170.179173] x11: 0720072007200720 x10: 0720072007200720 
[33170.184490] x9 : 0720072007200720 x8 : 0000000000000000 
[33170.189807] x7 : 0000000000000008 x6 : 00000000000002b8 
[33170.195125] x5 : ffffff800831f660 x4 : 0000000000000008 
[33170.200441] x3 : ffffff80088f8b6c x2 : 0000000000000004 
[33170.205757] x1 : 0000000000000004 x0 : 0000000000000026 
[33170.211075] Call trace:
[33170.213526]  refcount_sub_and_test_checked+0x50/0xa0
[33170.218494]  refcount_dec_and_test_checked+0x14/0x20
[33170.223462]  mptcp_mpcb_put+0x18/0x58
[33170.227124]  mptcp_sock_destruct+0xb0/0x120
[33170.231311]  __sk_destruct+0x2c/0x140
[... cut ...]
[17953.601159]  process_backlog+0xa8/0x170
[17953.604995]  net_rx_action+0x108/0x290
[17953.608745]  __do_softirq+0xdc/0x1d8
[17953.612324]  irq_exit+0x60/0xb8
[17953.615467]  handle_IPI+0xb0/0x140
[17953.618867]  gic_handle_irq+0xa0/0xb8
[17953.622529]  el1_irq+0xb0/0x140
[17953.625673]  arch_cpu_idle+0x10/0x18
[17953.629251]  do_idle+0xfc/0x218
[17953.632392]  cpu_startup_entry+0x24/0x28
[17953.636315]  secondary_start_kernel+0x150/0x160
[17953.640845] ---[ end trace 56500074bf0214a9 ]---
```

Ysurac, 2020-03-13:
Try to update to latest release (0.54.2)