Ysurac/openmptcprouter #695: Glorytun Read operation time out. Tunnel down for very short time.

Marctraider, 2019-10-14:
## Expected Behavior

Glorytun not malfunctioning/

## Actual Behavior

Glorytun malfunctioning and DC?

I wonder if this is a bug with glorytun tho? I dont know why else it would do it. There was no latency increase before it happened either. just totally random.

I run OMR with this in mind: As little as much in background that intentionally disable connection, everything is fixed as much as possible with the idea 'Everything always runs and connected'. Because I dont care for max aggregation speed and I dont care for fallback tech.

This might look like a glorytun-udp issue? Not sure...

## Steps to Reproduce the Problem

  1. Just stay online for few days. (uptime in this case little over a day.
2. Run without OMS tracker and all the stuffs (Except OMRVPN otherwise vpn doesnt start on reboot?)
3. Both aggregated connections were fine during this time, MPTCP and TCP connections all stayed alive.

## Specifications

  - OpenMPTCProuter version: 0.52.3
  - OpenMPTCProuter VPS version: 0.1001
  - OpenMPTCProuter platform: WRT3200ACM

Log, pay attention to this section:

Mon Oct 14 03:33:07 2019 daemon.notice netifd: wan2 (4216): udhcpc: sending renew to 192.168.1.1
Mon Oct 14 03:33:07 2019 daemon.notice netifd: wan2 (4216): udhcpc: lease of 91.141.139.197 obtained, lease time 300
Mon Oct 14 03:35:26 2019 daemon.err glorytun[15724]: read: Operation timed out
Mon Oct 14 03:35:26 2019 daemon.info glorytun[15724]: STOPPED tun0
Mon Oct 14 03:35:27 2019 daemon.err glorytun[15724]: 89.207.131.164.65001: connected
Mon Oct 14 03:35:28 2019 daemon.info glorytun[15724]: STARTED tun0
Mon Oct 14 03:35:37 2019 daemon.notice netifd: wan2 (4216): udhcpc: sending renew to 192.168.1.1
Mon Oct 14 03:35:37 2019 daemon.notice netifd: wan2 (4216): udhcpc: lease of 91.141.139.197 obtained, lease time 300
Mon Oct 14 03:36:28 2019 user.notice post-tracking-post-tracking: Get status and settings for wan2...
Mon Oct 14 03:37:20 2019 user.notice post-tracking-post-tracking: Get status and settings for wan2... Done

Full:

Mon Oct 14 00:50:33 2019 user.notice post-tracking-post-tracking: Get status and settings for wan1... Done
Mon Oct 14 00:53:00 2019 daemon.notice netifd: wan2 (4216): udhcpc: sending renew to 192.168.1.1
Mon Oct 14 00:53:00 2019 daemon.notice netifd: wan2 (4216): udhcpc: lease of 91.141.139.197 obtained, lease time 300
Mon Oct 14 00:55:30 2019 daemon.notice netifd: wan2 (4216): udhcpc: sending renew to 192.168.1.1
Mon Oct 14 00:55:30 2019 daemon.notice netifd: wan2 (4216): udhcpc: lease of 91.141.139.197 obtained, lease time 300
Mon Oct 14 00:58:00 2019 daemon.notice netifd: wan2 (4216): udhcpc: sending renew to 192.168.1.1
Mon Oct 14 00:58:00 2019 daemon.notice netifd: wan2 (4216): udhcpc: lease of 91.141.139.197 obtained, lease time 300
Mon Oct 14 01:00:30 2019 daemon.notice netifd: wan2 (4216): udhcpc: sending renew to 192.168.1.1
Mon Oct 14 01:00:30 2019 daemon.notice netifd: wan2 (4216): udhcpc: lease of 91.141.139.197 obtained, lease time 300
Mon Oct 14 01:03:01 2019 daemon.notice netifd: wan2 (4216): udhcpc: sending renew to 192.168.1.1
Mon Oct 14 01:03:01 2019 daemon.notice netifd: wan2 (4216): udhcpc: lease of 91.141.139.197 obtained, lease time 300
Mon Oct 14 01:05:31 2019 daemon.notice netifd: wan2 (4216): udhcpc: sending renew to 192.168.1.1
Mon Oct 14 01:05:31 2019 daemon.notice netifd: wan2 (4216): udhcpc: lease of 91.141.139.197 obtained, lease time 300
Mon Oct 14 01:08:01 2019 daemon.notice netifd: wan2 (4216): udhcpc: sending renew to 192.168.1.1
Mon Oct 14 01:08:01 2019 daemon.notice netifd: wan2 (4216): udhcpc: lease of 91.141.139.197 obtained, lease time 300
Mon Oct 14 01:09:31 2019 daemon.notice netifd: wan3 (4214): udhcpc: sending renew to 46.231.22.42
Mon Oct 14 01:09:31 2019 daemon.notice netifd: wan3 (4214): udhcpc: lease of 192.145.56.17 obtained, lease time 25200
Mon Oct 14 01:10:31 2019 daemon.notice netifd: wan2 (4216): udhcpc: sending renew to 192.168.1.1
Mon Oct 14 01:10:31 2019 daemon.notice netifd: wan2 (4216): udhcpc: lease of 91.141.139.197 obtained, lease time 300
Mon Oct 14 01:13:01 2019 daemon.notice netifd: wan2 (4216): udhcpc: sending renew to 192.168.1.1
Mon Oct 14 01:13:01 2019 daemon.notice netifd: wan2 (4216): udhcpc: lease of 91.141.139.197 obtained, lease time 300
Mon Oct 14 01:15:31 2019 daemon.notice netifd: wan2 (4216): udhcpc: sending renew to 192.168.1.1
Mon Oct 14 01:15:31 2019 daemon.notice netifd: wan2 (4216): udhcpc: lease of 91.141.139.197 obtained, lease time 300
Mon Oct 14 01:18:01 2019 daemon.notice netifd: wan2 (4216): udhcpc: sending renew to 192.168.1.1
Mon Oct 14 01:18:01 2019 daemon.notice netifd: wan2 (4216): udhcpc: lease of 91.141.139.197 obtained, lease time 300
Mon Oct 14 01:20:31 2019 daemon.notice netifd: wan2 (4216): udhcpc: sending renew to 192.168.1.1
Mon Oct 14 01:20:31 2019 daemon.notice netifd: wan2 (4216): udhcpc: lease of 91.141.139.197 obtained, lease time 300
Mon Oct 14 01:23:02 2019 daemon.notice netifd: wan2 (4216): udhcpc: sending renew to 192.168.1.1
Mon Oct 14 01:23:02 2019 daemon.notice netifd: wan2 (4216): udhcpc: lease of 91.141.139.197 obtained, lease time 300
[... 122 lines cut ...]
Mon Oct 14 03:35:37 2019 daemon.notice netifd: wan2 (4216): udhcpc: lease of 91.141.139.197 obtained, lease time 300
Mon Oct 14 03:36:28 2019 user.notice post-tracking-post-tracking: Get status and settings for wan2...
Mon Oct 14 03:37:20 2019 user.notice post-tracking-post-tracking: Get status and settings for wan2... Done
Mon Oct 14 03:38:07 2019 daemon.notice netifd: wan2 (4216): udhcpc: sending renew to 192.168.1.1
Mon Oct 14 03:38:07 2019 daemon.notice netifd: wan2 (4216): udhcpc: lease of 91.141.139.197 obtained, lease time 300
Mon Oct 14 03:39:23 2019 daemon.info glorytun[15724]: STOPPED tun0
Mon Oct 14 03:39:24 2019 daemon.err glorytun[15724]: 89.207.131.164.65001: connected
Mon Oct 14 03:39:24 2019 daemon.info glorytun[15724]: STARTED tun0
Mon Oct 14 03:40:37 2019 daemon.notice netifd: wan2 (4216): udhcpc: sending renew to 192.168.1.1
Mon Oct 14 03:40:37 2019 daemon.notice netifd: wan2 (4216): udhcpc: lease of 91.141.139.197 obtained, lease time 300
Mon Oct 14 03:43:07 2019 daemon.notice netifd: wan2 (4216): udhcpc: sending renew to 192.168.1.1
Mon Oct 14 03:43:07 2019 daemon.notice netifd: wan2 (4216): udhcpc: lease of 91.141.139.197 obtained, lease time 300
Mon Oct 14 03:45:37 2019 daemon.notice netifd: wan2 (4216): udhcpc: sending renew to 192.168.1.1
Mon Oct 14 03:45:37 2019 daemon.notice netifd: wan2 (4216): udhcpc: lease of 91.141.139.197 obtained, lease time 300
Mon Oct 14 03:48:08 2019 daemon.notice netifd: wan2 (4216): udhcpc: sending renew to 192.168.1.1
Mon Oct 14 03:48:08 2019 daemon.notice netifd: wan2 (4216): udhcpc: lease of 91.141.139.197 obtained, lease time 300
Mon Oct 14 03:49:56 2019 user.notice post-tracking-post-tracking: Get status and settings for wan1...
Mon Oct 14 03:50:06 2019 user.notice post-tracking-post-tracking: Get status and settings for wan1... Done
Mon Oct 14 03:50:38 2019 daemon.notice netifd: wan2 (4216): udhcpc: sending renew to 192.168.1.1
Mon Oct 14 03:50:38 2019 daemon.notice netifd: wan2 (4216): udhcpc: lease of 91.141.139.197 obtained, lease time 300
Mon Oct 14 03:53:08 2019 daemon.notice netifd: wan2 (4216): udhcpc: sending renew to 192.168.1.1
Mon Oct 14 03:53:08 2019 daemon.notice netifd: wan2 (4216): udhcpc: lease of 91.141.139.197 obtained, lease time 300
Mon Oct 14 03:55:38 2019 daemon.notice netifd: wan2 (4216): udhcpc: sending renew to 192.168.1.1
Mon Oct 14 03:55:38 2019 daemon.notice netifd: wan2 (4216): udhcpc: lease of 91.141.139.197 obtained, lease time 300
Mon Oct 14 03:56:52 2019 daemon.err uhttpd[4292]: luci: accepted login on / for root from 192.168.100.100

Ysurac, 2019-10-14:
There is a script on the VPS that check if the VPN is working. If it can't ping the other end then it's restarted. It's the omr services.

Marctraider, 2019-10-14:
Then I have to disable the VPS side tracker as well as everything was working fine.

I think this tracker is not fool proof enough.

Ysurac, 2019-10-14:
The script work as expected : client end doesn't answer to ping, restart VPN because there is a problem.
You can use systemctl to disable it, this will also disable 6in4 tunnel update.

closed 2019-10-14 by Ysurac