Ysurac/openmptcprouter #699: Problem port forwarding

silver9195, 2019-10-16:
## Expected Behavior

Good port forwarding 

## Actual Behavior

My actual port fowarding does'nt work...

My vps and vpn is good all is green.
i set up my gateway with the router address (192.168.100.1)
my firewall rules ar good (they worked well before)
In VPS configuration "rediriger tous les ports du serveur vers ce router" is checked

i give you my log after your response i just reboot my router

Thks

## Specifications

OpenMPTCProuter version: 0.52.3
OpenMPTCProuter VPS version: 0.1001 4.19.67
OpenMPTCProuter platform: Raspberry Pi 3 Model B Plus Rev 1.3

buseoti, 2019-10-22:
Hi,

i have exactly same problem, have you find solution ?

Thx
fred

silver9195, 2019-10-23:
No I haven't

SvenCarstensen, 2019-10-28:
Same problem here.

Ysurac, 2019-10-28:
You can check on the VPS, in /etc/shorewall/rules that DNAT lines are uncommented (without the # at the beginning). If it's ok and if params.vpn in this directory have IPs of the VPN used, then all ports are forwarded to the router (in next release, only needed ports are forwarded).

silver9195, 2019-10-28:
Salut Ysurac, 

I checked the file rules and DNAT are uncommented ... 
In params.vpn there are 
vps address 10.255.253.1 
Omr adresse 10.255.253.29
Vps iface mlvpn0

Any idea ?

Ysurac, 2019-10-29:
If you really use MLVPN, don't use it: It's not working at all, it segfault at launch.
If not, go to wizard and check in advanced parameters the VPN used and "Save & Apply" again.

silver9195, 2019-10-29:
i went to VPN , MLVPN and i unchecked the active case and "Save & Apply".

i reboot it but nothings .

It was well as you want i do ?

Ysurac, 2019-10-29:
You need to choose a VPN (not MLVPN) in the System->OpenMPTCProuter, wizard page, advanced settings.
Port forwarding use the VPN.

silver9195, 2019-11-13:
i used MLVPN and i changed this and after one reset it is ok .

Thanks Ysurac

closed 2019-11-13 by Ysurac