Ysurac/openmptcprouter #702: High latency when multipath is in use

max4186, 2019-10-17:
## Expected Behavior

Standard low ping, no jitter.

## Actual Behavior

Whilst multipath is on with something like A Dead Simple VPN, GloryTUN UDP/TCP - I get high/jittery ping. As soon as it turns off to switch to another one when I go to change it, ping stabilises.

## Steps to Reproduce the Problem

Reply from 1.1.1.1: bytes=32 time=35ms TTL=58
Reply from 1.1.1.1: bytes=32 time=43ms TTL=58
Reply from 1.1.1.1: bytes=32 time=48ms TTL=58
Reply from 1.1.1.1: bytes=32 time=47ms TTL=58
Reply from 1.1.1.1: bytes=32 time=31ms TTL=57 - Everything to here is prior to changing the settings.
Reply from 1.1.1.1: bytes=32 time=32ms TTL=57 
Reply from 1.1.1.1: bytes=32 time=31ms TTL=57
Reply from 1.1.1.1: bytes=32 time=31ms TTL=57
Reply from 1.1.1.1: bytes=32 time=31ms TTL=57
Reply from 1.1.1.1: bytes=32 time=31ms TTL=57
Reply from 1.1.1.1: bytes=32 time=31ms TTL=57 - everything to here is when it's switching
Request timed out. - switches
Reply from 1.1.1.1: bytes=32 time=40ms TTL=58 - back to high ping on a new multipath VPN setting
Reply from 1.1.1.1: bytes=32 time=31ms TTL=58
Reply from 1.1.1.1: bytes=32 time=31ms TTL=58
Reply from 1.1.1.1: bytes=32 time=35ms TTL=58
Reply from 1.1.1.1: bytes=32 time=36ms TTL=58
Reply from 1.1.1.1: bytes=32 time=37ms TTL=58
Reply from 1.1.1.1: bytes=32 time=50ms TTL=58
Reply from 1.1.1.1: bytes=32 time=50ms TTL=58
Reply from 1.1.1.1: bytes=32 time=36ms TTL=58
Reply from 1.1.1.1: bytes=32 time=61ms TTL=58

## Specifications

  - OpenMPTCProuter version: 0.52.3
  - OpenMPTCProuter VPS version: 0.1001 4.19.67-mptcp
  - OpenMPTCProuter platform: RPI3 x86

Ysurac, 2019-10-17:
You have ping between you and the VPS added, it's why VPS provider have to be as near as possible. And ICMP is low priorities, it's not a problem.

max4186, 2019-10-17:
So you're saying it is impossible to have a steady connection with no high jitter using bonding because I'm 30ms away? That doesn't sound right, it was steady ALL of last night - full on 0.5 jitter but now it's like 7 out of no where. I have had this issue before.

max4186, 2019-10-17:
When MLVPN is in use it's a lot more stable, very seldom jumps but there still is like 20ms jumps every I'd say 20 ICMP returns.

For some reason it also says VPN is offline and OMR bypass is not working with it.

Ysurac, 2019-10-17:
I will look for MLVPN with omr-bypass.
ICMP can be ignored by any equipment for many reason.
You can also look to the master interface, if it's not the best rtt interface this can make some problems.

Marctraider, 2019-10-17:
> 
> 
> When MLVPN is in use it's a lot more stable, very seldom jumps but there still is like 20ms jumps every I'd say 20 ICMP returns.
> 
> For some reason it also says VPN is offline and OMR bypass is not working with it.

How do u even get mlvpn to work? it seem half broken ...

max4186, 2019-10-17:
Unfortunately pretty much all of the VPN methods are pretty buggy for me, when changing between VPN's OMR seems to break half the time and takes 5 minutes to 'kick' back in. Some sort of restart button or something would likely help.

Marctraider, 2019-10-17:
I'm decided to create my own image and/or packages and try on clean system to get mlvpn or something like vtrunkd to work..

max4186, 2019-10-17:
An even weirder issue I just noticed is somehow wan2 automatically set itself to master, I assume this happened when I turned off my wan1 for a reboot - shouldn't it automatically set back?

Ysurac, 2019-10-17:
Changing VPN need to set changes on the VPS side, so yes this can take some times. This will be better in next release.
Master interface can be changed if you set the paremeter for that...

max4186, 2019-10-17:
I just seen it change my wan2 to master again when I switched to DSVPN, it did it automatically - any idea why it would do that and is it something I can disable? (If so, where)?

Ysurac, 2019-10-17:
in advanced parameters, but it's not related to this issue.

max4186, 2019-10-17:
I think it is @Ysurac - I think the high pings I've been seeing is because my wan2 is 4G so when I was changing the VPN it was setting wan2 as my master and that was where the ping was from. I've switched VPN, changed wan1 (my FTTN connection) back to master and my ping is stable as it can be.

`Reply from 172.217.169.46: bytes=32 time=34ms TTL=55
Reply from 172.217.169.46: bytes=32 time=31ms TTL=55
Reply from 172.217.169.46: bytes=32 time=31ms TTL=55
Reply from 172.217.169.46: bytes=32 time=31ms TTL=55
Reply from 172.217.169.46: bytes=32 time=31ms TTL=55
Reply from 172.217.169.46: bytes=32 time=31ms TTL=55
Reply from 172.217.169.46: bytes=32 time=31ms TTL=55
Reply from 172.217.169.46: bytes=32 time=31ms TTL=55
Reply from 172.217.169.46: bytes=32 time=31ms TTL=55
Reply from 172.217.169.46: bytes=32 time=31ms TTL=55
Reply from 172.217.169.46: bytes=32 time=31ms TTL=55
Reply from 172.217.169.46: bytes=32 time=31ms TTL=55
Reply from 172.217.169.46: bytes=32 time=31ms TTL=55
Reply from 172.217.169.46: bytes=32 time=32ms TTL=55
Reply from 172.217.169.46: bytes=32 time=31ms TTL=55
Reply from 172.217.169.46: bytes=32 time=31ms TTL=55
Reply from 172.217.169.46: bytes=32 time=31ms TTL=55
Reply from 172.217.169.46: bytes=32 time=31ms TTL=55
Reply from 172.217.169.46: bytes=32 time=31ms TTL=55
Reply from 172.217.169.46: bytes=32 time=31ms TTL=55
Reply from 172.217.169.46: bytes=32 time=31ms TTL=55
Reply from 172.217.169.46: bytes=32 time=31ms TTL=55
Reply from 172.217.169.46: bytes=32 time=31ms TTL=55
Reply from 172.217.169.46: bytes=32 time=31ms TTL=55
Reply from 172.217.169.46: bytes=32 time=31ms TTL=55
Reply from 172.217.169.46: bytes=32 time=32ms TTL=55`

max4186, 2019-10-17:
The above is using DSVPN FYI.

Ysurac, 2019-10-17:
It's what I said before...

max4186, 2019-10-17:
How is that the case? You mentioned previously it was due to having high latency but this shows that latency isn't the issue. The issue is wan2 got set to my master, do you know why it would do that? Just curious, not meaning to waste your time - you are literally pioneering in this space so I appreciate everything you are doing.

Ysurac, 2019-10-17:
By default this should select interface with best RTT as master after wizard changes. Sometimes it's not the case (High RTT on interface that is often low).
So in System->OpenMPTCProuter, advanced settings tab you can set "Master interface selection" to no change and set the master in Network->MPTCP. Then the master interface will never change (while it's up).

max4186, 2019-10-17:
Ah, that makes sense. The issue is my FTTN connection is brand new and in the UK we have something called DLM which basically configures your line over a week and can cause ping to be higher (which is occurring to me, 30ms when it's generally 23ms) - my 4G wan2 is currently getting 27ms so lower and that must be causing the switch.

I'll use your suggestion to see if this stops any further ping issues going forward.

Ysurac, 2019-10-17:
"Multipath TCP scheduler" in network->MPTCP or/and "Congestion control" can sometimes give better results with others values than default.

max4186, 2019-10-17:
I'll try that too.

max4186, 2019-10-17:
How can I solve 'VPN tunnel is DOWN' on the status page whilst using DSVPN? I've noticed ICMP traffic is going over wan1 rather than my actual VPN tunnel so I assume there must be an issue somewhere @Ysurac

Ysurac, 2019-10-18:
If ICMP is going over wan1, I think the issue is that VPN is really down. You can check in system log and try to restart it in System->Startup.

max4186, 2019-10-18:
Ah okay, I'll take a look. I've moved back to GloryTUN TCP for now as I've done some extensive testing by pinging using TCP & UDP packets in addition to ICMP at the same time and it's clear that ICMP appears to not be treated as a priority by GloryTUN so whilst I thought I had high ping, both the TCP & UDP packets were remaining at the same amount so case closed really.

max4186, 2019-10-19:
@Ysurac Could you let me know your most recommended way of running this for someone who downloads a lot on a main fibre connection and a secondary 4G connection but also games? I've ran into a new issue today where I was downloading and my ping spikes to 2000ms for about 5 seconds and then the entire thing drops, including the actual VPS with no logs other than stuff going offline.

It's an amazing idea but so far I'm finding 0 way to run this in a normal environment without having to check on it every 3 hours.

Bestle, 2019-10-19:
I've been using a primary fibre connection configured as master with multiple 4G connections to increase throughput when downloading for a few months now.

It might not be that helpful but I am not experiencing any of the issues you are reporting, including when playing games, though I am using shadowsocks.

max4186, 2019-10-19:
> I've been using a primary fibre connection configured as master with multiple 4G connections to increase throughput when downloading for a few months now.
>
> It might not be that helpful but I am not [...]

Hey Bestle, do you have Discord or something like that I can PM you directly on with a few config questions? If you do, my discord is Cat#9155

Ysurac, 2019-10-19:
You can try to use SQM, it's made for that.
For configuration, you have to test: nothing is the best in all case. Many MPTCP settings are available like scheduler or TCP congestion. By default it's optimized for speed.

max4186, 2019-10-23:
I have now done a complete reinstall of everything to latest versions on formatted disks for both the Pi as well as my VPS. So far, so good. Ping is stable, downloading isn't causing 2000ms ping. I have also changed ICMP priority in OMR-DSCP so try and achieve some more accurate results.

Question @Ysurac - what does CS5 and CS7 do? It isn't mentioned like it is for others.

closed 2019-12-20 by Ysurac