Ysurac/openmptcprouter #703: IPv6 not working

romuald-r, 2019-10-18:
## Expected Behavior

IPv6 working on the LAN.

## Actual Behavior

Hi, I quickly configured an openmptcprouter router on a rpi3 in a classic KVM VM with publics IPv4/v6 (v4 /32 and /64 v6), all on 2 different DSL pairs and 2 4G (LTE) backup access. Everything works visibly, the only concern... is that the router can not route the IPv6 addresses of the VPS, it is very awkward.

In fact, when I click on "Enable IPv6" in the wizard of the router, the check mark is disabled after validating the form. There is no error message that appears despite this.

## Steps to Reproduce the Problem

  1. Setup the VPS
  2. Setup the router at home
  3. IPv4 is working fine, but not IPv6. I'm unable to activate the "Enable IPv6" checkbox.

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.52.3 r0+11006-713561a10b / LuCI Master (git-19.260.67281-bb408f2)
  - OpenMPTCProuter VPS version: Debian 10.1.0 x64
  - OpenMPTCProuter platform: Raspberry Pi 3 Model B Rev 1.2 x64

Ysurac, 2019-10-18:
The IPv6 checkbox problem is a know issue. This will be solved in next release.
What you will get is private IPv6 via a NAT (difficult to do more than that with a /64).

Ysurac, 2019-12-20:
You can test with v0.53, if this still doesn't work you can reopen this issue.

closed 2019-12-20 by Ysurac