Ysurac/openmptcprouter #710: tun0 not appearing as a MPTCP interface although has multipath enabled.

Llorx, 2019-10-28:
In the latest version, if I create a `tun0` device with a static IP in OpenVPN and `multipath` set to `on`, it don't appears in the main status section, and is not used as MPTCP routing.

`tun0` is MTCP capable, tested with `curl http://multipath-tcp.org/ --interface tun0`, but router status shows only one router (which is `eth1`):
[image]

But I think that everything is fine configured:
[image]

ifconfig result:
[image]

The MAC is 00:00:00... altough I forced it, so I don't know if this is the problem actually.

Llorx, 2019-10-28:
Ok, setting a `tap` instead of `tun` it works.

Ysurac, 2019-10-28:
Tun0 is already used as internal VPN.
But I think interface will not be used in MPTCP routing, at least not like a classic WAN. MPTCP over VPN is available (in alpha state) in current snapshot release: https://github.com/Ysurac/openmptcprouter/wiki/Snapshots

Llorx, 2019-10-28:
Woah, nice.

closed 2019-11-01 by Ysurac