Ysurac/openmptcprouter #713: Weird CPU Usage

max4186, 2019-11-01:
## Expected Behavior

For some odd reason, every time I go to sleep and get back up - my RPI 4 is using an average load of 1.8-2.0 and it's spread across all 4 cores at about 25% usage on each. After a reboot, it goes back to 0.3 and the entire time I am online I don't see the load increase. I have checked running processes and there is none actually using it. I have noticed this occurring when it failsover because a connection drops and it never goes down, could this be the issue?

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.52.3 r0+11006-713561a10b
  - OpenMPTCProuter VPS version: 0.1001 4.19.67
  - OpenMPTCProuter platform: Raspberry Pi 4 Model B Rev 1.1

max4186, 2019-11-01:
I am fairly certain this is caused by my wan1 dropping it's auth session to move to a new gateway as my ISP has a load balancing system for this (I can tell because my latency goes from 23ms to 26ms until I restart the session back to 23ms). Still an issue though why failover would cause higher CPU forever, even when wan1 returns.

user747, 2019-11-01:
If you ssh into your pi and run htop you should see processes that are using cpu. What VPN are you using (Shadowsocks, Glorytun)?

max4186, 2019-11-01:
I did run htop and as mentioned in my post, the most a process was using was 2%. I'm using Shadowsocks for TCP and nothing for UDP/ICMP as it caused major lag so I just have it going over wan1.

user747, 2019-11-01:
And you are seeing 1.8-2 in Luci 'Realtime Load '(/cgi-bin/luci/admin/status/realtime) ? 
Maybe take a look at System Log and see if WAN's are dropping as well. 

And WAN is Ethernet based coming in from a switch or something like that, or is it a USB device plugged directly into your Rpi4, like a cellular USB modem?

max4186, 2019-11-01:
I hadn't checked it in there but next time it happens, I'll check sys log and that page. The ethernet port is plugged in from a switch, which has my wan1 and wan2 connected to it.

user747, 2019-11-01:
I wouldn't worry about that page actually. Uhm, I'm on same hardware and version as you, but I use Glorytun. I haven't come across this issue. 

Yeah System Log might have more answers.

max4186, 2019-11-02:
I have now found the below:

`Sat Nov  2 13:30:22 2019 kern.err kernel: [78434.123192] xhci_hcd 0000:01:00.0: Ring expansion failed
Sat Nov  2 13:30:22 2019 kern.err kernel: [78434.131592] cpufreq: __target_index: Failed to change cpu frequency: -22
Sat Nov  2 13:30:22 2019 kern.err kernel: [78434.136915] hub 1-1:1.0: activate --> -12
Sat Nov  2 13:30:22 2019 kern.err kernel: [78434.146296] bcm2835-cpufreq:bcm2835_cpufreq_set_clock:76: Failed to set clock: 1500000 (-12)
Sat Nov  2 13:30:22 2019 kern.err kernel: [78434.156157] bcm2835-cpufreq:bcm2835_cpufreq_driver_target_index:175: Error occurred setting a new frequency (1500000)
Sat Nov  2 13:30:22 2019 kern.err kernel: [78434.166812] cpufreq: __target_index: Failed to change cpu frequency: -22`

@Ysurac - Any ideas?

Ysurac, 2019-11-02:
It try to change speed of the CPU, but no success. No idea why but you can set fixed speed in System->OpenMPTCProuter->Advanced settings tab.

max4186, 2019-11-02:
What do you recommend for the scaling governor?

Ysurac, 2019-11-02:
I set it to performance.

max4186, 2019-11-02:
Alright, I've changed it from ondemand to performance with minimum frequency at 600000 and max at 1500000. Will see if any update, if it's still broke I'll force it in the middle ground.

max4186, 2019-11-03:
Just an update that setting it to performance has resolved it, no high load from yesterday.

user747, 2019-11-08:
also make sure rpi4 using latest firmware

sudo apt install rpi-eeprom

"Starting with the Raspberry Pi 4 the device actually has onboard upgradable firmware stored on an EEPROM chip separate from your storage. Updating this firmware is very important as one of the first released updates reduces power usage of the Pi by 30% which also reduces how hot it gets."

Ysurac, 2019-11-08:
apt will not work on OpenMPTCProuter. I will try to make a package for rpi-eeprom.

max4186, 2019-11-08:
Sounds good, let us know when it's done so we can upgrade it's firmware.

user747, 2019-11-15:
> apt will not work on OpenMPTCProuter. I will try to make a package for rpi-eeprom.

@Ysurac excellent work. 

was missing vl805

downloaded vl805 from here https://drive.google.com/file/d/1nUFfKGI118bJ6DmiDm9euE6kWbOxbG7q/view?usp=sharing
see https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=250990&sid=4951bd18096a489c8d475962db451eaf

scp vl805 to /usr/bin
export vl805=/usr/bin/vl805
install package pciutils blkid (openwrt lspci version doesn't recognize -d parameter)

change 
/etc/default/rpi-eeprom-update

FIRMWARE_RELEASE_STATUS="critical"
FIRMWARE_RELEASE_STATUS="beta"

root@test:/# rpi-eeprom-update 
*** UPDATE REQUIRED ***
BOOTLOADER: update required
CURRENT: Fri May 10 18:40:36 UTC 2019 (1557513636)
 LATEST: Tue Sep 10 10:41:50 UTC 2019 (1568112110)
VL805: update required
CURRENT: 00013701
 LATEST: 000137ab

root@test:/# rpi-eeprom-update -a
*** INSTALLING EEPROM UPDATES ***
BOOTLOADER: update required
CURRENT: Fri May 10 18:40:36 UTC 2019 (1557513636)
 LATEST: Tue Sep 10 10:41:50 UTC 2019 (1568112110)
VL805: update required
CURRENT: 00013701
 LATEST: 000137ab
EEPROM updates pending. Please reboot to apply the update.

reboot and pray to the nobrick gods

root@test:~# rpi-eeprom-update -a
BOOTLOADER: up-to-date
CURRENT: Tue Sep 10 10:41:50 UTC 2019 (1568112110)
 LATEST: Tue Sep 10 10:41:50 UTC 2019 (1568112110)
VL805: up-to-date
CURRENT: 000137ab
 LATEST: 000137ab

 very nice

Ysurac, 2019-11-15:
I didn't have the time to test.
I added vl805 in rpi-eeprom package.
I wasn't expecting ```rpi-eeprom-update``` script to work without changes...
I think I will have to make a LuCi interface for that...

max4186, 2019-11-15:
Is there a dev build we need to upgrade to run this?

max4186, 2019-11-15:
@Ysurac

Ysurac, 2019-11-16:
This : https://github.com/Ysurac/openmptcprouter/wiki/Snapshots
But should be possible to install only rpi-eeprom and rpi-userland packages: https://www.openmptcprouter.com/develop/4.19/rpi4/packages/aarch64_cortex-a72/openmptcprouter/

max4186, 2019-11-22:
Do you know when you plan on moving this to main build?

user747, 2019-12-08:
openwrt master is also a few months behind on rpi4 firmware commits as well. 
see 
[https://github.com/openwrt/openwrt/blob/deb835849aeb713968aaa48bf046101140ab4825/package/kernel/brcm2708-gpu-fw/Makefile#L12](url)

and compare to [https://github.com/raspberrypi/firmware/commits/master](url)

might try testing with latest commit and file hashes and see if it works. something tells me it cant be that easy though

closed 2019-12-20 by Ysurac