Ysurac/openmptcprouter #714: Unexpected reboot

jordan-richardo, 2019-11-01:
I am not sure how I can capture some logging info for you, I am happy to do whatever to help :)

## Actual Behavior
Running system will randomly crash, but I am unable to find anything meaningful in the system log to help diagnose. I had a few WANS coming in via Vlan and thought perhaps there was a buffer overrun or something related to that, so I have moved to physical interfaces.  Seemingly random crashes, and so I moved from using a VM for the client to a physical box but the issue remains.

## Steps to Reproduce the Problem
-Random crashes
-Does not appear to be related to load or specific traffic

## Specifications

  - OpenMPTCProuter version: 0.52.3
  - OpenMPTCProuter VPS version: 0.1102
  - OpenMPTCProuter platform: x86_64 (used non EFI Ext4 image)
  -Rebuilt configuration from scratch instead of backup just in case.

Ysurac, 2019-11-01:
I would need the crash log, without that I can't do anything for now.
Should be available at: ```/sys/kernel/debug/crashlog```

jordan-richardo, 2019-11-01:
I can't seem to locate that file in that path, I did some quick googling for a path as well but can't find anything useful.

Here is what I see:
[image]

Ysurac, 2019-11-01:
You can try with the develop release to check if you still have the issue: https://github.com/Ysurac/openmptcprouter/wiki/Snapshots

jordan-richardo, 2019-11-01:
Same result unfortunately, I have a monitor connected thinking that I could maybe see the debug, but it disappears to fast for my slow brain to process haha

user747, 2019-11-01:
With a fresh install, do the crashes happen before configuring a VPN (Shadsowsocks, Glorytun, ect ect) or do they happen after VPN setup ? 

What kind of WAN's ?

jordan-richardo, 2019-11-03:
After a fresh install everything is fine until all the wan's are configured and data starts passing over them. It feels like a certain type of traffic is causing the crash as sometimes it will run happily overnight or during the day until something changes (hypothesizing traffic type here) and it will keep consistently crashing. 

I thought perhaps a heat or RAM issue, tried swapping boards and playing with cooling but no luck.

jordan-richardo, 2019-11-03:
Forgot to mention that the WAN's are all DSL, credentials dialed on the individual modems in router mode with static IP's set on each interface of the openmptcp box. Originally each WAN was on a Vlan, but tried moving to physical ports as well.

user747, 2019-11-05:
Someone mentioned that to use /sys/kernel/debug/crashlog
the RAM allocated for it must not be cleared by the bootloader or the kernel after a reset. So it's possible that it's getting cleared after crashed and unavailable. 

Maybe setting 'Write system log to file' with Debug level ( System--> Logging) might capture come events and persistence after a crash ? 

My thought is a hardware issue at his point so if you haven't tried on different hardware it might be worth it at this point.

jordan-richardo, 2019-11-05:
I found the issue though unsure how it is happening. It appears as though whenever an interface is added, the same MAC address is being used for each. I manually went through and created unique mac's and all is working just great now!

It must be something to do with creation of Vlan interfaces within OpenWRT causing it I think.

closed 2019-12-20 by Ysurac