Ysurac/openmptcprouter #716: Duplicate packages? Not FEC.

silfversound, 2019-11-03:
Is it possible to duplicate, triplicate or even quadruplicate packages through each wan? On server side only keep the packages that arrives first. 
My reason for this request is that I have time sensitive UDP audio streams. Any ideas?

Ysurac, 2019-11-03:
In Network->MPTCP You can set Redundant Multipath TCP scheduler. You can also set ndiffports instead of fullmesh.

silfversound, 2019-11-04:
Great, thank you @Ysurac. Will check it out. I asume it will handle the traffic both dirrections? Is it possible to set a jitter buffer for this? And is it possible to get jitter and latancy data out, possibly as a graph, ( a suggestion for future versions, maybe under the Bandwidth tab ).

closed 2019-12-20 by Ysurac