Ysurac/openmptcprouter #718: VPS server could not be installed

hljsykj, 2019-11-05:
VPS server can't be installed. I tried several times, but it didn't work

hljsykj, 2019-11-05:
[image]

Ysurac, 2019-11-05:
It can't download ```https://github.com/shadowsocks/shadowsocks-libev/releases/download/v3.3.1/shadowsocks-libev-3.3.1.tar.gz```. What is your VPS provider ?

hljsykj, 2019-11-05:
Alibaba cloud server can be installed before, but it can't be installed recently

hljsykj, 2019-11-05:
Alibaba cloud server

Ysurac, 2019-11-05:
You can try to ```wget https://github.com/shadowsocks/shadowsocks-libev/releases/download/v3.3.1/shadowsocks-libev-3.3.1.tar.gz``` and ```tar xzvf shadowsocks-libev-3.3.1.tar.gz```. You should have the same error...
Else maybe a cache/network/GFW error.

suyuan168, 2019-11-05:
> 阿里云服务器

阿里云目前大多数无法访问shadowsock-libev。你可以尝试其他的运营商。但是目前在中国好像有GFW阻止安装这些软件。

closed 2019-12-20 by Ysurac