Ysurac/openmptcprouter #719: very low speed

imercier, 2019-11-05:
## Expected Behavior
omr-test-speed ~20MB/s

## Actual Behavior
omr-test-speed ~500KB/s
omr-test-speed wan1 ~10MB/s
omr-test-speed wan1 ~10MB/s

## Steps to Reproduce the Problem

[image]

## Specifications

  - OpenMPTCProuter version: (v0.52.3 r0+11006-713561a10b)
  - OpenMPTCProuter VPS version: (0.1001)
  - OpenMPTCProuter platform: (RPI4)

I've tried to enable/disable *vpn, ShadowSocks.
rpi4 cpu is quite idle when testing.
config file attached.
[config.txt](https://github.com/Ysurac/openmptcprouter/files/3810280/config.txt)

thank you for your help

Roken120, 2019-11-12:
I have same problem with OVH vps

Roken120, 2019-11-12:
Maybe i dont use rpi4 i use vm

Ysurac, 2019-11-12:
You can try to disable fastopen and to set lower MTU

imercier, 2019-11-14:
I can't disable fast open, when I apply disable option, tick disapear in web interface, and config stay 1:
`shadowsocks-libev.hi1.fast_open='1'
`
Setting all wan interfaces at mtu 1350 doesn't help either.

Ysurac, 2019-11-14:
True there a bug on interface to disable fast open, it's fixed in develop branch for next release.
I have no idea for now. Did you try to reinstall both ?

imercier, 2019-11-20:
so how can I disable fast open?
yes I did try reinstall from scratch both.
BTW, overthebox is working properly

Ysurac, 2019-11-20:
```sysctl -w net.ipv4.tcp_fastopen=0```
You can also disable fastopen on shadowsocks luci interface.

closed 2019-12-20 by Ysurac