Ysurac/openmptcprouter #720: Problem with multiple interfaces

Darakko, 2019-11-06:
## Expected Behavior

the OpenMPTCProuter computer has 2 network cards.
One network card is dedicated to LAN:
    192.168.0.254/24

One network card is dedicated to WANs for 3 links (tested with 2 links, same problem) with a 5 ports Ethernet switch:
    192.168.11.254/24 for modem on 192.168.11.1
    192.168.12.254/24 for modem on 192.168.12.1
    192.168.13.254/24 for modem on 192.168.13.1

OpenMPTCProuter should work with this setup.

note: the first network card is integrated into the Raspberry. The second network card is an USB/Ethernet adapter.

This setup is usefull for several cases. Here are 2 cases I know in a daily basis:
- having OpenMPTCProuter doing its job using internet links located on different LANs (the LANs are for example different departments in a company, each containg computers connected to a single modem for each LAN)
- having several modems fully configured and ready to replace OpenMPTCProuter in case of failure

## Actual Behavior

OpenMPTCProuter is unable to connect to any of the 3 modems, nor ping.
The menu entry "System / OpenMPTCProuter / Status tab" never display anything.

The modems can ping OpenMPTCProuter.
A computer plugged into the WAN switch can ping OpenMPTCProuter and ssh into it (?!!).

## Steps to Reproduce the Problem

  1. OpenMPTCProuter fresh install
  2. add a network card
  3. use the additionnal network card to configure 2 or 3 WANs on different spans

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.52.3 r0+11006-713561a10b / LuCI Master (git-19.260.67281-bb408f2) 
  - OpenMPTCProuter VPS version: irrelevant
  - OpenMPTCProuter platform: virtual x64 computer (KVM on Linux Debian) and also Raspberry Pi 3 Model B Plus Rev 1.3

## Bypass methods

Currently I create 3 virtual network cards (using ssh, I did not found a way to do it from the GUI), they are correctly handled by OpenMPTCProuter and seen as real network cards, so no more problem.

Another bypass method is to have all 3 modems ont the same LAN
    192.168.10.1/24 for modem on 192.168.10.11
    192.168.10.2/24 for modem on 192.168.10.12
    192.168.10.3/24 for modem on 192.168.10.13
But this is not exactly the same thing.
When doing this second bypass, "System / OpenMPTCProuter / Status tab" display "Network interface duplicated".

Ysurac, 2019-11-06:
OpenMPTCProuter is working with 1 network card for all WANs (or even for WANs and LAN). It's why MACvlan is used. You should check that your switch, or something else, doesn't filter MACs on interfaces.

Darakko, 2019-11-06:
I do not understand your answer: does it means that this kind of setup should work, or does it mean my setup is incorrect?

The switch is very basic, so it should not filter anything.

Ysurac, 2019-11-06:
This kind of setup work and it's the default configuration on a RPI.

Darakko, 2019-11-07:
So, what can I do to help to solve this?

Ysurac, 2019-11-07:
Use a blank image and use the wizard to configure your interfaces.
Check using ```ip a``` and ```ip route``` via ssh that the configuration is correct.

suyuan168, 2019-11-07:
Please use the wizard to add all the NICs correctly and configure the server.

Darakko, 2019-11-14:
Done with a fresh install, from the wizard: everything is ok.

But with a fresh image, then wizard, then the USB Ethernet card added later, this does not work.

closed 2019-11-14 by Darakko