Ysurac/openmptcprouter #721: Question: omr-tracker - do not switch off multipath

LovingFox, 2019-11-07:
## Expected Behavior

Is it possible to keep multipath state of the interface switched on always in spite of OMR_TRACKER_STATUS variable by modifying configuration? Or I need to modify the code?

## Actual Behavior

multipath interface state depends on the OMR_TRACKER_STATUS.

## Steps to Reproduce the Problem

If "omr-tracker" detect a problem on the interface, OMR_TRACKER_STATUS is marked as ERROR and "post-tracking" script switches off multipath state for the interface.

## Specifications

  - OpenMPTCProuter version: v0.52.3-5-g6f9cd2d
  - OpenMPTCProuter VPS version: does not metter
  - OpenMPTCProuter platform: does not metter

user747, 2019-11-07:
would be nice to have this for #715 
but i think this needs code modification 
[# An interface in error will never be used in MPTCP](https://github.com/Ysurac/openmptcprouter-feeds/blob/ee49175152dc2f1382339a2339062dd479d86b6f/mptcp/files/usr/share/omr/post-tracking.d/post-tracking#L237)

Ysurac, 2019-11-07:
I don't see any case where this can be needed.

user747, 2019-11-07:
user wants to preserve privacy and security that is  gained when using a vpn (shadowsocks, glorytun)

privacy and security take precedence over redundancy in this case.

when failover to individual wan due to multipath master error state, routing is done on what the user may perceive as being a less trusted, less secure wan interface.

Ysurac, 2019-11-07:
yes but it's not related to this issue.
The issue is about keep multipath flag as true even if omr-tracker detect the interface as not working well.

user747, 2019-11-07:
no not directly related to this issue ;)

LovingFox, 2019-11-08:
> I don't see any case where this can be needed.

I try to explain when the problem arises.

The case considered we use tcp-vpn any type. No udp-vpn, no shadowsocks. Just one tcp-tunnel and all traffic is sent through that.

It may occurs all uplink interfaces are failed any reason (internet providers problems for example). So omr-tracker marks all of uplinks as ERROR and switches off mptcp. But it is not a reason VPN application stops to try reconnects to the server. It will try to reconnect again and again.

Some time later problems are gone for the interface the default gateway has via. And so happened VPN application manages to set up tunnel through worked uplink before omr-tracker swithes on mptcp back for the interface.

And that is a problem! This tcp-tunnel is set up WITHOUT mptcp option and the VPN tcp-session will NEVER use multi uplinks in spit of omr-traker switch them back them on. Only one interface will be used, the interface wich started working first after problems have gone.

The only remedy to make the tunnel use multipath is to restart tcp session of the VPN application.

Ysurac, 2019-11-08:
OK
I will make some tests and change the behavior. I also think multipath flag should never be removed, it's MPTCP role to use a path or not.

Ysurac, 2020-03-10:
Fixed in OpenMPTCProuter v0.54

closed 2020-03-10 by Ysurac