Ysurac/openmptcprouter #723: Multiple VPS

Roken120, 2019-11-09:
Hello,
I have one question is possible to use multiple vps ?

Roken120, 2019-11-09:
If yes can you explain how to use multiple vps ?

Ysurac, 2019-11-10:
Yes it's possible.
You have to run the install script on each VPS and to add them via the Wizard in OpenMPTCProuter.

Roken120, 2019-11-10:
for the part installations on the vps it is good. but how should I do for the wizard.

Ysurac, 2019-11-10:
You have a "add a server" button, you enter a name and click on this button.

Roken120, 2019-11-10:
Ok, is done, Maybe i can active on the two VPS the port forwarding ? (like two vps is master)

Ysurac, 2019-11-10:
No you can't. Port forwarding use the VPN, the VPN can't connect to the two VPS at the same time.

flapvhd, 2019-11-20:
What is the purpose of adding many servers?

Ysurac, 2019-11-20:
For VPS failover. If a VPS fail, then a backup VPS is used.

closed 2019-12-20 by Ysurac