Ysurac/openmptcprouter #726: how could I configure the default path manager ?

wttat, 2019-11-11:
## Expected Behavior

I wanna to use the Pi4 as a MPTCP router, i set the path manager to default, and wificlient wlan0 as backup, a 4G usbnet eth1 as master , however the flow appears in the wlan0 interface, after i disable the wlan0 , eth1 works, then after I up the wlan0, most traffic is using wlan0.
is it something about the networks hops? the wlan0's metric  is 5 and the eth1 is 6, I tried to alter it both in luci and the /etc/config/network but it will back to 5 and 6 automaticly
what i assume to do  is that during the normal time the backup link will not have any or only little traffic, while the master interface is down, the traffic will turn to the backup and when the master is back the traffic will back to the master as will. am i understand it correct of the 【'default': This path-manager actually does not do anything. The host won't announce different IP-addresses nor initiate the creation of new subflows. However, it will accept the passive creation of new subflows.】?
or what is the handover option works ? how should i set the options ?

  - OpenMPTCProuter version: v0.52.3
  - OpenMPTCProuter VPS version: i use my own ss proxy server with mptcp patch installed, it worked fine in fullmesh
  - OpenMPTCProuter platform: RPI4

Ysurac, 2019-11-11:
So what is the result and what is really the expected behavior ?
You should use the issue template...

wttat, 2019-11-11:
Sorry about that, I didn’t make myself clear.

I assume the default path manager means that the traffic only goes through the master link while none through the backup link, while the master link fail it will turn to the backup link, when the master is back it turn back to the master link with none left on the backup link.

The fact is no matter the setting the traffic seam to created on the backup wlan0 instead of the 4G eth1, so  I set the wlan0 to master while eth1 is backup.then I disable the wlan0 the eth1 started traffic , when I turn the wlan0 back on the traffic didn’t move back to wlan0.

Is it something about the network hop?The wlan networks metrics is 5 and the 4G is 6. And I couldn’t change it manually because somehow it will automatically back to 5and 6.

Or I get the default path manager’s function description wrong

wttat, 2019-11-11:
Btw, how could I set the application use the certain link of the MPTCP ? For example, the google use my wlan and the YouTube use the 4G, thx a lot

Ysurac, 2019-11-11:
Default path manager do nothing.
You can use omr-bypass to select the interface you want to use by domain/ASN/Protocol/...

wttat, 2019-11-11:
Thx about the link change problem.

If it does nothing, what is the different between the link’s enable and the master status ?how could I set which link it should use first? I suppose it means some kinds of the priority of the links?And what about the handover option , I didn’t see it in the mptcp website of socket control.Really appreciate your help

Ysurac, 2019-11-11:
master is used for default routing in all case. Then priority use routing parameters in default mode (so nothing related to MPTCP), or RTT in fullmesh. Handover is not tested/used so this shouldn't change really anything.

wttat, 2019-11-11:
well..so in default mode it shall follows the master or the routing setting ? Somehow it seemed use the backup link because of the smaller network hops..Why can’t I should the interface’s network hops ?

Ysurac, 2019-11-12:
In default mode it will follow the master and the routing setting as it's the same think.
If you set default mode instead of fullmesh, it use backup link because this mode do nothing and ignore all settings. And backup interface should not be set as master.

wttat, 2019-11-19:
> In default mode it will follow the master and the routing setting as it's the same think.
> If you set default mode instead of fullmesh, it use backup link because this mode do nothing and ignore all settings. And backup interface should not be set as master.

thx for your help.

wttat, 2019-12-02:
> Default path manager do nothing.
> You can use omr-bypass to select the interface you want to use by domain/ASN/Protocol/...

the omr-bypass didn't work as expected .
path-manager is full-mesh . 4g_eth1 mptcp_master and wan_eth0.2 mptcp_enabled for uplink.
in omr bypass demoin setting:
youku.com set for eth0.2
bilibili.com set for eth1.
while youku.com use both 4g and wan.
the bilibili.com refuse to connect.

is the omr-bypass means  directly use the selected link instead of shadowsocks ?
so when they try to bypass the master link it failed?

I used to consider it as use selected for mptcp traffic, or how could i set it mptcp's subflow to use the specific link ?

Ysurac, 2019-12-02:
How do you know that domains use both 4g and wan ?
For omr-bypass to work, OMR IP need to be your DNS. Also this work only for the domain and subdomains, if something come from another domain than it's not bypassed (and for example youku use alicdn.com).

wttat, 2019-12-02:
> How do you know that domains use both 4g and wan ?
> For omr-bypass to work, OMR IP need to be your DNS. Also this work only for the domain and subdomains, if something come from another domain than it's not bypassed (and for example youku use alicdn.com).
> OMR-Bypass use directly the selected interface and bypass VPN and shadowsocks.

I use the iftop to monitor both interface, at the same time I only open single video live website like youku.com and there are huge traffic on both link.
OMR IP be my DNS means I have to set the router = default gateway 's IP as DNS on my PC manully ?
Under this situation, they will be no MPTCP but only directly TCP traffic for the domain?

Ysurac, 2019-12-02:
No, this mean DNS of clients need to be OMR IP. If you set DHCP on clients this should be the case.
If you use OMR-Bypass then it's direct TCP traffic for the domain. Only the domain, not all the website that can use multiples domains like some CDN.

closed 2019-12-20 by Ysurac