Ysurac/openmptcprouter #727: Public MPTCP graph

JorgeMoort89, 2019-11-11:
Hi,
how can I have public MPTCP graph?

I've tried adding new file in /usr/lib/lua/luci/controller/ but is not working, if I add the module luci.controller.public_stats it ask me for the admin password and I want that graphs public

```
module("luci.controller.mptcp", package.seeall)

function index()
    assign({"bandwidth"}, {"admin", "network", "mptcp", "bandwidth"}, nil)
end
```

PRO2XY, 2019-11-13:
I am running after this too. Could you work it out?

JorgeMoort89, 2019-11-14:
Sadly no, I've created this but is not working without admin login

bandwidth.lua

module("luci.controller.bandwidth", package.seeall)

function index()
    assign({"bandwidth"}, {"admin", "network", "mptcp", "bandwidth"}, nil)
end

closed 2020-09-14 by github-actions[bot]