Ysurac/openmptcprouter #728: how do i connect my vps with shadowsocks

tianyun, 2019-11-14:
## Expected Behavior

 i have run the script of "debian9-x86_64.sh" in ubuntu 18.04.2, but when i connect the vps by shadowsocks-libev and show the follow information
`Nov 14 16:15:29 iZwz9cn5k7bpubs6pykj3tZ ss-server[655]:  2019-11-14 16:15:29 ERROR: failed to handshake with address: invalid address type`,
yes , I found out what caused it that  the password or cipher from https://github.com/shadowsocks/shadowsocks-libev/issues/1905

but i check it and there is no password and cipher problem, so, what do you think of it, thanks

Ysurac, 2019-11-14:
Do you use the wizard ?
On the VPS a key is used, not a password, so a key must be used on OpenMPTCProuter too.

tianyun, 2019-11-14:
> Do you use the wizard ?
> On the VPS a key is used, not a password, so a key must be used on OpenMPTCProuter too.

Thank you for your replay, i use the shadowsocks client at my pc(window 7) for test, and the password is the key in vps, there is any wrong, or i couldn't connect vps by ss clinet. Thank you again!

tianyun, 2019-11-25:
Yes, i reslove the problem, the client shadowsocks only offer the option of password, if it set up in openwrt, it works. Thank you!

closed 2019-11-25 by tianyun