Ysurac/openmptcprouter #729: Petite question technique (AS)

Slivikkus, 2019-11-14:
## Actual Behavior

Bonsoir, j'ai juste une simple question (vu que je suis un noob en réseau) et que je vais avoir un AS y'a la possibilité de relié l'AS au routeur ou c'est encore plus complexe que ça ?

Ysurac, 2019-11-14:
Ce n'est pas si simple que ça, mais je crois qu'il y a déjà une issue sur le sujet que je n'arrive pas à retrouver pour le moment.

Slivikkus, 2019-11-14:
Ok :)

flapvhd, 2019-11-16:
Le bloc d'IP que tu vas utiliser est annoncé par qui?

Slivikkus, 2019-11-16:
C'est à dire ?

flapvhd, 2019-11-16:
Le bloc d'IP qui va être associé a ton AS doit être annoncé sur internet par un routeur qui parle BGP.
Qui fait cette annonce? Car c'est depuis c'est ce routeur qui pourra router ces IP vers ton VPS, par un tunnel VPN ou autre...

Slivikkus, 2019-11-16:
Aucune idée

closed 2019-11-19 by Slivikkus