Ysurac/openmptcprouter #730: Boot BPI-R2 v0.52.3

phignx, 2019-11-15:
## Expected Behavior

 Boot BPI-R2 with openmptcprouter sd card image

## Actual Behavior
No boot, black screen and no response from 192.168.100.1 (WAN interface) 

## Steps to Reproduce the Problem

  1. dd bs=4M if=./openmptcprouter-v0.52.3-r011006-713561a10b-mediatek-mt7623-7623n-bananapi-bpi-r2-ext4-sdcard.img of=/dev/sdb conv=fsync
  2. install sd card in the slot (board boot-switch on sd)
  3. There are two rows of LEDs, one near the edge of the board and the other near the connectors.
Description of the lighting sequence after a reset:
  1 - simultaneous lighting of the leds (board edge)
  2 - extinction of the blue LED (board edge)
  3 - simultaneous lighting of the red and blue leds (connectors)
  4 - extinguishing of the blue and brief lighting of the green (connectors)
  5 - extinguishing the red led (connectors)
  6 - green LED off (board edge)
  7 - only the red led stays on (board edge)

## Specifications

  - OpenMPTCProuter version: 0.52.3
  - OpenMPTCProuter platform: BPI-R2

Ysurac, 2019-11-28:
I've not yet tested BPI-R2 on dev branch, if you want you can test them: https://www.openmptcprouter.com/release/develop/4.19/bpi-r2/targets/mediatek/mt7623/
(both -sdcard.img and sysupgrade-sd, they are not using the same bootloader).
You need to also use dev server script: https://github.com/Ysurac/openmptcprouter/wiki/Snapshots

I will test them myself ASAP.

phignx, 2019-12-02:
It was a stupid mistake
After connecting the USB converter cp2101, I found that I had not set the right mask: 32 instead of 24

closed 2019-12-02 by phignx