Ysurac/openmptcprouter #731: Tunnel down with saturated upload

JorgeMoort89, 2019-11-16:
I've experienced in every version of OMR router lot of problems when upload is saturated, for example:
- VPN tunnel DOWN -> A lot of tun0 VPN tunnel DOWN or ShadowSocks DOWN, probably caused by the high latency caused by the upload saturated, partially resolved with DSCP rules (Port 65001 with CS6 priority and ICMP on CS5), how to reserve bandwidth to OMR to do it checks?

- Fluctuating download -> When upload is satured I experience a lot of flactuating on download, how can I resolve that? I've already tried QoS with layer_of_cake and got minimum improvements

Ysurac, 2019-11-16:
You can configure SQM for that.

JorgeMoort89, 2019-11-16:
Already configured it on WANs, can maybe have any sense to set it LAN side?
What do you thinks is the best method to reserve bandwidth to OMR to do its checks?

Ysurac, 2019-11-16:
It should be needed only on WANs. You can be limited by LAN only if you can use more than max port bandwidth.
Best method is enabling SQM in the wizard with download and upload defined.

JorgeMoort89, 2019-11-17:
I've writed a script that automatically set limit for LAN 3 mbit/s less than the real aggregated bandwidth, I can adapt that for the WANs easly.
But even with that QoS sometimes tunnel goes down :/

Ysurac, 2020-05-18:
This will be solved in 0.55. If not, you are free to reopen this issue.

closed 2020-05-18 by Ysurac