Ysurac/openmptcprouter #732: Sessions not restarting on topology change?

WitoTV, 2019-11-17:
## Expected Behavior

I'm building a portable streaming solution and I wanted to add bonding for multiple network adapters (Wi-Fi, Ethernet, LTE Usb modems).
Whole thing is automated to select avaliable modems with little script I wrote for myself for sake of not needing to open the ui every time new modems is added/removed to/from the system (for now script is disabled)
OBS should use balance loaded network and not drop bitrate to 0 if one of the modems is unplugged

## Actual Behavior

If one of the modems or Wi-fi/Ethernet is turend off or unplugged OBS drops the bitrate to 0 and never goes back up. As well I noticed that from some reason it's not using max speed that is avaliable (1 modem + wi-fi connected should be able to use full speed from my home network as wi-fi by itself is able to achive 6k bitrate with no problem but when used in the OpenMPTCProuter it looks to be choking super hard)

## Steps to Reproduce the Problem

1. Install OpenMPTCProuter on the Hyper-V,
2. Set up 7 virtual switches (For now all internal) (Management, VM Modem 1-6),
3. Set up 7 network adapters in the Hyper-V VM (Management as first adapter so it will act as eth0),
4. On host system run NAT command to create new NAT:
4.1. `Get-NetAdapter` find `ifIndex` of `vEthernet (Management)`,
4.2.  `New-NetIPAddress –IPAddress 192.168.100.2 -PrefixLength 24 -InterfaceIndex IFINDEX-OF-MANAGEMENT`,
4.3. `New-NetNat -Name Management -InternalIPInterfaceAddressPrefix 192.168.100.2/24`,
5. Configurate OpenMPTCProuter:
5.1. Add your VPS details,
5.2. Add interfaces from VM Modem adapters (eth1 to eth6) on DHCP,
5.3. In System/Advances Settings/Network Settings set `IPv4 TPC FIN timeout` to 1,
6.  Change `VM Modem 1-6` depending of uyour needs for adapters (I have made myself automated script for that, 5 is always an ethernet port and 6 is wi-fi),
7. Install OBS: 
7.1. Configurate obs for streaming to twitch (this is relativly easy),
8. Set in OBS in Advanced/Network:
8.1. Bind to IP to `vEthernet (Management)`,
8.2. Enable `Dynamic change bitrate to manage congestion`,
9. Start the stream,
10. Unplug/disable one of the modems,
11. Stream will drop the bitrate to 0 and will not go back anymore. 

## Specifications

  - OpenMPTCProuter version: v0.52.3-r011006-713561a10b-x86-64
  - OpenMPTCProuter VPS version: OpenMPTCProuter Server 0.1001
  - OpenMPTCProuter platform: x86_64 (Hyper-V)
  - Device: Lattepanda 800s with Windows 10 Pro N

Ysurac, 2019-11-18:
I don't know how Hyper-V work, I don't use that.
All is green on the status page ?

WitoTV, 2019-11-18:
Yes all green on the status page. I’m suspecting it’s somehow related to fact that I’m not physically unplugging the device from  the router but Hyper-V switches the Virtual Switch to nternal from external. 

I used Hyper-V as I didn’t know and other solution to router network adapters this way to the OpenMPTCProuter.  (I went this path because my modems are being detected as usb storage when I plug them in and I wanted to reduce need of configurations every time I plug something into it.

Ysurac, 2019-11-18:
In any case when a modem doesn't answer anymore, if it's not the modem defined as master, you should not have any network failure.
Aggregation is really working ? You can test speed of each interfaces via SSH on the router with ```omr-test-speed wan1```,```omr-test-speed wan2``` (where wan1 and wan2 are real interfaces see with ```ip a```) and aggregated with ```omr-test-speed```

WitoTV, 2019-11-18:
Oh the master thing this is a good info. Is there a way to select for me which one is the master one? From what I see sometimes master is selected on one of the modems that is set as internal one in the UI.

WitoTV, 2019-11-18:
I do speed tests when I get home from work. But I'm guessing I'm somehow "lucky enough" to unplug master every single time.

WitoTV, 2019-11-18:
Bonus question: Is there "some way" to switch from master-slaves mode to something else?

WitoTV, 2019-11-18:
Fun fact
[image]
Nothing is plugged in from WAN2 to WAN5.

So why WAN2 is showing up as Master?

Ysurac, 2019-11-18:
Even if down, master stay defined.
You can change behavior in System->OpenMPTCProuter->Advanced parameters tab and set it to fixed. This can then be set in Network->MPTCP.

closed 2019-12-20 by Ysurac