Ysurac/openmptcprouter #733: Question about MPTCP: Transport Layer Keepalive

LovingFox, 2019-11-18:
## Expected Behavior

Is it possible to send a little bit data through the all of MPTCP sub-sessions (like a keepalive message) to keep the sub-session alive from the Middlebox's point of view?

## Actual Behavior

It may be heppend that the Middleboxes (like a provider's NAT-system or a Firewall) breaks out MPTCP sub-sessions due to timeout ("sleeping" sub-session). It is not good for a long MPTCP session: VPN-TCP-Tunnel, huge file downloading, etc. It can be broken and has to be re-established.

## Specifications

root@host:~# uname -a
Linux client1 4.19.72 #0 SMP Fri Sep 13 14:40:29 2019 aarch64 GNU/Linux

root@host:~# cat /etc/openwrt_release 
DISTRIB_ID='openmptcprouter'
DISTRIB_RELEASE='v0.52.3-5-g6f9cd2d'
DISTRIB_REVISION='r0+11006-713561a'
DISTRIB_TARGET='mediatek/mt7622'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='openmptcprouter v0.52.3-5-g6f9cd2d r0+11006-713561a'
DISTRIB_TAINTS='busybox override'

Ysurac, 2019-11-18:
You can configure TCP keepalive time but I don't really understand your problem.

LovingFox, 2019-11-19:
> ... I don't really understand your problem.

I try to explain.
There are two interfaces on the OMR. OMR connected to the Internet using these interfaces and through NAT/Firewall on the providers side. Also there is a long tcp-session on the OMR - vpn-tunnel. The MPTCP set up two sub-sessions for the VPN-tunnel through the two interfaces.

One of the interface has RTT=10ms to the VPN-server and other TT=100ms. And due to RTT the MPTCP use only interface RTT=10ms for data transfer. MPTCP only had set up tcp-sub-session through the interface RTT=100ms and nothin alse. That is a problem. Provider's NAT/Firewall over time just close that tcp-sub-session due to timeout. And if RTT=10 interface will gone for a some reason, vpn-tcp-session cannot use RTT=100ms interface.

Ysurac, 2019-11-19:
It's why there is "Re-create fullmesh subflows after a timeout" in network->MPTCP config.

LovingFox, 2019-11-19:
> It's why there is "Re-create fullmesh subflows after a timeout" in network->MPTCP config.

Ok. But how can I set a value of the timeout?

Ysurac, 2019-11-19:
It's TCP options, you can set net.tcp.retries2 for example.

LovingFox, 2019-11-20:
@Ysurac thx!
Decreasing this parameter improves behavior of long MPTCP session.

closed 2019-12-20 by Ysurac