Ysurac/openmptcprouter #734: Reverse Stream

nikfoxsentia, 2019-11-18:
Hi folks,

Just a small question. 

I have an application that sends out a UDP stream on port 5001 lets say, and the outbound side is working fine. It also then gets sent back another stream on port 5001 inbound from the remote PC (return audio). But that does not appear to be making it back.

I've got a PORT forward in for 5001 on OMPTCP to the device but no joy. Also tried UPNP. 

Any suggestions?

Raggazzoo, 2019-12-18:
Hi,

I got the same problem, I think we both want to use link aggregation to access to a SYNOLOGY NAS.

Thank you,
Thom

rubaz100, 2019-12-18:
Hi I got the same issue! I have a synology nas too!

Raggazzoo, 2019-12-18:
I think I figured out by enable udp in GloryTunVPN

closed 2020-05-18 by Ysurac