Ysurac/openmptcprouter #735: Problème avec le routeur sur ESXI

Slivikkus, 2019-11-19:
## Actual Behavior

J'ai mon routeur sur ESXI sur mon deuxième serveur mais j'ai souvent des problèmes au niveau du réseau... J'attribue des ip fixe au machines mais quand le routeur merde, donc je dois le redémarrer et ensuite la machine à une nouvelle ip ... Serez vous pourquoi ?

## Specifications

  - OpenMPTCProuter version: (0.52.3)
  - OpenMPTCProuter VPS version: (0.1001)
  - OpenMPTCProuter platform: (ESXI)

kevin39, 2019-11-19:
"La machine a une nouvelle IP" ?!
Tu parles d'OpenMTCPRouter ? De l'interface "lan" ? L'interface est en statique ip... C'est impossible sauf si tu l'a passée en DHCP.

Slivikkus, 2019-11-19:
Je parle d'une ip locale qui change tout le temps alors qu'elles sont définies fixes dans les paramètres DHCP

kevin39, 2019-11-19:
Quelle interface ?

Slivikkus, 2019-11-19:
LAN

kevin39, 2019-11-19:
Une interface en static ip ne peut pas changer d'IP à priori ;)
Ta config ressemble bien à ça ? 

[image]

Slivikkus, 2019-11-19:
Of course but je comprend pas pourquoi le routeur attribu très souvent une nouvelle ip à mes machines :/ (souvent à cause d'un redémarrage du routeur)

kevin39, 2019-11-20:
Tu n'es pas clair.
Si tu parles de tes machines du réseau c'est normal, c'est le principe du DHCP. Les IPs allouées aux périphériques réseaux sont renouvelables régulièrement.

Si tu veux que tes périphériques aient une IP fixe : 
* Option 1 : configurer les paramètres réseaux en "IP statique"
* Option 2 : Configurer le DHCP pour allouer une IP fixe aux périphériques concernés

Dans tous les cas, je ne crois pas qu'une issue soit l'endroit concerner pour parler de ça ;)

Ysurac, 2019-11-20:
Si j'ai bien compris, les baux DHCP sont définis en statique dans OpenMPTCProuter, mais après un reboot l'IP n'est plus la bonne ?
Est-ce que la configuration des baux est toujours présente dans OpenMPTCProuter ? Est-ce que tu n'as pas de soucis de lecture seule (indiqué dans la page status) ?
Est-ce qu'un autre serveur DHCP n'est pas présent sur le réseau ?

Slivikkus, 2019-11-20:
Yes, la configuration est toujours présente. Aucun autre serveur DHCP dans le réseau et pour la lecture seule je sais pas je redirais ça ce soir

Slivikkus, 2019-11-20:
Aucun problème d'écriture de mon côté 🤔🤔

closed 2019-11-26 by Slivikkus