Ysurac/openmptcprouter #736: Shadowsocks not stable.

bournehlm, 2019-11-20:
## Expected Behavior

  Sometimes shadowsocks not stable,  must restart shadowsocks to get back to work.

## Actual Behavior

## Steps to Reproduce the Problem

  1.PC connect to RPI3 lan port.
  2.After some time, PC can't connect to internet.
  3.PC ping BPI3 (192.168.100.1), OK
  4.BPI3 ping 8.8.8.8, OK.
  5.PC ping 8.8.8.8, FAIL.
  6.Restart shadowsocks, PC ping 8.8.8.8, OK.

Here is the log before restart shadowsocks.
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.267231] usbcore: registered new interface driver r8712u
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.275679] usbcore: registered new interface driver rndis_host
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.283739] usbcore: registered new interface driver rtl8150
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.292874] usbcore: registered new interface driver sierra_net
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.300848] usbcore: registered new interface driver sr9700
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.309753] usbcore: registered new interface driver usb_gigaset
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.316865] usb_gigaset: USB Driver for Gigaset 307x using M105
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.326680] usbcore: registered new interface driver usbhid
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.333402] usbhid: USB HID core driver
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.342109] usbcore: registered new interface driver usbserial_generic
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.349961] usbserial: USB Serial support registered for generic
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.358202] usbcore: registered new interface driver visor
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.364923] usbserial: USB Serial support registered for Handspring Visor / Palm OS
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.374896] usbserial: USB Serial support registered for Sony Clie 5.0
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.382734] usbserial: USB Serial support registered for Sony Clie 3.5
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.518127] xt_ndpi v1.2 ndpi 2.8.0 IPv6=YES debug_message=no
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.518127]  BT: hash_size 0k, hash_expiation 0 sec, log_size 128kb
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.518127]  sizeof hash_ip4p_node=44 id_struct=296 PATRICIA_MAXBITS=128
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.518127]  flow_struct=2064 packet_struct=1400
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.518127]    flow_tcp_struct=42 flow_udp_struct=27 int_one_line_struct=16
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.518127]  ndpi_ip_addr_t=16 ndpi_protocol=8 nf_ct_ext_ndpi=176
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.518127]  spinlock_t=4
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.518127]  NF_EXT_ID 7
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.568880] xt_ndpi: MAX_PROTOCOLS 320 LAST_PROTOCOL 243
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.575443] xt_ndpi: flow acctounting OFF
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.592530] xt_time: kernel timezone is -0000
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.599254] usbcore: registered new interface driver ark3116
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.606422] usbserial: USB Serial support registered for ark3116
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.616793] usbcore: registered new interface driver bas_gigaset
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.624203] bas_gigaset: USB Driver for Gigaset 307x
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.631589] usbcore: registered new interface driver belkin_sa
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.638862] usbserial: USB Serial support registered for Belkin / Peracom / GoHubs USB Serial Adapter
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.652275] usbcore: registered new interface driver cdc_mbim
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.702356] usbcore: registered new interface driver ch341
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.709287] usbserial: USB Serial support registered for ch341-uart
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.718185] usbcore: registered new interface driver cp210x
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.725221] usbserial: USB Serial support registered for cp210x
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.733862] usbcore: registered new interface driver cypress_m8
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.741191] usbserial: USB Serial support registered for DeLorme Earthmate USB
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.749861] usbserial: USB Serial support registered for HID->COM RS232 Adapter
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.758579] usbserial: USB Serial support registered for Nokia CA-42 V2 Adapter
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.771573] usbcore: registered new interface driver ftdi_sio
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.778739] usbserial: USB Serial support registered for FTDI USB Serial Device
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.788832] usbcore: registered new interface driver garmin_gps
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.796281] usbserial: USB Serial support registered for Garmin GPS usb/tty
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.809698] usbcore: registered new interface driver keyspan
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.816896] usbserial: USB Serial support registered for Keyspan - (without firmware)
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.827587] usbserial: USB Serial support registered for Keyspan 1 port adapter
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.836508] usbserial: USB Serial support registered for Keyspan 2 port adapter
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.845361] usbserial: USB Serial support registered for Keyspan 4 port adapter
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.916995] usbcore: registered new interface driver mct_u232
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.924177] usbserial: USB Serial support registered for MCT U232
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.933706] usbcore: registered new interface driver mos7720
Wed Nov 20 01:40:34 2019 kern.info kernel: [   10.940732] usbserial: USB Serial support registered for Moschip 2 port adapter
[... 679 lines cut ...]
Wed Nov 20 06:48:15 2019 daemon.info /usr/bin/ss-redir[7431]: initializing ciphers... chacha20
Wed Nov 20 06:48:15 2019 daemon.info /usr/bin/ss-redir[7431]: listening at 0.0.0.0:1100
Wed Nov 20 06:48:15 2019 daemon.info /usr/bin/ss-redir[7431]: tcp port reuse enabled
Wed Nov 20 06:48:15 2019 daemon.info /usr/bin/ss-redir[7431]: UDP relay enabled
Wed Nov 20 06:48:15 2019 daemon.info /usr/bin/ss-redir[7431]: udp port reuse enabled
Wed Nov 20 06:48:15 2019 daemon.info /usr/bin/ss-redir[7431]: running from root user
Wed Nov 20 06:48:15 2019 user.notice Shadowsocks: Rules DOWN
Wed Nov 20 06:48:15 2019 user.notice omr-tracker: Launching...
Wed Nov 20 06:48:15 2019 user.notice omr-tracker: Launched
Wed Nov 20 06:48:32 2019 daemon.info omr-tracker-ss: Shadowsocks is up (can contact 1.0.0.1)
Wed Nov 20 06:48:32 2019 daemon.info omr-tracker-ss: Reload Shadowsocks rules
Wed Nov 20 06:48:32 2019 user.notice Shadowsocks: Rules UP
Wed Nov 20 06:48:33 2019 user.notice Shadowsocks: Reload omr-bypass rules
Wed Nov 20 06:48:33 2019 user.notice omr-bypass: Starting OMR-ByPass...
Wed Nov 20 06:48:34 2019 user.notice omr-bypass: Reload dnsmasq...
Wed Nov 20 06:48:34 2019 daemon.info dnsmasq[15508]: read /etc/hosts - 4 addresses
Wed Nov 20 06:48:34 2019 daemon.info dnsmasq[15508]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Wed Nov 20 06:48:34 2019 daemon.info dnsmasq-dhcp[15508]: read /etc/ethers - 0 addresses
Wed Nov 20 06:48:34 2019 user.notice omr-bypass: OMR-ByPass is running

## Specifications

  - OpenMPTCProuter version: (0.52.3)
  - OpenMPTCProuter VPS version: (0.52.3)
  - OpenMPTCProuter platform: (RPI3)

Ysurac, 2019-11-20:
The problem is: "omr-tracker-ss: Server (101.37.66.246) seems down, no answer to ping".
Ping doesn't use Shadowsocks but the VPN, Shadowsocks is only used for TCP.
You can check the master route and set it to a better connection if possible (and set it as fixed in System->OpenMPTCProuter->Advanced parameters tab).
You can also check VPS logs.

bournehlm, 2019-11-20:
I'm sorry, You are right, Ping doesn't use Shadowsocks but the VPN, Shadowsocks is only used for TCP.
So, ignore the ping steps, just consider PC can't connect to internet.
Because after I restart shadowsock, PC can connect to internet again. I haven't do anything with the VPS server.
Cloud you please tell me where is the VPS logs location? thank you!

Ysurac, 2019-11-20:
If it's a debian, you can look at /var/log/daemon.log

closed 2019-11-23 by bournehlm