Ysurac/openmptcprouter #738: Discord ping important

deathcult456, 2019-11-20:
## Actual Behavior

Bonjour , désolé pour la question de noob, mais jai 2 WAN:
-WAN1: adsl2 free (ligne fixe) 
-WAN2: ligne 4G  free (box B525s-23a ) 
et discord/teamspeak  me fait des des monté de ping < 2000ms aléatoirement , je pense que c'est lorsque ça passe par WAN2 en 4G, donc je voudrais savoir si il est possible avec une plage de port de passer que par WAN1 .
Parce que si j'ai bien compris le OMR-BYPASS ne fait que de tagué l'adresse IP de WAN1 mais ne fait pas passer tous le flux par WAN1 ?
Voila sinon tous le reste fonctionne super bien c'est formidable merci beaucoup.

## Specifications

- OpenMPTCProuter version: 0.51.5
-OpenMPTCProuter VPS version: 0.999 4.19.56-mptcp / SCALEWAY 200M/b
- OpenMPTCProuter platform: x86  /intel nuc x86

Ysurac, 2019-11-20:
Avec omr-bypass il n'est pas encore possible de bypasser des ports, c'est pour la prochaine version en développement.

deathcult456, 2019-11-20:
ok merci beaucoup bon courage !

closed 2019-11-20 by deathcult456