Ysurac/openmptcprouter #739: No status page when no DNS services configured on the OMR router

sebvallee, 2019-11-22:
## Expected Behavior

Correct status page in any DNS configuration case (with and without DNS), even if some informations are not retrievable.

## Actual Behavior

When no DNS services is set (Unbound disabled, and no external DNS configured on the LAN interface), the status page doesn't show anything. In deeper inspection, it appears that the JSON is empty.

## Steps to Reproduce the Problem

  1. Disable Unbound in Services → Recursive DNS → Unbound → Basic → Uncheck "Enable Unbound"
  2. Remove any external DNS to the LAN interface (Network → Interfaces → "Edit" on LAN → "Use custom DNS servers" is empty)
  3. Navigate to System → OpenMPTCPRouter → "Status" tab

## Specifications

  - OpenMPTCProuter version: 0.52.3 (r0+11006-713561a10b)
  - OpenMPTCProuter VPS version: 0.1001 (Kernel 4.19.67-mptcp)
  - OpenMPTCProuter platform: x86_64 on Virtualisation (Proxmox VE 6.0-9)

Ysurac, 2019-11-22:
This should be working when "no external test" in advanced parameters is enabled, but I will make it working also if no DNS are working for next release.

Ysurac, 2019-12-20:
You can test with v0.53, if this still doesn't work you can reopen this issue.

closed 2019-12-20 by Ysurac