Ysurac/openmptcprouter #740: Question, How to modify VPS shadowsocks port?

bournehlm, 2019-11-22:
How to modify VPS shadowsocks port?
After I modify the server_port in /etc/shadowsocks-libev/config.json, suchas to 65102, that port will return to the default 65101 after few seconds.
And how to change other service port? suchas Glorytun port etc.

Ysurac, 2019-11-22:
Changing the port in shadowsocks-libev luci interface should change it on the VPS too. At least after a first working configuration. Same for glorytun.

bournehlm, 2019-11-23:
Got it, thanks.
When modify the shadowsocks port in the router's shadowsock luci interface, the VPS /etc/shadowsocks-libev/config.json server_port will be changed automaticlly.
How about two router use the same VPS, one router change the shadowsocks port in it's shadowsocks luci interface, another router will update the new shadowsocks port automaticlly?

Ysurac, 2019-11-23:
For now multiple OpenMPTCProuter on one VPS is not supported, so it's not an issue.
But there is a control on last change config date, so this should update also other router but this may also break.

bournehlm, 2019-11-23:
Oh. But I have test multiple OMR router with on VPS, it seems can work at the same time except the port forward.

Ysurac, 2019-11-23:
It's not a supported configuration. You can also have problems with the VPN, with IPv6,...

bournehlm, 2019-11-26:
I have test two OMR routers with on VPS again.
1. Both Glorytun and DSVPN support only one router, another one report VPN tunnel DOWN.
2. One router using Glorytun and another one using DSVPN, both VPN work fine.
3.Shadowsocks working fine with tcp and udp for both two routers.
4. If VPN tunnel DOWN, udp will switched to shadowsocks automaticlly.
5.If disable the VPN, the default router will be set to the master interface. 

So, if I disable the VPN,  tcp and udp traffic should pass to shadowsocks-->MPTCP-->VPS-->Internet ?
If I just want tcp and udp, can I use shadowsocks only without VPN, so I can use multiple router with one VPS.

And I found there is docker version OMR VPS https://github.com/Ysurac/docker-openmptcprouter-vps, but it seems that not working, and not continued.

Ysurac, 2019-11-26:
By default, VPN is used to aggregate all that is not TCP.
Shadowsocks only aggregate TCP. UDP can't be aggregated by shadowsocks.

Docker versions are not in development for now, maybe later.

bournehlm, 2019-11-26:
UDP can't be aggregated by shadowsocks? Even set "ss-redir for UDP" to "hi - tcp_and_udp"?

Ysurac, 2019-11-26:
TCP aggregation with shadowsocks use MPTCP. MPTCP is only for TCP, so no aggregation with shadowsocks and UDP.

closed 2019-12-20 by Ysurac