Ysurac/openmptcprouter #744: transplant to newifi3 failed with no ip address

wttat, 2019-11-25:
## Expected Behavior

Try to transplant the project to my newifi3 which based on ramips mt7621

## Actual Behavior

the pc connected to the router couldn't get an IP address both on the wann port or the lan port.

## Steps to Reproduce the Problem

created a config-newifi3 in the root folder with ：
```
CONFIG_TARGET_ramips=y
CONFIG_TARGET_ramips_mt7621=y
CONFIG_TARGET_ramips_mt7621_DEVICE_d-team_newifi-d2=y
CONFIG_TARGET_BOARD="ramips"
CONFIG_TARGET_SUBTARGET="mt7621"
CONFIG_TARGET_PROFILE="DEVICE_d-team_newifi-d2"
CONFIG_TARGET_ARCH_PACKAGES="mipsel_24kc"
CONFIG_DEFAULT_kmod-leds-gpio=y
CONFIG_DEFAULT_kmod-mt7603=y
CONFIG_PACKAGE_kmod-mt76x02-common=y
CONFIG_DEFAULT_kmod-mt76x2=y
CONFIG_DEFAULT_kmod-usb-ledtrig-usbport=y
CONFIG_PACKAGE_kmod-usb2=y
CONFIG_DEFAULT_kmod-usb3=y
CONFIG_PACKAGE_block-mount=y
CONFIG_PACKAGE_blockd=y
CONFIG_PACKAGE_kmod-usb-net=y
CONFIG_PACKAGE_kmod-usb-net-cdc-ether=y
CONFIG_PACKAGE_kmod-usb-net-rndis=y
```

then add 
```
elif [ "$OMR_TARGET" = "newifi3" ]; then
        OMR_REAL_TARGET="mipsel_24kc"
```
to the build.sh

after the build I use the breed console to install openmptcprouter-v0.52.3-7-g49e2b16-r0+11358-f526e85426-ramips-mt7621-d-team_newifi-d2-squashfs-sysupgrade.bin to the router

but my PC couldn't get the IP address, seems the DHCP not working？in that case I couldn't ssh to the router either, so where could the problem be?

## Specifications

  - OpenMPTCProuter version: 0.52.3.7
  - OpenMPTCProuter platform: ramips mt7621

wttat, 2019-11-25:
I am also trying to use the orgin openwrt git then add the openmptcprouter feed, choose the openmptcprouter-full in the menuconfig and the MPTCP's options in the kernel_menuconfig, it said that the dnsmasq in the base system is conflict with the dnsmasq-full,so i unselected the dnsmasq option, maybe this could work ?it is still building

wttat, 2019-11-25:
> I am also trying to use the orgin openwrt git then add the openmptcprouter feed, choose the openmptcprouter-full in the menuconfig and the MPTCP's options in the kernel_menuconfig, it said that the [...]

it failed the same as no IP address showed up. I wondered which part of the feed could affect the routing tables , or how could I solve this?
In the PC's network setting interface, there is no traffic from the router no whether which port I plug in the wire

Ysurac, 2019-11-25:
With the console you should be able to see what is the IP, what port is used and if it's really configured.
By using OpenMPTCProuter feed only, this can't work because you need a patched kernel at least.
And mt7621 is a slow CPU, I don't think this can work well...

wttat, 2019-11-25:
I have imported the MPTCP patch to the kernel and select the options in the kernel_menuconfig.

it seemed caused by the default setting of the omr, I have removed the root/ and the openmptcprouter feeds' network setting and it finally works. seems like the default omr setting is not work well on my router

wttat, 2019-11-25:
> With the console you should be able to see what is the IP, what port is used and if it's really configured.
> By using OpenMPTCProuter feed only, this can't work because you need a patched kernel at least.
> And mt7621 is a slow CPU, I don't think this can work well...

I have tryied lots of times configure the menuconfig ,  now I am pretty sure is the openmptcprouter/files/etc/uci-defaults makes it won't work, but i look through the code couldn't found anything related to specific router , I just wonder why it doesn't work on mine? the dhcp server or the firewall setting?  any ideas plz ?

wttat, 2019-11-25:
> With the console you should be able to see what is the IP, what port is used and if it's really configured.
> By using OpenMPTCProuter feed only, this can't work because you need a patched kernel at least.
> And mt7621 is a slow CPU, I don't think this can work well...

my PC can't acquire any IP and no gateway at all so that I couldn't log in to the router neither luci nor ssh， and there is no TTL to connect, so I could't check out whether it is working

Ysurac, 2019-11-25:
Without a TTL this will be difficult, you don't know if this boot well. You use a SD card ? flash available on this MIPS if  also too small.

wttat, 2019-11-26:
> Without a TTL this will be difficult, you don't know if this boot well. You use a SD card ? flash available on this MIPS if also too small.

this router got a 32mb rom so it's not the problem.
however the openmptcprouter/files/etc/uci-defaults 's setting makes it wrong .
could you show me how to modifiy these file that not turn the lan and wan around but leave anything else works? thank you very much
I tried simply not to select the openmptcprouter in menuconfig and it works but cause the unbound failed permission error and the uttpd error kept coming in the logread while lauching and the ss-redir somehow doesn't work either

wttat, 2019-11-26:
> Without a TTL this will be difficult, you don't know if this boot well. You use a SD card ? flash available on this MIPS if also too small.

I am trying to use one wan port and a 4g lte modem as uplink, the four lan port set as lan still, how i am suppose to change the openmptcprouter's setting , I changed it in luci before but I cant loggin my router and there is nums of setting regarding it ..the 1920-omr-network and 1980-omr-firewall..?  i trying to delete some wan3 and wan4 setting in other files ,but these two files I confused about the lan and the wan setting. would you plz help on these ?

Ysurac, 2019-11-26:
32Mo rom is a problem with a 600Mo images (and minimum 170Mo).
You can delete the ```1920-omr-network``` file.

wttat, 2019-11-26:
> 32Mo rom is a problem with a 600Mo images (and minimum 170Mo).
> You can delete the `1920-omr-network` file.

it works !!! thx very much~

i found that i tried deleted this file but i didnt copy the right bin to flash so  i consider it can't work. sad

thanks again!

closed 2019-11-26 by wttat