Ysurac/openmptcprouter #745: Quel interface pour l'IPv6

Slivikkus, 2019-11-25:
## Question

Bonsoir, 

J'aimerais savoir quel interface il faut lier pour avoir l'IPv6 (sur le VPS) ?

Merci :)

Ysurac, 2019-11-25:
Sur le VPS il faut juste configurer l'IPv6 sur l'interface principale portant normalement aussi l'IPv4 publique.

Slivikkus, 2019-11-25:
Mais justement je ne sais pas c'est la quel chez OVH, j'utilise le KVM directement que donne OVH donc pas facile de savoir :(

Ysurac, 2019-11-25:
Il faut aller voir la documentation (notamment pour avoir la passerelle et la configuration à mettre sur Debian) et la console d'administration (pour avoir l'IPv6). Et pour savoir sur l'interface à configurer : ```ip r``` et c'est celle à la ligne "defaut via... dev xxx"

Slivikkus, 2019-11-25:
OK je vois, ma config doit pas fonctionner vu que j'arrive pas encore à ping un site avec IPv6 (google.fr)

Slivikkus, 2019-11-25:
Je n'arrive pas à configurer l'IPv6 ça fait planter le réseau 🤔🤔 J'ai pourtant suivi le tutoriel d'OVH mais ça n'a pas l'air de fonctionner (j'ai peut être mal config)

closed 2019-11-26 by Slivikkus