Ysurac/openmptcprouter #747: Problème activation IPv6

Slivikkus, 2019-11-26:
## Actual Behavior

Bonsoir j'ai un problème quand je veux activer l'IPv6 sur le routeur, quand je l'active le routeur le désactive tout de suite alors que l'IPv6 est parfaitement configurer sur le VPS :)

C'est normal que je peut pas l'activer ?

## Steps to Reproduce the Problem

  1. Aller dans les paramètres
  2. Activé IPv6
  3. Le routeur le désactive

## Specifications

  - OpenMPTCProuter version: (0.52.3)
  - OpenMPTCProuter VPS version: (0.1001)
  - OpenMPTCProuter platform: (ESXI x86_64)

Ysurac, 2019-11-27:
Peux-tu me donner le résultat de ```ip -6 a``` sur le VPS ?

Slivikkus, 2019-11-27:
```
root@vps656421:~# ip -6 a
1: lo: <LOOPBACK,UP,LOWER_UP,80000> mtu 65536 state UNKNOWN qlen 1000
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 state UP qlen 1000
    inet6 2001:41d0:305:2100::7498/64 scope global
       valid_lft forever preferred_lft forever
    inet6 fe80::f816:3eff:fec7:b176/64 scope link
       valid_lft forever preferred_lft forever
3: dsvpn0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 9000 state UNKNOWN qlen 500
    inet6 64:ff9b::aff:fb01 peer 64:ff9b::aff:fb02/96 scope global
       valid_lft forever preferred_lft forever
    inet6 fe80::6d6e:9669:c8db:7267/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
5: omr-6in4@NONE: <POINTOPOINT,NOARP,UP,LOWER_UP,80000> mtu 1480 state UNKNOWN qlen 1000
    inet6 fe80::aff:ff01/64 scope link
       valid_lft forever preferred_lft forever
    inet6 fe80::a00:1/64 scope link
       valid_lft forever preferred_lft forever
6: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 state UNKNOWN qlen 100
    inet6 fe80::b268:86a7:d082:1579/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
7: gt-udp-tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1280 state UP qlen 500
    inet6 fe80::f97f:db93:bb8:fe62/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
9: mlvpn0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1444 state UNKNOWN qlen 500
    inet6 fe80::3e86:8175:3a8a:fa69/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
17: gt-tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 state UNKNOWN qlen 500
    inet6 fe80::59b4:89de:2519:727c/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
```
root@vps656421:~#

Ysurac, 2019-11-27:
Pas de soucis ici. As-tu essayé dans les onglets OpenMPTCProuter->assistant de configuration et dans la page OpenMPTCProuter->paramètre avancés ? Il y a un bug dans la version actuelle il me semble. Ca sera réglé dans la prochaine version.

Slivikkus, 2019-11-27:
Que ce soit dans paramètres avancés ou bien dans l'assistant de config ça ce décoche une fois la page refresh (pourtant paramètres save)

Slivikkus, 2019-12-01:
Bonsoir, du coup je dois attendre la prochaine mise à jour ?

Ysurac, 2019-12-01:
oui

Slivikkus, 2019-12-01:
D'accord une idée de quand la mise à jour sortira ? À ce que j'ai vu aussi OpenWRT prend en charge le BGP vous avez déjà regarder de ce côté ? :)

closed 2019-12-12 by Slivikkus