Ysurac/openmptcprouter #749: wan2 timing out constantly when using any VPS other than my original one

max4186, 2019-11-29:
For some reason whenever using a VPS other than what I originally configured all my settings on, my wan2 instantly times out and the bonding stops working. Nothing in system or kernel logs other than wan2 times out after 1 pings.

Is there a way to copy config from 1 VPS to another? I'm running Ubuntu 18.04 on both.

## Specifications

openmptcprouter v0.52.3 r0+11006-713561a10b / LuCI Master (git-19.260.67281-bb408f2)
OpenMPTCProuter VPS 0.1001 
RPI 4 Model B 64bit

max4186, 2019-11-29:
It just did it again and logged this time:

`Fri Nov 29 18:39:14 2019 user.notice post-tracking-post-tracking: omrvpn down
Fri Nov 29 18:39:14 2019 user.notice post-tracking-post-tracking: omrvpn down. Replace default route by 192.168.1.1 dev wan1
Fri Nov 29 18:39:14 2019 daemon.err uhttpd[2424]: luci: accepted login on /admin for root from 192.168.100.188
Fri Nov 29 18:39:28 2019 user.notice post-tracking-post-tracking: Get status and settings for wan2...
Fri Nov 29 18:39:31 2019 user.notice post-tracking-post-tracking: Get status and settings for wan2... Done
Fri Nov 29 18:39:42 2019 user.notice post-tracking-post-tracking: Get status and settings for wan2...
`

max4186, 2019-11-29:
It now appears to have just stopped doing it out of no where but I'm fairly confident it's going to do it again given it claims it couldn't ping when I was pinging directly from the 4G router itself and it was pinging with no spikes.

max4186, 2019-11-30:
It's happened again now and I can't fix it, it also doesn't let me change any settings like making my wan2 master which I assume means something hasn't installed properly? Have reinstalled 3 times, no fix.

Ysurac, 2019-11-30:
Did you try to reboot the router ?
For the master you need to make it static in OpenMPTCProuter->Advanced parameters tab if you want to change it.
I also need more logs.

max4186, 2019-12-08:
I rebooted everything numerous times, unfortunately there was no logs being generated more than I already posted. The issue hasn't happened in some time so I'm putting this down to having just been a temp bug and this can be closed.

Ysurac, 2019-12-20:
You can test with v0.53, if this still doesn't work you can reopen this issue.

closed 2019-12-20 by Ysurac

max4186, 2019-12-20:
Wish I didn't update now, it's now doing it again after not having done it for the past 20 days.

Fri Dec 20 08:26:43 2019 user.notice omr-tracker: Launching...
Fri Dec 20 08:26:43 2019 user.notice omr-tracker: Launched
Fri Dec 20 08:26:50 2019 user.notice post-tracking-post-tracking: Get status and settings for wan1...
Fri Dec 20 08:26:53 2019 user.notice omr-tracker: Can't create route to 8.8.8.8 via 192.168.8.1 dev wan2 src 192.168.8.2. waiting...
Fri Dec 20 08:26:57 2019 user.notice post-tracking-post-tracking: Get status and settings for wan1... Done
Fri Dec 20 08:27:02 2019 user.notice post-tracking-post-tracking: Get status and settings for wan1... Done
Fri Dec 20 08:27:10 2019 user.notice post-tracking-post-tracking: Get status and settings for wan1... Done

That is all both log files show. I've upgraded both the VPS and Openmptcp itself - is there a particular way I can tell which VPN for UDP is enabled on the VPS itself?

max4186, 2019-12-20:
2 more logs:

Fri Dec 20 08:40:37 2019 user.notice post-tracking-post-tracking: wan2 switched off because ping from 192.168.8.2 error (8.8.8.8,8.8.4.4,9.9.9.9)
Fri Dec 20 08:40:37 2019 user.notice post-tracking-post-tracking: Delete vps (185.42.223.161) route via 192.168.8.1 dev wan2

I can assure you that it CAN ping it however.....

max4186, 2019-12-20:
@Ysurac

Ysurac, 2019-12-20:
Can you ```ping -B -I wan2 192.168.8.1```, ```ping -B -I wan2 vpsip``` and ```ping -B -I wan2 8.8.8.8``` via SSH on the router ?

max4186, 2019-12-20:
I left it for a while and it's magically working again, same issue as I happened before - hopefully it's good going forward again. With this update, can we run that firmware update cmd? I forgot what the cmd was.

Ysurac, 2019-12-20:
yes, I will make a wiki page about that later today :)

max4186, 2019-12-21:
Did you manage to finish this? :)

Ysurac, 2019-12-21:
You can use ```rpi-eeprom-update -a```, then you need to reboot.

max4186, 2019-12-21:
root@OpenMPTCProuter:~# rpi-eeprom-update -a
-ash: rpi-eeprom-update: not found

Doesn't appear to work? @Ysurac

Ysurac, 2019-12-21:
I just did it, you have a RPI4 and use 64bits image ?

max4186, 2019-12-21:
Correct, Raspberry Pi 4 Model B Rev 1.1 running 0.53 64bit.

max4186, 2019-12-21:
Is there a particular way I can 100% confirm I'm running 64bit just to make sure?

Ysurac, 2019-12-21:
with ```uname -a```

max4186, 2019-12-21:
Linux OpenMPTCProuter 4.19.84 #0 SMP Tue Nov 19 13:59:03 2019 armv7l GNU/Linux

max4186, 2019-12-21:
Just realized I ran the 32bit one instead of 64.... Let me do it again.

Ysurac, 2019-12-21:
32bits and I forget to add it for 32bits image.
I will have to compile new images with it ;)

max4186, 2019-12-21:
No probs, hopefully sysupgrade to 64bit won't kill anything for now.

max4186, 2019-12-21:
Now running 64bit and have run the cmd, seems to have all went through without an issue other than the fact my 'Architecture' on front page says ? but all good :D