Ysurac/openmptcprouter #752: OpenMPTCProuter v0.53

Ysurac, 2019-12-02:
Status of v0.53rc1 before release of final release.

Router tests:
- [X] x86_64
- [ ] x86
- [ ] RPI4 64Bits
- [ ] RPI3 64Bits
- [ ] RPI2/3/4 32Bits
- [ ] BPI-R2
- [ ] WRT32X/WRT3200ACM

Server script tests:
- [X] Debian 9
- [X] Debian 10
- [X] Ubuntu 18.04
- [ ] Ubuntu 19.04

How to test: https://github.com/Ysurac/openmptcprouter/wiki/Snapshots

If you want to know what's new look at commit logs or wait for final release.

cayenne17, 2019-12-02:
[image]
Hi,
I just updated my OMR and OMV. 
after the update, I restarted my OMR and OMV.

Since the update, I have my downflow rate which has been halved.

On OMR, I have the following message: "Can't get public IP address from ShashowSocks"

On the VPS, when I do "systemctl status shadowsocks-libev.service" it tells me that it is disabled.

is that normal?

[image]

Ysurac, 2019-12-03:
You can try to restart VPS install script again.

user747, 2019-12-03:
congrats on v0.53rc1 milestone!

RPI4 64Bits
-also noticed down/up throughput to be halved when using `wget -O - http://www.openmptcprouter.com/server-test/debian9-x86_64.sh | sh` to set up a test VPS. 
I changed the server to a 2nd Debian 9 VPS I had running (set up using a modified version of debian9-x86_64.sh) and my throughput values were back to normal. 

-shadowsocks encrypted method defaulted to aes when vps-admin kicked in. unfortunately, they didn't license AES-NI for bcm2711 so it's not available.  

low priority edge case stuff:

- after rebuilding config from vanilla,  still seeing ss-redir flooding 10.255.254:65101 when omr-pihole is enabled, using gt-udp-tun0 on vps. conntrack goes up quite high and freezes rpcd. 

this rule stops the tcp flooding 
```
DROP       tcp  --  anywhere             10.255.254.1         tcp dpt:65101 /* !fw3: ss-redir */
DROP       udp  --  anywhere             10.255.254.1         udp dpt:65101 /* !fw3: ss-redir */

- glorytun-udp only (shadowsocks disabled) can only use one wan when using a omr-bypass domain example: setup netflix domains in omr-bypass. start watching netflix. now, start a download and see if you can use any other wan throughput. 
```

Ysurac, 2019-12-03:
MTU auto detect will be resolved for next RC.
AES will be fixed too (error in default VPS config).
I'm not able to reproduce pihole issue and glorytun with omr-bypass issue.

user747, 2019-12-03:
very nice. 
ill  investigate the edge cases further, but its looking like a local config issue on my end.  

auto mtu configures virtual macvlan
`uci -q set openmptcprouter.$OMR_TRACKER_INTERFACE.mtu=$mtu
`
curious to know if other uci mtu values should mirror `openmptcprouter.wan.mtu`
```
network.wan.mtu
network.wan_dev.mtu
```

noticed something in log about omr-tracker-ss and an xtables lock.
it could be "delays caused by slow DNS lookups"
if dns lookups are needed can ignore this. adding `-n` skips resolving. 

`[if [ -z "$(iptables -t nat -L | grep ss_rules_forward)" ] && [ "$(uci -q get shadowsocks-libev.ss_rules.disabled)" != "1" ]; then](https://github.com/Ysurac/openmptcprouter-feeds/blob/dffb81118bf00b0828380092be0c083ab23311f4/omr-tracker/files/bin/omr-tracker-ss#L66)`

Ysurac, 2019-12-03:
For MTU, I allow users to set it also manually. It's why I don't force it in network.wan.mtu but it's set manually via ip link, else I would have to add a checkbox or something like that to enable/disable auto-mtu...
Fix with -n will be added in next release everywhere it's needed (not only in ss-tracker)

user747, 2019-12-05:
> For MTU, I allow users to set it also manually.
Given 1 WAN, with a properly configured MTU, the performance gains can be somewhat small, depending on the underlying protocols (maybe 0.50% reduction in overhead in some cases) However, my guess is that when you start adding more WANs, the impact of a bad MTU value could have some sort of cascading packet fragmentation affect, which increases the overhead to a point where there is a noticeable performance drop. No data or tests to back this up though.

Ysurac, 2019-12-05:
True. It's why there is MTU detection.

rubaz100, 2019-12-21:
Hey, can you update your Port forwarding (How To Page)? I can't find the SNAT Configuration

8252610, 2019-12-22:
When running the wget -O - http://www.openmptcprouter.com/server/ubuntu18.04-x86_64.sh | sh 
Error occurred
/tmp/shadowsocks-libev-3.3.3.tar 100%[==========================================================>]   1.84M  13.9KB/s    in 2m 45s  

2019-12-22 11:36:05 (11.4 KB/s) - '/tmp/shadowsocks-libev-3.3.3.tar.gz' saved [1930597/1930597]

--2019-12-22 11:36:05--  https://raw.githubusercontent.com/Ysurac/openmptcprouter-feeds/master/shadowsocks-libev/patches/020-NOCRYPTO.patch
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.228.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.228.133|:443... failed: Connection refused.

My server is alicloud China

Scaff31, 2019-12-22:
@Ysurac hello you can check validate on ESXI ;)

PhatHub, 2019-12-25:
@Ysurac Just a few questions:

- Where is the configuration for MLVPN? I can't seem to find it on my router (x86_64 efi ext4) on the luci UI.
- I'm not sure if this is related, but the router can't get the Server Admin script from my VPS?
[image]
- Also not sure if this is related, but the system logs is full of OMR-VPS errors?
> Wed Dec 25 01:55:45 2019 user.notice OMR-VPS: Can't get token, try later
> Wed Dec 25 01:56:01 2019 user.notice OMR-VPS: Can't get token, try later

Someday I'm going to have to really crawl through your setup so I don't bother you with tiny things like this! :)

Ysurac, 2019-12-25:
MLVPN is removed, not able to make it working again without crash.
It's not related but Server was updated and rebooted ? If yes you can check in /var/log/daemon.log on the VPS if there is errors.

PhatHub, 2019-12-26:
Hey Merry Christmas @Ysurac ! 
I restarted the VPS and it works somehow? :confused: :disappointed_relieved: :laughing: 

Currently the router's MPTCPRouter (luci) status has the following errors:
> Can't get public IP address from ShadowSocks
> Can't contact Server Admin Script (No token) 

...but I'm getting network connectivity...?

closed 2020-03-10 by Ysurac

rifaterdemsahin, 2020-07-28:
from time to time the debian vps in gcp gets stuck
[image]

[image]

Ysurac, 2020-07-28:
Old closed issue and your issue is not related to this issue subject.
This doesn't seems to be related to OpenMPTCProuter at all.

rifaterdemsahin, 2020-07-28:
well said i realized it was the problem with the port change from 22 to 65222