Ysurac/openmptcprouter #756: Update VPS  version 0.1005

flimo44, 2019-12-21:
## Steps to Reproduce the Problem

  1.  wget -O - http://www.openmptcprouter.com/server/debian10-x86_64.sh | sh

  Update fail error : 
    update-alternatives --set iptables /usr/sbin/iptables-legacy  update-alternatives: erreur: pas d'alternatives pour iptables

If i comment go to end fine :
#if [ "$ID" = "debian" ] && [ "$VERSION_ID" = "10" ]; then
        #update-alternatives --set iptables /usr/sbin/iptables-legacy
        #update-alternatives --set ip6tables /usr/sbin/ip6tables-legacy
#fi

## Specifications

  - OpenMPTCProuter version: 0.53
  - OpenMPTCProuter VPS version:  0.10001 ( OVH Vps )
    Distributor ID: Debian
    Description:    Debian GNU/Linux 10 (buster)
    Release:        10
    Codename:       buster

  - OpenMPTCProuter platform:  x86_64)

Ysurac, 2019-12-21:
I modified the script and added "apt-get install iptables" this should be fixed.

flimo44, 2019-12-21:
merci

closed 2019-12-21 by flimo44